rerdavies/pipedal #103: More presets / examples

callelj, 2023-05-09:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I wish there were more examples or a way to share presets between users. Setups for specific song for example.

**Describe alternatives you've considered**
I guess each user could create their own presets. :-/

**Additional context**
N/A

rerdavies, 2023-05-10:
You can upload presets to,  or download presets from PiPedal using the Preset Manager in PiPedal (in the "..." menu beside the preset dropdown).

To be honest, I'm a jazz guitarist, and I don't have a great ear for rock or blues presets. I know they're there, but I'm probably not the best person to create them. Feel free to share presets in this discussion group. I'm more than open to adding contributed presets to the default set of presets, if there's suitable momentum. Or even creating a site to share presets if there's really serious momentum.

If you're using LV2 plugins that are not among the pre-installed set of plugins, don't forget to let people know what you used when you upload the presets.

rerdavies, 2025-05-14:
Since this issue was logged, patchstorage.com has added support for sharing PiPedal patches. Not a lot of traffic (yet), but hopefully it will catch on. See https://patchstorage.com/platform/pipedal/

And downloadable and uploadable presets and banks do provide a mehanism for sharing presets.

While I agree that there are still not enough samples and presets, I'm not sure that I'm going to make serious progress on this issue beyond what has already been done, so I'm going to close the issue (and give some thought as to whether there's anything else I can do at this point).

Thanks for your feedback. Please know that I have always given it serious consideration. 

Regards,

-R

closed 2025-05-14 by rerdavies