rerdavies/pipedal #109: Failed to connect to server

EE1977, 2023-05-16:
As of now, I have pipedal connected to my local ethernet network and I get a constant error while trying to connect from my desktop computer. The web browser refuses to load the pipedal interface and returns "Error. Failed to connect server". Any suggestions? As I recall, the last time I tried to connect via ethernet was on version 1.0.19 and I had no problems.

rerdavies, 2023-05-16:
1. Make sure pipedal is running.

Log onto your Raspberry pi, and issue the following command:

    sudo systemctl status pipedald

A couple of lines in, you should see:

    Active: active (running) since ....

If pipedal isn't running, issue the following command:

    sudo systemctl restart pipedald

and then rerun 

    sudo systemctl status pipedald

You should see something like the following:

```
May 16 14:54:50 raspberrypi pipedald[2244372]: Scanning for LV2 Plugins
May 16 14:54:51 raspberrypi systemd[1]: Started PiPedal Web Service.
May 16 14:54:52 raspberrypi pipedald[2244372]: Audio started. ALSA, hw:M2, 32-bit integer little-endian, 48000, 64x3, in: 1/2, out: 2/2
May 16 14:54:52 raspberrypi pipedald[2244372]: Listening on raspberrypi.local:80
May 16 14:54:52 raspberrypi pipedald[2244372]: Listening on 10.0.0.252:80

Pay particular attention to the lines that say "Listening on...". 

Try both the _**raspberrypi.local**_ address, and the explicit IP address.

Let me know how it goes.

EE1977, 2023-05-16:
Tried this and it stops @ systemd[1]: Started PiPedal Web Service. Below there are some ~ symbols and at the end, it says "lines 1-13/13 (END)"

EE1977, 2023-05-16:
By the way, I have the wifi disabled, if it anyhow matters...

EE1977, 2023-05-16:
OK, did this once more and it worked, although only @ 192.168.1.16:80. The "pi.local:80" also provided, seems like a dead link...

EE1977, 2023-05-16:
Is there a way to make this permanent? or even better auto-configurable?

rerdavies, 2023-05-16:
Ok. You may have to use the IP address instead of the pi.local address on your desktop. 

Windows doesn't seem to do that well with the DNS/SD address. (The pi.local address is published via DNS/SD). Sometimes it takes a while to discover the address; sometimes it doesn't find it at all. I don't think it's actually a bug with PiPedal. 

Things you might try, if you really want to pursue the pi.local address:

- Try rebooting your Windows system -- the theory being that Windows has cached the old Wi-Fi address and hasn't properly let go of it yet. That is a distinct possibility. I know Macs have a serious problem with this. And it's not impossible that Windows has the same problem too. PiPedal does un-publish the address when shutting down or rebooting. But avahi (the Linux DNS/SD server) may not un-publish if you turn off wi-fi on the Pi. 

- Try adding a period at the end of pi.local:

     http://pi.local.

See what 

     ping pi.local.

does (with and without the period).

- Try rebooting the Raspberry Pi and see if that makes a difference.

Again, let me know you find.

EE1977, 2023-05-16:
🤔I tried without the ".local" (just http://pi) and it just connects!!! without even a reboot...

rerdavies, 2023-05-17:
Huh. Did pi.local ever work for you?

EE1977, 2023-05-17:
I tried both but this rather turned out a disaster...😟 Now, neither of the addresses works anymore. Neither http://pi/ , nor Http://pi.local... I am kind of frustrated... I tried rebooting both the pi and my mac but the issue persists. Also, tried from my Android phone and I have the exact same issue. Server error! But for the facts, the address that gave a response while pinging was the pi.local one (@ 192.168.......). Any suggestions?

rerdavies, 2023-05-17:
Ok. Reset. Step by step. Precise answers please. I'm having a hard time understanding what you're seeing, because you're providing partial responses to my questions at each step. Details matter.

First step:

### Make, model, version of your pi.

What is the model of your Raspberry pi. How much memory?

Run the following command on your pi, and report the results.

   uname -a

Do you have more than one router on your home network?

Do you happen to remember which version of pipedal you're currently using? If you've upgraded in the last couple of weeks, there's no compelling need to be on the exact current version.

### Which address do and do not work?

My understanding from your last message is that `ping pi.local` works, but `ping pi` does not. Could you confirm that. If not, give me the exact response from the ping commands.

If neither address works, run 

    ping 192.168.1.16

Does that work?

If it does, does http://192.168.1.16 work?

If any of those work, skip the remaining steps, and report back.

### What is the exact error message you get from your browser? 

Assuming that any of the above addresses respond (pi, pi.local, or 192.168.1.16), what is the exact error you're seeing in your browser? Is it actually "Server Error"? Or is it "Address not found"? "Connection refused"?

### Wifi or Ethernet? 

Are you connected via wifi or ethernet? Have you changed recently (within the last couple of hours)?

### Does a local connection work?

Can you connect to the pipedal website from your raspberry pi?
 
### Is the pipedald service running? 

Connect to your raspberry pi. Check that the pipedald service is running. At a bash prompt on your pi:

     `sudo systemctl status pipedald`

Look for 

      `Active: active (running) since ....`

If the service is not active, issue the following command at a bash prompt on the raspberry pi:

     journalctl -b 0  | tail -n 100  

Are there significant errors?

If so, run the following command:

     journalctl -b 0  | tail -n 100  > /tmp/journal.txt

And then attach the file /tmp/JournalLogs.txt to your next response (drag to the bottom line of the editor), or email it to me if you prefer: rerdavies@gmail.com 

### Does restarting the pipedald service help?

Note that the pipedald service is configured to automatically restart if it crashes; so if you're seeing significant errors, you can skip this step.

If you do **_not_** see serious errors, try restarting the pipedal service, and see if that helps.

    'sudo systemctl restart pipedald`

EE1977, 2023-05-17:
I'll try to answer the questions one per sentence below.
Pi4B+ 8Gb. 
One router. 
Pipedal 1.1.24. 
Only 192.168.1.16 works. Full access on Mac (via ethernet) and on Android (via wifi on the same router). 
"Error: Failed to connect to server" with a Reload prompt.
Only Ethernet. Wifi is disabled on Pi side.

I have full access. to pipedal via Chromium on Pi on 127.0.0.1.

Hope that helps.

rerdavies, 2023-05-17:
Ok. Two problems.

### Problem 1: Name resolution not working.

To be honest, pipedal has very little to do with that. It's all handled by
Bonjour services on the Mac, DNS/SD services on Windows 11, and Avahi
services on the Raspberry Pi.

The one thing we haven't tried, but should: reboot your router! Your router
plays an important role in dispatching broadcast packets used by DNS/SD.
Perhaps your router is misbehaving.

### Problem 2: establishing the server connection.

I'd like to see the server log files I requested in the previous message,

    journalctl -b0 | tail -n 100 > /tmp/journal.txt

And the result of running

    uname -a

Three more things to check.

a) could you check the CPU use on your Raspberry Pi. Run `htop` (or use
another system utility of your choice if you prefer). You may have to
install it first: `sudo apt install htop`. This would show up in the system
log files; but you may as well do it first. Htop would also show whether
there are memory issues as well.

b) Check that your Pi isn't temperature throttling. Load PiPedal in the
browser on your Raspberry Pi. Temperature information, audio-thread CPU
use, and overruns are reported by Pipedal at the top of the Setting dialog,
and (by default) the bottom right-hand corner of the web page. A
temperature above 70C would be reason for concern, as would a significant
number of XRuns. The CPU use figure there, is for the audio thread only, so
it's less significant than the CPU usage figures reported by `htop`.

c) Possibly, the web server is running out of
resources/connections/something. Could you close PiPedal web pages in the
browsers of all your devices, and see if that allows devices to connect on
the 196... address.

rerdavies, 2023-05-17:
Re closing all the browsers... Are you using Safari on your Mac? Pay
particular attention to whether Safari is open, or running in the
background. It's possible that Safari does something that hangs the Pipedal
web server when it goes into the background. On Chrome/Chromium/Edge, when
the pipedal page goes into the background, the software disconnects its web
socket to save power. It's possible that Safari doesn't like that piece of
code, or doesn't allow the web socket to run in the background at all,
causing the web server to hang.

EE1977, 2023-05-18:
Looks like it is solved!! At last, I can now connect from all my devices. Just a router reboot and everything works great again!! I cannot see why that happened, but anyway, the job is done!! Thank you again for your precious help!!!

rerdavies, 2023-05-19:
Glad to hear it. Let me know if it happens again.

Out of curiosity, do you have an ASUS router? There were global reports of issues with a botched ASUS router update yesterday, that would potentially match the symptoms you were experiencing.

### If name resolution fails

The router is definitely involved with broadcast routing of DNS/SD name resolution. That's a problem that will need a router reboot if it happens again. 

### if PING pi works, but connecting to the web server doesn't work.

If PING works for for the pi or pi.local address, but the browser won't connect, keep in mind the following. There is a quirk in how Chrome handles http connections. I wonder if it may have contributed to the problems we saw. 

IoT devices and appliances are explicitly _not allowed_ to use https connections. (Sounds ridiculous, but it is true. Browsers will not allow use of HTTPS on private 10. or 192.168. addresses). So pipedal uses an http connection. If you type a web page address into Chrome's address bar, and don't specify the http:// or https:// part of the address, Chrome defaults to using https.  So when connectiong to pipedal, you _must_ explicitly type the http part! The address must be entered in full.

    http://pi

or 

   http://pi.local

Is it possible that you were entering "pi" instead of "http://pi"? It's any easy mistake to make because of the way that Chrome's address bar works.

### If the initial connection works, but you get stuck at "loading.... loading... Cannot connect to server".

If I understood your descriptions earlier, there is a second problem: a situation where you can connect to the web server, but the web page cannot establish a web socket connection to the pipedal server. 

If this happens you will see a PiPedal error message (instead of a default browser error message). If you see this, please report back here.  I may have briefly reproduced it here; but it self-corrected, for me, and I haven't been able to replicate it since.

This is by far the oddest part of the problem; and it potentially indicates a problem with pipedal that needs to be corrected. 

So. If the initial web page loads, but you get stuck at "Loading... Loading... Loading..." and then get an error message "Cannot connect to server", do the following for me, please.

    1. Close and reopen your browser, and try again.
    2.   If that doesn't work, do the following: restart the pipedald service, and try again.
    3. If that doesn't work, do the following; reboot your router and try again.

Please report the results here (whether successful or unsuccessful).

It would also REALLY be helpful (and save much pain and grief) if you could send the system logs after doing so. The following command captures only log messages from pipedal only (so no personal information disclosure):
 
    `journalctl -b 0 | grep pipedald > /tmp/journal.txt`

Although the following is more useful to me:

    `journalctl -b 0 | tail -n 100 > /tmp/journal.txt`

Email the result to me at rerdavies@gmail.com. If you have concerns about security, review the contents yourself, and edit or snip bits that you feel might disclose undesirable information. There shouldn't be much other than a few IP addresses, and very limited disclosure that you are running standard Linux services. But your mileage may vary.

rerdavies, 2023-05-21:
Provisionally closed. Keep open for a while, in case the issue recurs.

rerdavies, 2023-05-23:
Rebooting the router solved all problems. Possibly a symptom of a botched Asus router update.