rerdavies/pipedal #111: No input signal with Harley Benton MP500 USB audio interface

TheMojoMan, 2023-05-18:
I have installed PiPedal on Armbian running on my X96Max+ TV box (CPU: Amlogic S905X3, 4 x Cortex A55, kernel: 6.1.27).

I use a Harley Benton MP500 (aka MeloAudio TS Mega) USB footswitch/audio interface. When switching the sample rate of the MP500 in the settings window of PiPedal the rate changes accordingly in the display of the MP500.

Unfortunately, no matter which input of the MP500 I choose in the settings menu (Stereo/left/right) PiPedal does not get any input from my guitar. But I can hear the guitar clearly when using direct monitoring. When disabling pipedal.service I can also clearly hear music when I play it with the command `aplay -Dhw:Card=MP500 sample.wav`.

When I use Manjaro KDE instead of Armbian I have no problems working with the MP500 using Ardour. MOD-HOST and MOD-UI work as well. No problems with input and outputs.

Do you have any idea how to get input/output working when using PiPedal and the MP500?

I have also successfully compiled and installed PiPedal from source (after testing with the supplied Debian package first) but it does not change anything.

rerdavies, 2023-05-19:
Thanks for your problem report.

Out of curiosity, how did you get kernel 6.1 running on an Amlogic device? Last time I checked, they were stuck on 4.x or early 5.x kernels.  If you're really running a 6.1 kernel, I may drag a couple of old S905s out of a shoebox and put them to actual practical use. 

The question of kernel version is critically important because USB audio is effectively non-functional on kernel versions prior to 5.10 (except for a handful of special-cased USB audio adapters, most of which are no longer shipping. And there are huge improvements to USB audio performance and stability that occur c. 5.15-5.20. 

If you could just confirm for me that you are using a  6.1.27 kernel, I'd be grateful.  Run `uname -a` to check. If it's less than 5.15  , all bets are off, I'm afraid.

Questions, to get us started:

- Once you choose a frequency and buffer size in the PiPedal settings dialog, does the status information reflect that audio playback has started successfully? If successful, the STATUS section should look something like: 

         XRuns: 0 CPU: 3.4% 46.7C
         Governor: performance 1.800 GHz
    Anything else means that the audio isn't actually running. And a large number of XRuns may indicate that the audio thread isn't running _for all practical purposes_.

- how much RAM on your device? 

- Can you route audio input to audio output when using Ardour on Armbian?  (Or qjackctl, if that's easier).

- Check system logs to make sure that the pipedal service is running properly.

        journalctl -b 0 | grep pipedald | less 
 
- did you build Pipedal on your device (I'm looking for a lower bound on how much RAM is required to build Pipedal, so if you did, that would be useful information for me). I know that a VSCode build needs at least 4GB, and will be uncomfortable to debug.

TheMojoMan, 2023-05-19:
Thank you so much for your reply and help!

I have the 4GB model and I am indeed running the 6.1.27 kernel. You can find respective Armbian images for S9xx here

   [https://github.com/ophub/amlogic-s9xxx-armbian ](https://github.com/ophub/amlogic-s9xxx-armbian)

(Little history:
When @Balbes150 (main developer of Armbian) ditches support for Amlogic CPUs some years ago I switched to Manjaro which supported every kernel from 4.x to 5.x and now 6.x; but setup is a little more tricky).
I only recently switched back to Armbian because it is easier to quickly compile stuff when Debian dependencies are listed in the 'How to compile'.)

I actually use two SD cards for testing:
1. Manjaro KDE / Wayland / Pipewire
2. Armbian server image / no desktop environment / ALSA

So far, everything works very well when using the Manjaro image. I can run Neural Amp Modeler and IR loader in Ardour. But latency is bad. I need to use a buffer of 720 samples to not get XRuns due to running full desktop. When disabling desktop I can get down to 512 samples when using MOD-HOST & MOD-UI. Still not too great.
I did not compile Pipedal on Manjaro, yet (did not have time, yet, to figure out the needed packages for compiling).

The problems I describe here refer to using the Armbian server image.
As you have suggested I have now also tried to connect audio input to audio output via Jack and I do hear my guitar!

Here are the output messages (pipedal services not running):
```
$ uname -a
Linux armbian 6.1.27-ophub #1 SMP Mon May  1 13:03:40 EDT 2023 aarch64 GNU/Linux
$ jackd -R -dalsa -dhw:MP500 -r48000 -p256 &
...
$ jack_connect system:capture_1 system:playback_1
$ jack_connect system:capture_1 system:playback_2
$ jack_lsp -c
system:capture_1
   system:playback_1
   system:playback_2
system:capture_2
system:playback_1
   system:capture_1
system:playback_2
   system:capture_1
```

Answering your remaining questions (after starting pipedal services again and reboot):
1. Status section in settings shows XRuns: 0 CPU: 0.5% 54.4C, Governor: performance 2.100 GHz after setting frequency and buffer size in pipedal.
2. Yes, pipedal.service is running:
```
$ journalctl -b 0 | grep pipedald | less
Mai 19 11:14:18 armbian pipedald[1133]: Document root: /etc/pipedal/config Threads: 10
Mai 19 11:14:19 armbian pipedald[1133]: Scanning for LV2 Plugins
Mai 19 11:14:24 armbian pipedald[1133]: Audio started. ALSA, hw:MP500, 32-bit integer little-endian, 480
00, 512x2, in: 1/2, out: 2/2
Mai 19 11:14:24 armbian pipedald[1133]: Listening on armbian.local:80
Mai 19 11:16:16 armbian pipedald[1133]: Web Socket Address: 192.168.0.16:80
Mai 19 11:16:16 armbian pipedald[1133]: WebSocketSession opened. ([::ffff:192.168.0.133]:50432)
Mai 19 11:20:58 armbian pipedald[1133]: WebSocketSession opened. ([::ffff:192.168.0.133]:50508)
```
3. Yes, I did build pipedal on my 4GB X96Max+ running Armbian.

I will try later to build Pipedal using the Manjaro image (but that might take me a while to figure everything out).

Thank you again for your help!

TheMojoMan, 2023-05-19:
BTW, since I started from scratch several times I currently only have the lv2 plugins installed that come with pipedal.
```
ls /usr/lib/lv2/
atom.lv2         event.lv2            options.lv2      presets.lv2      ToobAmp.lv2  worker.lv2
buf-size.lv2     instance-access.lv2  parameters.lv2   resize-port.lv2  ui.lv2
core.lv2         log.lv2              patch.lv2        schemas.lv2      units.lv2
data-access.lv2  midi.lv2             port-groups.lv2  state.lv2        urid.lv2
dynmanifest.lv2  morph.lv2            port-props.lv2   time.lv2         uri-map.lv2
```

rerdavies, 2023-05-19:
Everything looks good in the log files, and in the status info.

Do you see any activity on the vu meters. Maybe start with an empty patch (click/press on the ... menu to the right of the patch selection dropdown, and click "New...".

Select the empty node in the upper patch view, and see if there's any activity on the VU meter on the left-hand side. Click on the starting node before that, and check that the volume dial hasn't been set to -INF. Is there any activity in the VU meter there?

rerdavies, 2023-05-19:
Not much point in building for Manjaro if the latency is dismal. 

A note to file away for after we solve this: pipedal has been mostly tested on "SMP PREEMPT" kernels. Rapsberry Pi OS comes with an SMP PREEMPT kernel by default. Linux Mint allows you to install an SMP PREEMPT kernel via 'apt install linux-realtime'; and Ubuntu...  I Think I may have installed an SMP RT PREEMPT kernel, or used Ubuntu Studio, which comes with an SMP PREEMPT kernel pre-installed. Armbian supports it, but may require a custom kernel build, which doesn't look hugely difficult, since it's menu driven. See here: https://github.com/armbian/build

I'm not 100% certain, but I think PREEMPT will make a significant difference; and that PREEMPT_RT would be even better. (But it won't fix the problem we're currently dealing with).

TheMojoMan, 2023-05-19:
I was finally able to compile Pipedal using my Manjaro image but unfortunately pipedal.service crashes:
ˋˋˋ
$ journalctl -b 0 | grep pipedal
Mai 19 21:04:15 X96MaxPlus pipedald[455]: Document root: /etc/pipedal/config Threads: 10
Mai 19 21:04:15 X96MaxPlus pipedald[455]: Scanning for LV2 Plugins
Mai 19 21:04:58 X96MaxPlus pipedald[455]: lilv: lilv_world_load_bundle(): warning: Replacing version 2.0 of <http://kxstudio.sf.net/carla/plugins/audiogain> from <file:///usr/lib/lv2/carla-audiogain.lv2/>
Mai 19 21:04:58 X96MaxPlus pipedald[455]: lilv: lilv_world_load_bundle(): note: New version 4.25 found in <file:///usr/lib/lv2/carla.lv2/>
Mai 19 21:04:58 X96MaxPlus pipedald[455]: lilv: lilv_world_load_bundle(): warning: Replacing version 2.0 of <http://kxstudio.sf.net/carla/plugins/audiogain_s> from <file:///usr/lib/lv2/carla-audiogain.lv2/>
Mai 19 21:04:58 X96MaxPlus pipedald[455]: lilv: lilv_world_load_bundle(): note: New version 4.25 found in <file:///usr/lib/lv2/carla.lv2/>
Mai 19 21:04:58 X96MaxPlus pipedald[455]: lilv: lilv_world_load_bundle(): warning: Replacing version 4.25 of <http://kxstudio.sf.net/carla/plugins/audiofile> from <file:///usr/lib/lv2/carla.lv2/>
Mai 19 21:04:58 X96MaxPlus pipedald[455]: lilv: lilv_world_load_bundle(): note: New version 150.22 found in <file:///usr/lib/lv2/carla-files.lv2/>
Mai 19 21:04:58 X96MaxPlus pipedald[455]: lilv: lilv_world_load_bundle(): warning: Replacing version 4.25 of <http://kxstudio.sf.net/carla/plugins/midifile> from <file:///usr/lib/lv2/carla.lv2/>
Mai 19 21:04:58 X96MaxPlus pipedald[455]: lilv: lilv_world_load_bundle(): note: New version 150.22 found in <file:///usr/lib/lv2/carla-files.lv2/>
Mai 19 21:04:58 X96MaxPlus pipedald[455]: lilv: lilv_world_load_bundle(): warning: Replacing version 0.2 of <http://lsp-plug.in/plugins/lv2/impulse_responses_mono> from <file:///usr/lib/lv2/lsp-plugins-impulsantworten.lv2/>
Mai 19 21:04:58 X96MaxPlus pipedald[455]: lilv: lilv_world_load_bundle(): note: New version 0.12 found in <file:///usr/lib/lv2/lsp-plugins.lv2/>
Mai 19 21:04:58 X96MaxPlus pipedald[455]: lilv: lilv_world_load_bundle(): warning: Replacing version 0.2 of <http://lsp-plug.in/plugins/lv2/impulse_responses_stereo> from <file:///usr/lib/lv2/lsp-plugins-impulsantworten.lv2/>
Mai 19 21:04:58 X96MaxPlus pipedald[455]: lilv: lilv_world_load_bundle(): note: New version 0.12 found in <file:///usr/lib/lv2/lsp-plugins.lv2/>
Mai 19 21:04:58 X96MaxPlus pipedald[455]: Audio started. ALSA, hw:MP500, 32-bit integer little-endian, 48000, 512x2, in: 2/2, out: 2/2
Mai 19 21:04:58 X96MaxPlus pipedald[455]: Failed to create plugin 'TooB Tuner'.
                                                  #0  0x0000aaaad1263a7c _ZN7pipedal13Lv2Pedalboard12PrepareItemsERSt6vectorINS_14PedalboardItemESaIS2_EES1_IPfSaIS6_EERNS_22Lv2PedalboardErrorListE (pipedald + 0x1e3a7c)
                                                  #1  0x0000aaaad12642e0 _ZN7pipedal13Lv2Pedalboard7PrepareEPNS_5IHostERNS_10PedalboardERNS_22Lv2PedalboardErrorListE (pipedald + 0x1e42e0)
                                                  #2  0x0000aaaad12d32fc _ZN7pipedal10PluginHost19CreateLv2PedalboardERNS_10PedalboardERNS_22Lv2PedalboardErrorListE (pipedald + 0x2532fc)
                                                  #3  0x0000aaaad11f8b48 _ZN7pipedal12PiPedalModel21LoadCurrentPedalboardEv (pipedald + 0x178b48)
                                                  #4  0x0000aaaad11f0a7c _ZN7pipedal12PiPedalModel4LoadEv (pipedald + 0x170a7c)
                                                  #5  0x0000aaaad10b0558 main (pipedald + 0x30558)
                                                  #8  0x0000aaaad10af830 _start (pipedald + 0x2f830)
                                                  #3  0x0000aaaad1314b88 _ZN7pipedal10RingBufferILb0ELb1EE8readWaitEv (pipedald + 0x294b88)
                                                  #4  0x0000aaaad131383c _ZN7pipedal16HostWorkerThread10ThreadProcEv (pipedald + 0x29383c)
                                                  #5  0x0000aaaad1313bdc _ZZN7pipedal16HostWorkerThreadC4EvENKUlvE_clEv (pipedald + 0x293bdc)
                                                  #6  0x0000aaaad13144e4 _ZSt13__invoke_implIvZN7pipedal16HostWorkerThreadC4EvEUlvE_JEET_St14__invoke_otherOT0_DpOT1_ (pipedald + 0x2944e4)
                                                  #7  0x0000aaaad131449c _ZSt8__invokeIZN7pipedal16HostWorkerThreadC4EvEUlvE_JEENSt15__invoke_resultIT_JDpT0_EE4typeEOS4_DpOS5_ (pipedald + 0x29449c)
                                                  #8  0x0000aaaad1314438 _ZNSt6thread8_InvokerISt5tupleIJZN7pipedal16HostWorkerThreadC4EvEUlvE_EEE9_M_invokeIJLm0EEEEvSt12_Index_tupleIJXspT_EEE (pipedald + 0x294438)
                                                  #9  0x0000aaaad131440c _ZNSt6thread8_InvokerISt5tupleIJZN7pipedal16HostWorkerThreadC4EvEUlvE_EEEclEv (pipedald + 0x29440c)
                                                  #10 0x0000aaaad13143ec _ZNSt6thread11_State_implINS_8_InvokerISt5tupleIJZN7pipedal16HostWorkerThreadC4EvEUlvE_EEEEE6_M_runEv (pipedald + 0x2943ec)
                                                  #2  0x0000aaaad129fed4 _ZN7pipedal14AlsaDriverImpl10ReadBufferEP8_snd_pcmPhm (pipedald + 0x21fed4)
                                                  #3  0x0000aaaad12a0258 _ZN7pipedal14AlsaDriverImpl11AudioThreadEv (pipedald + 0x220258)
[... cut ...]
InvokerISt5tupleIJZN13AudioHostImpl17StartReaderThreadEvEUlvE_EEEEE6_M_runEv (pipedald + 0x1daea8)
Mai 19 21:05:15 X96MaxPlus systemd[1]: pipedald.service: Main process exited, code=dumped, status=11/SEG
V
Mai 19 21:05:15 X96MaxPlus systemd[1]: pipedald.service: Failed with result 'core-dump'.
Mai 19 21:05:15 X96MaxPlus systemd[1]: pipedald.service: Consumed 21.713s CPU time.
Mai 19 21:05:40 X96MaxPlus systemd[1]: pipedald.service: Scheduled restart job, restart counter is at 1.
Mai 19 21:05:40 X96MaxPlus systemd[1]: pipedald.service: Consumed 21.713s CPU time.
ˋˋˋ

TheMojoMan, 2023-05-19:
> Do you see any activity on the vu meters. Maybe start with an empty patch (click/press on the ... menu to the right of the patch selection dropdown, and click "New...". ...
I started with an empty patch but I do not see any activity on the vu meters. I have also checked that input and output is set to 0 db.

rerdavies, 2023-05-19:
Ok. Questions.

1. Does your device have 2 input channels or 3 input channels. It seems to me that there should be 3: 1 mic input and L/R guitar inputs. But PiPedal seems to think there are only 2. PiPedal is supposed to support more than 2 channels; but the code is untested. If there really are three channels, PiPedal should show a dialog allowing selection of 3 channels independently by channel name, instead of the choice of Stereo, Left, Right that it shows when there are only two channels.

2.  Are the input VUs active on the display of the MP-500?

3. Dump some configuration files for me. 
    Run `aplay -l` to figure out what the card number is for your MP-500. Probably "card 2".
     Then run the following commands, adjusting the card number if necessary:

         cat /proc/asound/card2/usbmixer

         cat /proc/asound/card2/pcm0c/sub

rerdavies, 2023-05-19:
Final answer:

     cat /proc/asound/card2/usbmixer
     amixer -D hw:MP500 contents

rerdavies, 2023-05-19:
And if you could provide the contents of the following file, please.  Before you do so, select Stereo as the channel selection.

    /var/pipedal/JackChannelSelection.json

And the following would be interesting: plug your gutar into the Left Input (2nd from the left); and select Mono (Right Channel) in the settings dialog.

TheMojoMan, 2023-05-20:
1. Physically, the device has 3 inputs: L-IN, R-IN and XLR IN(R). But you cannot use R-IN or XLR IN(R) at the same time. Ardour, Cubasis and Jack therefore only list 2 inputs.
2. The display of the MP500 shows something that look like VU meters but they show nothing when using Ardour/Cubasis/Jack.
3. See below:

```
$ cat /proc/asound/card1/usbmixer
USB Mixer: usb_id=0x2eee0012, ctrlif=3, ctlerr=0
Card: MA MP500 at usb-xhci-hcd.3.auto-2, high speed
USB Mixer: usb_id=0x2eee0012, ctrlif=0, ctlerr=0
Card: MA MP500 at usb-xhci-hcd.3.auto-2, high speed
  Unit: 10
    Control: name="MP500  Playback Volume", index=1
    Info: id=10, control=2, cmask=0x0, channels=1, type="S16"
    Volume: min=-32688, max=0, dBmin=-12768, dBmax=0
  Unit: 10
    Control: name="MP500  Playback Volume", index=0
    Info: id=10, control=2, cmask=0x3, channels=2, type="S16"
    Volume: min=-32688, max=0, dBmin=-12768, dBmax=0
  Unit: 10
    Control: name="MP500  Playback Switch", index=1
    Info: id=10, control=1, cmask=0x0, channels=1, type="INV_BOOLEAN"
    Volume: min=0, max=1, dBmin=0, dBmax=0
  Unit: 10
    Control: name="MP500  Playback Switch", index=0
    Info: id=10, control=1, cmask=0x3, channels=2, type="INV_BOOLEAN"
    Volume: min=0, max=1, dBmin=0, dBmax=0
  Unit: 11
    Control: name="Mic Capture Volume", index=1
    Info: id=11, control=2, cmask=0x0, channels=1, type="S16"
    Volume: min=-32688, max=0, dBmin=-12768, dBmax=0
  Unit: 11
    Control: name="Mic Capture Volume", index=0
    Info: id=11, control=2, cmask=0x3, channels=2, type="S16"
    Volume: min=-32688, max=0, dBmin=-12768, dBmax=0
  Unit: 11
    Control: name="Mic Capture Switch", index=1
    Info: id=11, control=1, cmask=0x0, channels=1, type="INV_BOOLEAN"
    Volume: min=0, max=1, dBmin=0, dBmax=0
  Unit: 11
    Control: name="Mic Capture Switch", index=0
    Info: id=11, control=1, cmask=0x3, channels=2, type="INV_BOOLEAN"
    Volume: min=0, max=1, dBmin=0, dBmax=0
  Unit: 41
    Control: name="MA Internal Clock Validity", index=0
    Info: id=41, control=2, cmask=0x0, channels=1, type="BOOLEAN"
    Volume: min=0, max=1, dBmin=0, dBmax=0
$ amixer -D hw:MP500 contents
numid=7,iface=CARD,name='MA Internal Clock Validity'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=8,iface=MIXER,name='Mic Capture Switch'
  ; type=BOOLEAN,access=rw------,values=2
  : values=on,on
numid=9,iface=MIXER,name='Mic Capture Switch',index=1
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=10,iface=MIXER,name='Mic Capture Volume'
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=128,step=0
  : values=124,124
  | dBminmax-min=-127.68dB,max=0.00dB
numid=11,iface=MIXER,name='Mic Capture Volume',index=1
  ; type=INTEGER,access=rw---R--,values=1,min=0,max=128,step=0
  : values=124
  | dBminmax-min=-127.68dB,max=0.00dB
numid=3,iface=MIXER,name='MP500  Playback Switch'
  ; type=BOOLEAN,access=rw------,values=2
  : values=on,on
numid=4,iface=MIXER,name='MP500  Playback Switch',index=1
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=5,iface=MIXER,name='MP500  Playback Volume'
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=128,step=0
  : values=128,128
  | dBminmax-min=-127.68dB,max=0.00dB
numid=6,iface=MIXER,name='MP500  Playback Volume',index=1
  ; type=INTEGER,access=rw---R--,values=1,min=0,max=128,step=0
  : values=128
  | dBminmax-min=-127.68dB,max=0.00dB
numid=2,iface=PCM,name='Capture Channel Map'
  ; type=INTEGER,access=r----R--,values=2,min=0,max=36,step=0
  : values=0,0
  | container
    | chmap-fixed=FL,FR
numid=1,iface=PCM,name='Playback Channel Map'
  ; type=INTEGER,access=r----R--,values=2,min=0,max=36,step=0
  : values=0,0
  | container
    | chmap-fixed=FL,FR
```
4.
```
 $ cat /var/pipedal/JackChannelSelection.json
{"inputAudioPorts": [ "system::capture_0","system::capture_1"],"outputAudioPorts": [ "system::playback_0","system::playback_1"],"inputMidiDevices": [{"name": "hw:CARD=MP500,DEV=0","description": "MP500"}]}
```

I also tried every permutation of physical inputs and input/output settings in Pipedal but I do not hear any sound whatever I choose.

rerdavies, 2023-05-20:
Make sure that the pipedald service is running:

     sudo systemctl start pipedald

Connect your guitar to the LEFT gutiar input (2nd from the left, not the leftmost jack). According to the manual, the position are reversed. Select STEREO in the pipedal settings dialog.

Run 

    alsamixer -D hw:MP500

Does it run? What are the default settings? Can you change them? Does that have any effect?

TheMojoMan, 2023-05-20:
**I was finally able to pinpoint the problem - and I know exactly how to reproduce it and how to recover sound!**

**TL;DR** Sound disappears as soon as I press the OK button in pipedal's MIDI settings window when choosing the MIDI device. Many times this is accompanied with error messages like
```
[  time stamp] sched: RT throttling activated

```
or
```
[] retire capture urb: 78 callbacks suppressed
[] xhci-hcd xhci-hcd.3.auto: ERROR transfer event TRB DMA ptr not part of current TD exp_index 2 comp_code 13
[] xhci-hcd xhci-hcd.3.auto: Looking for event DMA 0000...
```
**Pure guesswork:**
 - MIDI initialization code is mistakenly written to audio device and not MIDI device
 - some pointers got mixed up; maybe due to referencing by name of devices which in case of my footswitch/interface is the same for the audio interface and MIDI device, namely 'MP500'

So, could you have a closer look into your MIDI setup code, please?

**Long story:**
I never realized that sound is actually working after installing Pipedal because as a conscientious user I first read the complete documentation of PiPedal (which, btw, is superb) and then I installed Pipedal, opened the webinterface of Pipedal on my iPad, did the setup of the audio interface and finally plugged in my guitar and put on my headphones and then - no sound.
Since I already knew from my previous tests that I was able to play and record sound using ALSA and that I also got sound when connecting input and output of my card using Jack, today, I started testing if the TooB .lv2 plugins were OK. And, indeed they worked as expected when starting them using jalv and doing the connections with Jack (all in terminal because I am using a server image with no desktop environment; I tried installing one inbetween but all the 'armbian-bullseye-desktop-....deb' packages all had errors so I first gave up on it).

After many, many failed attempts it lead me to starting all over again i.e. burn Armbian to SD card again, doing the initial setup of it, downloading and installing Pipedal. But this time I still had my box connected to a monitor and I had my headphones already on when installing Pipedal. And after installation, fixing a port problem with dnsmasq and starting the pipedal.service - voila - I heard my guitar. Setup of the sound card in Pipadal was still OK but when I hit the OK button for MIDI setup sound suddenly disappeared and at the same time an error message appeared in the terminal. I then returned to the MIDI setup, deselected the 'MP500' and pressed OK - but sound was not recovered. Restarting pipedal.service or rebooting did not change anything. Pipedal and the audio interface were stuck in a strange non functional state.

So, I had a closer look into where the MIDI setup data is saved. And, indeed, by doing the following I was able to recover sound:
```
$ sudo pipedalconfig --stop
$ sudo rm /var/pipedal/JackChannelSelection.json
$ sudo pipedalconfig --restart
```
I repeated everything several times and it always works. Also, when I reboot when MIDI device is not set I always get sound after rebooting. But as soon as I set the MIDI device again and sound disappears rebooting does not help. There is no sound after reboot.
I repeated the above steps again several times to be sure that I really found the bug and suddenly - to my surprise - I could press the MIDI setup OK button and sound did not disappear. Therefore, I repeated the recovery and testing procedure a bit more and as a conclusion it seems that once in a while MIDI device is set correctly but most of the times sound disappears which led me to the guesswork I described above.

My personal resume:
It took me so long to figure out the problem because I repeated the same mistake over and over again i.e. I first did the setup and then started playing guitar and listening. I was stuck in a rut and only when I changed my behavior and did it differently I realized that initially the sound is working which led me to further steps. I remember somebody saying 'Change is possible!' and I have to add '... and sometimes it is sorely needed.'.

rerdavies, 2023-05-20:
Very nice catch! Thank you. No reason why you should have expected the two things to be related. I certainly didn't. 

So. When I try to repro on my MOTU M2, which also has MIDI input and output; both audio and MIDI inputs work fine for me. We are close, but not yet home free.

Your MIDI device address -- as far as I can tell -- looks correct.  It was present in earlier files you posted. The device address for me: hw:CARD=M2,DEV=0  and yours looks comparable. So I don't think device address is the issue.

Things to do:

1. Would you mind trying to reproduce the issue using jack? Connect audio inputs to outputs; and then connect an MP-500 MIDI input to the MIDI output (making sure you don't have a MIDI cable that's looped back, of course). PiPedal is using ALSA raw midi, if that makes any difference; I think Jack defaults to doing the same.

2. And if you could provide me with a complete snippet of the system log/journalctl messages that occur after selecting a MIDI device, that might help me narrow the problem down. You can email the snippet to me at rerdavies@gmail.com, if you would prefer to not disclose system log fragments in a public forum.

I agree with your assessment that low level errors like that might have something to do with a wild write, or things that are even worse. So the error messages would be more than helpful, especially if I have to dive into kernel code to sort this out.

In the meantime, I will review the MIDI initialization code to see if I can gain any insight.

rerdavies, 2023-05-20:
Something else to try:  USB 2.0 ports vs USB 3.0 ports (Blue shells on the SBC).  Whatever you're using now, try the other one.

The fragments of systemlogs you have provided are... not a happy thing. I can make some wild guesses as to what PiPedal might be doing that breaks things (something to do with timing around the opening of the MIDI channel and audio channel) . But I really do need a set of systemlog messages, starting from just before you select the MIDI device, with accurate timestamps. Timestamps are going to be important, I think.

rerdavies, 2023-05-21:
Thank you for the logs. 

I've reviewed the code as carefully as I can. The MIDI code, in particular, is pretty straightforward. If there's a problem or a weakness in there, I can't see it immediately. It looks ok, as far as I can tell. There's nothing dangerous or risky in the MIDI code, at least, which is not to say that there isn't a wild write elsewhere, I suppose.

I've reviewed the logs. Unfortunately, they were less helpful than I had hoped. There is a single error message, which is discussed below at some length.

**Question:**  If the audio thread is running, you should see slight fluctuations in the reported CPU use and temperature indicator in the PiPedal Status message. If the values don't change at all, that's significant. Could you check for me that the values are changing. 

**Question:** When the audio is running, what kind of CPU use is being reported by PiPedal Status messages on the default preset? Are you seeing a large numbers of overruns? It's not uncommon to see a few overruns (less than 10) when booting, or restarting. But hundreds or thousands of overruns would cause no audio to be heard.

** Question:** How hard would it be for you to get your hands on either an Armbian "SMP PREEMPT" or Armbian "SMP PREEMPT RT" kernel? There's currently confusion as to what a "realtime" kernel is right now. It can refer to either "SMP PREEMPT" or "SMP PREEMPT RT" kernels depending on the distro; and many install an SMP PREEMPT kernel out of the box. Either will do. If you have access to an "Armbian realtime kernel", either way, that's what you need. I'm not sure whether you're building your own Armbian kernel. If you are, there's a menu item in the configuration utility that allows you to select what type of kernel to build. I couldn't say whether "SMP PREEMPT RT" is still an option; but I would be highly surprised if "SMP PREEMPT" is not an option. There's further discussion and context later in this message discussing _what_ and _why_.

**Question:** Is ARM Ubuntu an option for you? Ubuntu has a readily available SMP PREEMPT kernel (either stock, or installable via 'apt install linux-realtime', or bundled with Ubuntu Studio --  I'm uncertain which); and is -- as far as I know -- compatible with the PiPedal .deb package, and PiPedal build instructions.

** Something to try:** As far as jack is concerned, there is a substantial and real difference between "hw:MP500", and "hw:MP500,0" (or "hw:CARD=MP500,DEV=0", which I suspect is the same thing). The former leaves the ALSA midi ports alone; the latter seems to create Jack Midi ports from the ALA Midi ports, which will then show up when you use "jack_lsp". That being said, it doesn't really matter. What I'd like to know is whether the ALSA audio ports that jack is using will still work when the ALSA midi ports are connected. It's pretty trivial to connect the ALSA midi ports using ALSA utilitiles. 

-  Start jack, connect the audio ports, as you've been doing.
- Connect the ALSA midi ports. 
       aconnect -l   # lists the available ALSA ports. Use this to get the number of the MIDI port.
       aconnect <n> <n>  # where N is the numeric value for the midi port given by `acconnect -l`
- The critical question: does audio still work after that?

For context, qjackctl tracks connections made by aconnect. And all I'm worried about is the effect of connecting the ALSA midi ports.

**Something to try:**  There's a vague possibility that I've mis-specified the audio address. I'm struggling to figure out what the difference between hw:MP500, hw:MP500,0, and hw:CARD=MP500,DEV=0. If you could do the following for me, I'd be grateful. The configuration for the audio device is stored in `/var/pipedal/AudioConfig.json`. 

- Do what you do to get audio running, but with no selected midi device. 
- Edit the file `/var/pipedal/AudioConfig.json`, and change 
         "alsaDevice": "hw:MP500",
    to 
         "alsaDevice": "hw:MP500,0",
    (add ",0" after the device name; I'd prefer this variant to the CARD=, DEV= syntax)).
- restart PiPdeal, or reboot. Either will do. Is audio still working?
- Select the MIDI port in the Pipedal Settings dialog, without touching the audio configuration at all.
- Does that kill audio again?

The rest of this message is to background on where I am right now, that you may or may not find useful.

I have worked my way through the log files.  I can't find any examples of the  "xhci-hcd" errors you mentioned earlier. I think they're probably a USB3 thing, which we're not seeing because you currently have your device connected to a UBS2 port. That's a relief, honestly. Kernel messages complaining about DMA pointers is not something anyone really wants to deal with.

The one error I do see, which is pretty consistent is:

    "<time> armbian kernel: sched: RT throttling activated   

which seems to occur immediately after you select a MIDI device (but not otherwise).

I would have to interpret that message as "The PiPedal audio thread is using much too much CPU", or perhaps that the audio thread has locked up completely  It's a curious error, since PiPedal/ALSA isn't reporting any audio underruns, and there's no indication that the audio thread isn't running. There should at least be a PiPedal message if the audio thread has terminated.  PiPedal does poll for MIDI messages on the audio thread, so there is additional CPU use on the audio thread when a MIDI device is selected. But overrunning the RT scheduler would be a pretty big [... cut ...]
What you want: an Armbian kernel that is either SMP PREEMPT, or SMP PREEMPT RT. If you are building your own kernel, the armbian kernel configuration utility allegedly provides an out-of-the-box selectable menu option for "kernel variant". I couldn't say whether that is SMP PREEMPT, or SMP PREEMPT RT. Either would be fine..

Is that a realistic ask? Let me know.

Thanks for your ongoing assistance. If we can sort this out, we'll be sorting it out for hundreds of other people as well, who would otherwise encountered the same problem. So I am grateful.

TheMojoMan, 2023-05-21:
First, let me please say that your generosity in spending time on this particular issue is second to none. The community and I can consider us very fortunate that experts like you are willing to share their love and passion for solving intricate problems by giving us such great free software like PiPedal and TooB Amp and even offer detailed support. I am overwhelmed and speechless. You are such a kind soul!

I want to first explain what else I found out yesterday before i went to bed:
It seems that the Armbian 6.1.27 kernel that I am using is not configured for using MIDI with Jack. It is missing snd_seq module which was the cause for me not being able to use a2j midi bridge and therefore to connect MIDI using Jack. See logs below. I found a thread in the Hardkernel/Odroid forum from 2019 discussing the same issue and they eventually fixed the kernel to solve the problem.

```
armbian:~:% lsmod | grep snd
snd_usb_audio         315392  0
snd_hwdep              20480  1 snd_usb_audio
snd_usbmidi_lib        36864  1 snd_usb_audio
snd_rawmidi            45056  1 snd_usbmidi_lib
snd_soc_meson_axg_sound_card    16384  0
snd_soc_meson_g12a_tohdmitx    20480  1
snd_soc_meson_card_utils    16384  1 snd_soc_meson_axg_sound_card
snd_soc_meson_axg_tdmout    24576  1
snd_soc_meson_axg_frddr    36864  3
snd_soc_meson_codec_glue    16384  1 snd_soc_meson_g12a_tohdmitx
snd_soc_meson_axg_tdm_interface    16384  2 snd_soc_meson_axg_sound_card
snd_soc_meson_axg_fifo    20480  1 snd_soc_meson_axg_frddr
snd_soc_meson_axg_tdm_formatter    16384  2 snd_soc_meson_axg_tdm_interface,snd_soc_meson_axg_tdmout
armbian:~:% amidi -l
Dir Device    Name
IO  hw:1,0,0  MP500 MIDI 1
armbian:~:% aconnect -i
ALSA lib seq_hw.c:466:(snd_seq_hw_open) open /dev/snd/seq failed: No such file or directory
can't open sequencer
armbian:~:% aconnect -o
ALSA lib seq_hw.c:466:(snd_seq_hw_open) open /dev/snd/seq failed: No such file or directory
can't open sequencer
armbian:~:% cat /proc/asound/MP500/midi0
MP500
Output 0
  Tx bytes     : 0
Input 0
  Rx bytes     : 0
```
Therefore, before spending more time on testing stuff using this Armbian Bullseye kernel I would suggest that I first have a try using the Armbian Ubuntu Jammy kernel that ophub provides on github.

As far as I would like to compile my own kernel I unfortunately have zero knowledge in how to do it. I am just a hobby enthusiast with some basic knowledge of C/C++, Python and Swift who can only fix the simplest of problems by his own. I am also new to linux audio and started learning a little more about it in the last couple of days.
I think that I would first have to setup a cross compiling environment on my best machine which currently is an M1 Mac Mini. I remember that I indeed once compiled a kernel that the youtuber Jeff Gerling (who is very active experimenting with RaspberryPi; I guess that you might know him) provided and it took my Mac about 7 min. to compile a RaspberryPi kernel which I consider a great result.
But he was using a docker image and I am not sure if I will be able to adjust his setup to my needs. I still need to learn a lot more about it all.

A quick answer to your first two questions:
At Rate: 48000Hz, Buffer size: 256, Buffers: 2
 - Jazz preset: XRuns: 0, fluctuating CPU: 12.9% - 13,7%, temperature is constantly increasing: 39° ... 40° ... 41° ...
 - Heavy preset: XRuns: 0, fluctuating CPU: 18% - 19%, 42° ... 43°...

So, it should be more than enough CPU headroom to do more stuff on the audio thread, I think.

I hope that I can find a little more time for further experiments later this day (it's early in the morning now where I live) and I will report back later today and/or in the following days.

rerdavies, 2023-05-21:
Again, another excellent catch. Seems like you've actually done all the hard work here. :-) Your perseverance is greatly appreciated.

The CPU results look great. It's going to work well on your device.

I'm not experienced enough in Linux to not be horrified by results like this. Not so much as a decent error result or log message anywhere! Very strange.  But I too have certainly learned interesting things about the Linux audio stack during our misadventure, that will have to be filed away for future reference. I imagine, at some point or another, some PiPedal user is going to get bitten by ALSA volume controls. 

Incidentally, recompiling a kernel: It's actually not that bad. Kind of intimidating, but not that bad.  I've only done it once as well. You got more than enough memory to compile a kernel; its' going to take longer than on an M1 obviously, but certainly not more than an hour or two. Compiling it on the SBC would be much easier than a cross-compile.  Compiling a PREEMPT_RT kernel, on the other hand, is not so easy. 

I'm going to close this issue for now. If it comes up again on your next kernel candidate, feel free to re-open this issue.

rerdavies, 2023-05-21:
Closed. Misconfigured kernel.

closed 2023-05-21 by rerdavies

TheMojoMan, 2023-05-21:
It was an enjoyable, educational and very inspiring conversation. Thank you so much!
I'll definitely try to stay on the ball and maybe I'll be able to compile my own kernel soon. To be honest, I prefer the 7 minutes to the two hours and try to see how it works with cross-compiling.