rerdavies/pipedal #123: [BUG] PiPedal app crashes while trying wifi direct connection

redcloud80, 2023-07-01:
**Describe the bug**
PiPedal app crashes while trying wifi direct connection on Android.

**Server (please complete the following information):**
 - OS: Raspberry Pi OS LITE (64-bit)
 - PiPedal Version 1.1.27

**Client (please complete the following information):**
 - Device: Android Phone
 - OS: Android 10
 - PiPedal app Version 1.1 21 Apr 2023

rerdavies, 2023-07-01:
If you get asked to submit a crash report, please do so. I'll keep an eye out for it on the Google Play developer console.

Supplementary information, please:

- Are you using Raspberry Pi 4?
- Run the following command, at a bash prompt, and report back with the results:
```
   uname -a
```
- Reboot your Raspberry PI; attempt to connect with the Pipedal app. Run the following command at a bash prompt, and report back with the results:
```
    sudo  journalctl -b 0 | grep pipedald 
```

rerdavies, 2023-07-05:
fyi, if you have switched to the new network stack in Rasberry Pi OS, Wi-Fi direct will probably not work. It hasn't been tested, and I honestly don't expect it to work. Officially, not supported; but I need an excuse to fix it because it will need to be fixed eventually.

If you are using a Pi 3, there is a significant chance that you can't have both a Wi-Fi direct connection and a regular Wi-Fi connection at the same time. You may be able to get it to work by 

(1) not enabling Wi-Fi Direct. You don't need to enable Wi-Fi direct if both devices are on the same Wi-Fi network.

(2) Turning off regular Wi-Fi. The app should attempt to make a Wi-Fi direct connection as long as Wi-Fi direct is enabled on the Pi 3.

rerdavies, 2023-09-05:
Fixed in Pipedal Client 1.0.5.

closed 2023-09-05 by rerdavies