rerdavies/pipedal #124: [BUG] pipedalconfig error: Invalid option: --enable-legacy-ap

redcloud80, 2023-07-01:
**Describe the bug**
pi@raspberrypi:~ $ pipedalconfig --enable-legacy-ap
Error: Invalid option: --enable-legacy-ap

**Server (please complete the following information):**
 - OS: Raspberry Pi OS LITE (64-bit)
 - PiPedal Version 1.1.27

rerdavies, 2023-07-01:
Please try '--enable-ap' instead.  The help text was changed; but the actual option was not.

Please be forewarned that it has been a while since this feature has been tested, and that the feature is not currently supported. 

The option name will be corrected in the next release of PiPedal.

rerdavies, 2023-07-01:
Also, this probably won't do what you want it to do. The Android PiPedal app does not support legacy wi-fi connections. 

You may want to try the following: using Raspberry Pi tools, connect your Raspberry pi to the same WiFi router that the the Android PiPedal app is connected to.  Not as good as Wi-Fi direct access, because you would need a Wi-Fi router if you're performing on stage; but it works fine for home use.

rerdavies, 2023-07-05:
Fixed in 1.1.28. Specifically, the command-line option has been renamed to agree with the --help text. It is still probably not something that's useful to you.

closed 2023-07-05 by rerdavies