rerdavies/pipedal #128: [BUG] Description

EE1977, 2023-07-11:
Raspi 4B+ 8Gb RAM

Version 1.1.28 Dark mode

Toob Convolution reverb. Although I can upload impulses, the **impulse name**** does not display properly over the impulse file frame.
It is stuck over St. Margaret Church, whatever IP I upload. I can clearly hear the difference on sound while loading different impulses, but the Impulse file name is always St.Margaret's Church...

EE1977, 2023-07-11:
Also tried the Toob Convolution stereo version, and noticed excessive cpu usage and extreme unresponsiveness from pi
[image]

rerdavies, 2023-07-11:
Fixed in version 1.1.29, just posted.

closed 2023-07-31 by rerdavies