rerdavies/pipedal #137: PiPedal now fails to install

Sedaxel, 2023-10-14:
I've como across a failure while installing PiPedal. It's maybe not a bug but a particular error for my Pi, so if that was the case, just please ignore me and delete this report.

I've installed pipedal 1.31 several times while testing (maybe more than 5 times) without problems.

Sometimes it was for testing different OSs (Ubuntu, RaspiOS...). Other times because the plugin stopped working anymore and I made fresh installs.

I was researching an error with Xruns while the web server is working either on the PC or my mobile phone. Then stopped working at all... So flashed again.

But now I can't even install PiPedal. It throws an error related to dependencies (libflac++6v5).

[image]

Has anything changed lately that could cause this error? I tried both in Raspberry Pi OS Lite 64 and Ubuntu server 64. I've got a Pi 4 (4 Gb) with an Hifiberry DAC+ADC.

My web browser is Brave, both in the pc and the mobile.

Thank you!

rerdavies, 2023-10-14:
The libflac++6v5 dependency has been present for quite some time (at least
since May, probably much earlier than that). But I would guess that it
indicates that your OS is too old, rather than too new.

It would be helpful if you could probe apt to see what the the closest
available libflac++ library is on the distributes involved.

    apt search libflac++

Could you please provide a more detailed description of the platforms
involved. Post the output of:

    uname -a

That provides more helpful information than "raspberry pi lite", or "ubuntu
server" (which doesn't given information about the OS version)..

I'm going to assume that the problem with held packages doesn't affect any
of pipedal's dependencies. But you might want to check that as well. It
might be a good idea to force the held updates.

Sedaxel, 2023-10-14:
Thank you for your reply!

Yeah, sorry. Both OS went flashed from Raspy Imager just before installing. And I grabbed the newest ones from the list... Just the same I always did when trying this project. They were fresh flashed disks for each installation. No risk of having an old OS.

And the PiPedal was just downloaded yesterday too. First I tried the file I had in my pc (which worked all this time) and, when this other problem happened, I downloaded again... Just in case.

I'm not at home now, can't make the uname in this moment. I'll try to do.

The weird thing is that... It could seem that the latest OSs lack some files needed for Pipedal? Just especulating, I've got no idea. If you can install it without problems, then obviously should be something in my Pi that stopped working somehow.

rerdavies, 2023-10-14:
Raspberry Pi OS has just dropped a major OS version bump for Pi4 and Pi5.
Which is why I am particularly curious about the OS version. I'm also
curious whether you are on the LTS or tip version of Ubuntu server. So yes,
providing the uname -a info would be useful.

It is possible that the flac++ library has aged out. But I'd like to
confirm that before working on a fix for it. And to establish what versions
are available.

I'll wait for your reply.

Sedaxel, 2023-10-15:
Hi there!

Here is the results of the libflac and uname:

[image]

rerdavies, 2023-10-16:
Perfect, thank you.  That tells me all I need to know.

In retrospect, I guess it makes sense that the library should have been
statically linked instead of dynamically linked. Yet another linux lesson
learned. :-/

Unfortunately, you've caught me at an awkward moment -- I'm just completing
final testing and merging of major 1.2 release of the ToobAmp components.
It contains the fix and it appears to work.. If you could bear with me for
a day or two, I should be able to get the fix published in PiPedal.

I'll let you know when the fix is available..

Regards,

Robin Davies.

Sedaxel, 2023-10-16:
Hey, thank you!

How wasn´t I to bear with you for a day or two? You deserve all the time in the world. Even if you ever published a fix, it would have been worth it to have tried your fantastic project. It´s a real joy, and I couldn´t be more grateful.

Awesome job, mate.

bitmat1, 2023-10-26:
I had the same problem after a fresh install of Pi os 64 bit (bookworm)
After failing to install the dependencies, I reflashed the sd card with Pi OS Legacy, 64bit, (bullseye with desktop) and now pipedal installs and works perfectly. Thanks for this excellent software, hope this helps.

Sedaxel, 2023-10-26:
Thank you for the suggestion!

rerdavies, 2023-10-27:
Confirmed. It's an issue with bookworm. A fix is in progress. Just working through the details of how to get a build that works for bullseye and bookworm. :-/ In the meantime, using Pi OS Legacy does provide a workaround.

Smith8154, 2023-12-13:
Just tried installing as well and getting the same error. I will move to Pi OS Legacy for now, but wanted to follow up and see if this was still in progress?

remixmabix, 2024-03-16:
Hello any news for this ?

FoolHen, 2024-05-15:
I compiled pipedal for bookworm, check https://github.com/rerdavies/pipedal/issues/140#issuecomment-2111850147

rerdavies, 2024-08-08:
Fixed in v1.2.33. 

The current version statically links against LIBFLAC++10, which entirely removes the install dependency moving foward.

Previous versions statically linked against LIBFLAC++6.5, except for one llibogg, for which, due to a bug in the release was not made available as a static library.

closed 2024-08-08 by rerdavies