rerdavies/pipedal #140: Pi5 support

EE1977, 2023-12-31:
Hello there!! That would be awesome if there would be an update to make possible an installation on a Pi5. As others mention, pipedal refuses to work on Debian 12 (bookworm). The pipedal project is one of the coolest projects on a raspi and, hopefully, this is not an abandoned project!!

maarthome, 2023-12-31:
Pipedal on PI5 would be awesome indeed!

Best wishes to you all!

FoolHen, 2024-05-15:
@maarthome @EE1977 I've compiled ToobAmp and pipedal for bookworm. Works fine in a raspberry pi 4 with bookworm, should be working on pi 5 too.

[toobamp_1.1.30_arm64.zip](https://github.com/rerdavies/pipedal/files/15318296/toobamp_1.1.30_arm64.zip)
[pipedal_1.1.31_arm64.zip](https://github.com/rerdavies/pipedal/files/15318298/pipedal_1.1.31_arm64.zip)

You only need to install pipedal, as it has toobamp included already. I added ToobAmp .deb file in case anyone needs just that.

EE1977, 2024-05-15:
Thats awesome!!! I can't wait to give it a try on my pi5!!!

maarthome, 2024-05-15:
Really GREAT!! Wow.... thanks!!

Did anyone of you hear anything from the creator? .... Robin Davies

Greetings,
Maarten

EE1977, 2024-05-15:
Great news!!! I can confirm it works like a charm on the pi5!!!! Thank you very very much!!!!  @FoolHen as you seem to be experienced enough with the project, do you think you could help with debugging some other issues as well? Robin seems not to be available for some time now and the project here is a real gem!!

> Really GREAT!! Wow.... thanks!!
> 
> Did anyone of you hear anything from the creator? .... Robin Davies
> 
> Greetings, Maarten

No news from Robin...

FoolHen, 2024-05-15:
Glad it worked 😄 sadly I have not heard from the creator, which is a shame because my knowledge in this topic is really basic. I would like to help with the development but I'm afraid I can't do much at the moment. I will still try tho, might be able to do some stuff.

Hopefully others with more experience in c++ and audio are interested to contributing, so together we might be able to continue this project. Because I also agree that it's a really good idea.

EE1977, 2024-05-15:
I really hope this project continues because it's a fantastic solution for live sound. As for the major issue I face... I really don't know how complicated might this be for a solution to be found. Anyway, I've encountered some issues with MIDI bindings on the FX bypass switches. Even though I can easily toggle an FX on or off using my touchscreen or mouse, assigning it to my MIDI controller doesn't seem to bypass the audio FX, despite changing the status of the bypass switch on the interface.

FoolHen, 2024-05-15:
Hmm I haven't dived into the logic just yet. I haven't encountered that issue although I plan to use MIDI bindings too, just haven't got around it yet. I would suggest you open a new issue if you haven't already, so we can keep track of it 😄

FoolHen, 2024-05-17:
Be advised, I've noticed that WiFi direct does not work in bookworm. Seems like the way the network connection settings are stored has changed. Will try to fix it

rerdavies, 2024-08-08:
Fixed! The newly released v1.2.33 works on bookworm. 

Wi-Fi direct should also work on bookworm. 

Support for Wi-Fi direct on the NetworkManager stack is badly broken, but after a lot of heavy lifting (four month's worth rummaging through NetworkManager and wpa_supplicant source code), I managed to get it to work. Nightmare.! :-(

closed 2024-08-08 by rerdavies