rerdavies/pipedal #146: MIDI bindings cannot bypass effects

EE1977, 2024-05-15:
I've encountered some issues with MIDI bindings on the FX bypass switches. Even though I can easily toggle an FX (ON or OFF) using my touchscreen or my mouse, assigning it to my MIDI controller doesn't seem to bypass the audio FX, despite changing the status of the bypass switch on the interface.

maarthome, 2024-05-16:
Hi EE1977,

perhaps you should state that this issue is related to FoolHens compilation on Bookworm OS? (or isn't it?)

Cheers,
Maarten

EE1977, 2024-05-16:
Hey there! Just wanted to let you know that the issue is still there, even with the previous (Bullseye) version.

EE1977, 2024-05-16:
It is definitely not a @foolhens issue.

FoolHen, 2024-07-16:
@EE1977 @maarthome  Hi. I have forked this repository and fixed some small issues, including this one, and added 150 presets to ToobML, some with knob control. https://github.com/FoolHen/pipedal

I created a release for bookworm with the changes, let me know how it goes if you try it. https://github.com/FoolHen/pipedal/releases/tag/v1.1.32

rerdavies, 2024-08-08:
Fixed in v1.2.33

closed 2024-08-08 by rerdavies