rerdavies/pipedal #149: Is it possible to load the models that Foolhen made available in his version of pipedal?

maarthome, 2024-08-08:
Hi Robin,

Great to have a update on pipedal!!

Is it possible to load the models that Foolhen made available in his version of pipedal? Or can you tel me how to do this?

Aslo... would it be possible to have two channels? I mean one channel for guitar; left input via plugins to stereo ouput and right input via plugins to stereo output. (The in and outputs are from my scarlett 2i2)

Thanks for being active again!!
Maarten

PS is there a flowchart available for the pipedal software?

rerdavies, 2024-08-09:
> Is it possible to load the models that Foolhen made available in his version of pipedal? Or can you tel me how to do this?

I need to talk to Foolhen about this. I haven't seen exactly what he's done.  I still have one more of his fixes to merge (relating to disabling of the gain knob in the Neural Amp modeller). And I'll take a look at how he's handled the additional NAM models and presets next.

One of my concerns is that I've been very careful to maintain proper copyright discipline with pipedal. The few media files that are included with pipedal have explicitly copyright grants given by the authors.. And ToneHunt makes it virtually impossible to determine the licensing of .NAM files,  or contact the authors of the .NAM files to obtain permissions to incorporate them into the PiPedal installer. Which is why there are currently no NAM models distributed with the Neural Amp Modeller. 

It's also a current deficiency in PiPedal's bank/preset files. When presets were first implemented, Pipedal didn't have the concept of uploaded audio and .nam files. Properly implemented, the media files should be included in the preset files. (On the short list of features to implement right now). Once that's done, it should be possible to export all of FoolHen's patches to a bank file, and upload them en-mass to a new machine. Very high on the priority list of features to implement right now.

In the meantime, with existing features, it's an awkward process. 

My version of pipedal is takes great care to not overwrite user existing presets or banks when installing. so you might be able to install my Pipedal over an install FoolHen's install (depending what he's done).

Expect a better response shortly.

> Aslo... would it be possible to have two channels? I mean one channel for guitar; left input via plugins to stereo ouput and right input via plugins to stereo output. (The in and outputs are from my scarlett 2i2). 

It is possible. Insert a splitter at the start of the preset, and set it to LR mode. This routes your left channel over the top route, and your right channel over the bottom route. The configure the splitter to pan left and right channels hard left and hard right (respectively). 

> PS is there a flowchart available for the pipedal software?

I'm not quite sure what you mean. There is an architectural diagram, which lays out the major components of the pipedal software, and how they interact.

    https://rerdavies.github.io/pipedal/Architecture.html

Just vague enough to not be dangerous.  But not vague enough to be ever so slightly out of date anyway.  Major omissions: The LV2Host component which manages Lv2 plugins; and  pipedal_nm_p2p2 which is the service that handles Wi-Fi direct connections. if that's not what you're looking for, ask again, and I'll see what I can do.

maarthome, 2024-08-09:
Hi Robin,

Thanks for your elaborate reply :-)
I will more carefully read your documentation.... who knows whether I can contribute in tge future.

Your suggestion for splitting channels:
"It is possible. Insert a splitter at the start of the preset, and set it to LR mode. This routes your left channel over the top route, and your right channel over the bottom route. The configure the splitter to pan left and right channels hard left and hard right (respectively)."

Works only 'half'.... the output is separated into left and right... :-( 
I mean when input left is guitar and right is voice...  I can apply separate plugins via a L/R split but unfortunately in the output signal the guitar is only at the right and the voice at the left channel... :-(
I would like to have the stereo guitar and stereo voive mixed in one stereo output signal.

Any other suggestions?

Cheers,

Maarten

EE1977, 2024-08-09:
Hey Robin!!! Glad to see that you are back and ready to take pipedal to the next level!!! As for the stereo setup, it is something that I'd also really like to see in a more straightforward way (e.g. maybe with a separate FX string for each channel). Also, mind that I had an issue trying to assign MIDI pots to the splitter as well as the input & output gain (issue #134).

rerdavies, 2024-08-12:
After investigating, the solution seems to be to allow uploading of Proteus patch zip files directly. Requires some reworking... a dialog instead of a dropdown.  And the addition of folders to the selection dialog.

rerdavies, 2024-08-18:
You can now upload Proteus .zip files and .json files to TooB ML.

Fixed in v1.2.37.

closed 2024-08-18 by rerdavies