rerdavies/pipedal #150: Vertical file lists

rerdavies, 2024-08-09:
### Discussed in https://github.com/rerdavies/pipedal/discussions/138

<div type='discussions-op-text'>

<sup>Originally posted by **38github** December  9, 2023</sup>
The way it is now I have to rename my files to short names because of the vertical file listings when opening files that I have uploaded in plugins.

Wouldn't vertical listing be better where one file per row is shown which should allow more of a filename to be shown?</div>

maarthome, 2024-08-09:
Hi Robin,

Showing filenames is fine for me!! In this way the user (read 'I' :-)) can decide herself what names are shown :-)
(I noticed by the way that the [P].... proteus model files are not loaded by ToobML)

Cheers,

Maarten

rerdavies, 2024-08-09:
@maarthome: Looking at it right now.  I need to talk to Fulghen to see how he wants to handle it.  So many! Are they good?

maarthome, 2024-08-09:
Great!
duimen draaien intussen.... :-)

Cheers

rerdavies, 2024-08-18:
File property directories have been completely redesigned to allow for use of sub-directories.

File names are now displayed full-width.

Fixed in v1.2.37.

closed 2024-08-18 by rerdavies