rerdavies/pipedal #151: Uploading files in plugins, like uploading IR files in   , fails

maarthome, 2024-08-09:
**Describe the bug**
 Uploading files in plugins, like uploading IR files in   , fails.

(BTW: It would be nice to have these files... IR and models in a **central** location. Now a lot of redundancy is created by copying files to the plugin directory....)
 
 

**Screenshots**
**Convolution Reverb gives:**

[image]
[image]
[image]

And in other plugin:
**CabIR plugin gives:**
[image]
 

** Environment (please complete the following information):**
 - OS: [e.g. Raspbian, Ubuntu Gnome3 21.04, Windows 10]
- if Linux, run `uname -a` and include the results.
 - Device that the pipedal server is running on (pi 3, pi 4, &c, how much system memory)
 - PiPedal Version [See X/Settings/About,...]
If UI-related:
 - Browser [e.g. pipedal Android app, safari, Chrome]

rerdavies, 2024-08-09:
I was unable to replicate the issue using Chrome.  And the fact that there are no  CabIR impulse files is troubling, unless you deleted all the factory installed CabIR files.

Could you provide platform details please:

- Browser you're using, or the Android app, if that's the case.
- os on which the PiPedal server is running.
- PiPedal version(See Main menu/Settings/About... in the web app.

Also please provide the output of the following command line command, run on the PiPedal server.

    ls -l  /var/pipedal

FYI, audio uploads are stored in different folders by category.  But they are shared between plugins. I chose to keep CabIR and reverb impulse files in separate isolated folders. Reverb impulse files and CAB Impluse reponse files should generally not be used interchangeably.  I'm open to feedback on this point, but I do think this is the Right Thing To Do.

Currently, 3rd party plugins do place their audio files in a separate category, unless they indicate to PiPedal that they want to use the CabIR or ReverbImpulseFiles (or other) isolated folders using pipedal-specific LV2 extension. By default, audio files do go into a single non-isolated Samples directory.

If you find the upload process tedious, you could manually copy files into the following folders from the command line:

     /var/pipedal/audio_uploads/CabIR/
    /var/pipedal/audio_uploads/ReverbImpulseFiles/

rerdavies, 2024-08-09:
Also, if you could scroll over to see the tail end of the error message in the upload dialog, and let me know what it said,that would be enormously useful. There should be detailed description of why the upload failed at the end.

maarthome, 2024-08-09:
Hi Robin,
I tried in Chrome and Brave browser. There is no difference in behavior.
I was one subdirectory missing. This one: **_/var/pipedal/audio_uploads/CabIR_**
The IR's load OK now in the Toob plugins.
The loading of files is stil NOK. The error messages are not indicating any specifics.
[image]

This is from journalctl -f: (you or in fact I... never know....)
**_Aug 09 15:00:33 pipedal pipedald[686]: WebSocketSession opened. ([::ffff:192.168.1.181]:37071)
Aug 09 15:00:33 pipedal dhcpcd[630]: wlan0: requesting DHCPv6 information
Aug 09 15:00:43 pipedal dhcpcd[630]: wlan0: requesting DHCPv6 information
Aug 09 15:00:49 pipedal pipedald[686]: WebSocketSession closed. [::ffff:192.168.1.181]:37071
Aug 09 15:00:53 pipedal dhcpcd[630]: wlan0: requesting DHCPv6 information_**

The directory listing

root@pipedal:~# ls -l  /var/pipedal
**_total 44
-rw-rw-r-- 1 pipedal_d pipedal_d   197 Aug  9 00:13 AudioConfig.json
drwxr-xr-x 7 root      pipedal_d  4096 Aug  9 14:46 audio_uploads
drwxrwxrwx 2 pipedal_d pipedal_d  4096 Aug  8 21:25 home
-rw-rw-r-- 1 pipedal_d pipedal_d   251 Aug  9 10:45 JackChannelSelection.json
drwxrwxrwx 2 pipedal_d pipedal_d 20480 Aug  9 10:19 plugin_presets
drwxrwxrwx 2 pipedal_d pipedal_d  4096 Aug  9 14:44 presets_**

But I can copy files with sftp using filezilla... :-)

That's it... 

Cheers,

Maarten

rerdavies, 2024-08-09:
The permissions on /var/pipedal/audo_uploads are not correct. Run the following command and let me know if things improve:

     # set the ownership of all files and directories in the audio uploads directory
     sudo chown -R pipedal_d:pipedal_d /var/pipedal/audio_uploads
     # files and folds subsequently created in these directories will have the group set to pipedald.
     sudo chmod 2775 /var/pipedal/
     sudo chmod 2775 /var/pipedal/audio_uploads
     sudo chmod 2775 /var/pipedal/audio_uploads/*

The permissions on the directories `/var/pipedal` and `/var/pipedal/audio_uploads` (and every other directory in `/var/pipedal` should be 

     drwxrwsr-x pipedal_d pipedal_d

If you want to edit or modify files in those directories yourself, the easiest way to do so is to add your userid to the pipedal_d group.

      sudo usermod -a -G pipedal_d _youruserid_

Let me know what your results are.

I will take as personal actions, the following:

- More specific error message. "Permission denied" would have been nice.
- Review the install code to see what it does about restoring permissions on existing files and folders.

rerdavies, 2024-08-10:
Version 1.2.34 Beta2 addresses this issue. 

- Permissions are forced to correct values on new and existing files and folders in /var/pipedal at install time.
- pipedalconfig has a new -fix-permissions option to re-apply permissions.
- More detailed errors are not reported through the UI (don't want to report filenames in the web UI), but detailed errors are logged in journal logs.

If you want to edit files in /var/pipedal, probably the safest way to do it is to add your userid to the group pipedald. 

  sudo usermod -a -G pipedal_d _youruserid_

You could of course, also do it using `sudo`.

closed 2024-08-10 by rerdavies