rerdavies/pipedal #152: MIDI control works, but I have to press the momentary switch twice before the status of the plugin on/off changes state?

maarthome, 2024-08-09:
**Describe the bug**
MIDI control works (at least switching plugins ON/OFF), but I have to press the momentary switch twice before the status of the plugin on/off changes state....
Apparently the first switch press is memorised... since I can wait some time before pressing again... before it switches the plugin.
SO there is no MIDI signal error or such. Also tested the output of the foot pedal and the MIDI signals are OK.

** Environment (please complete the following information):**
I have a M-VAVE chocolate foot controller.
I checked on my PC via MIDIVIEW whether the MIDI signal is OK... And YES foot switch works OK.

Pipedal OS:pipedal_1.2.33_arm64

Thanks for your time and effort,
Maarten

EE1977, 2024-08-09:
Hi Maarten! Try to change the MIDI settings in pipedal menu. I had the same issue. Switching from momentary to toggle or vice versa solved my issue.

rerdavies, 2024-08-09:
Ha! makes senses. Your midi controller has probably configured THEIR switch to be a toggle. If your midi controller, and pipedal are both configured to treat the control as a toggle control, strange things are going to happen.

EE1977, 2024-08-09:
I think that it is OK as it is. It is pretty easy to get used to the function.

maarthome, 2024-08-09:
Hi EE1977,

(Starwars?)
You (also) made my day!!
Indeed... setting the control to momentary fixed the 'double clicking' issue... :-)

Thanks and enjoy making music,
Maarten

maarthome, 2024-08-09:
Sorry Elen... I see now (in the mail message) it is Elen... :-)

maarthome, 2024-08-09:
But.....
I seem to miss the meaning of 'toggle'.
I thought that each time a same midi cc event (only CC nr and data value no meaning) occurred the status would change.
SO starting an preset in which a AMP is on and pressing the footswitch would result in OFF.
But I noticed that sending a CC with value 127 only switches it on and never off.

So what is the story?
And is it possible to make such a 'toggle'? So a CC nr, data value unimportant would switch the switch to the opposite state...
This is important for me since after changing the preset I do not want to remember whether I have to step on the foot pedal switch once or twice to get the opposite state...

Cheers,

Maarten

rerdavies, 2024-08-09:
I agree. Created a bug for that. Thanks.

rerdavies, 2024-08-10:
After reading more carefully... You actually need to send TWO midi cc messages: one with a value of 127, one with a value of 0.

PiPedal toggles the LV2 control when the MIDI control changes from OFF to ON (not for each message). The idea is that PiPedal is defending itself against the case where a MIDI continuous controller (a rotary dial, pitchbend, maybe a pad control sending pressure information as CCs). Toggling an LV2 control for each CC message in that case would ... not good.

closed 2024-08-12 by maarthome