rerdavies/pipedal #153: After reboot status is "Not configured"

bbischler, 2024-08-09:
Hi there! Thanks so much for your amazing work!

I successfully installed PiPedal on a Pi 5 with the latest Pi OS, and it's running great. However, I noticed that after rebooting, the status of PiPedal shows "Not configured." My audio interface (a Volt 2) is listed and selected in the "Audio devices" field, but the input/output channels are disabled. I have to reselect the audio interface and press "OK" to change the status to "Xrun" and get the guitar sound working.

Is there something I can configure to have PiPedal working directly after a reboot? 

Thanks again, and best regards!

rerdavies, 2024-08-10:
Could you please install v1.2.34 (just posted), and see if it corrects the problem.

I'll wait for your response before closing the bug.

bbischler, 2024-08-11:
I did but the issue is still there.

rerdavies, 2024-08-11:
Could you run the following two commands, and let me know what the output
is:

    ls -l /var/pipedal

and

    journalctl -b 0 | grep pipedald

bbischler, 2024-08-11:
Thanks for your help- Here is the output:
ls -l /var/pipedal:
```
insgesamt 24
-rw-rw-r-- 1 pipedal_d pipedal_d  196 11. Aug 12:08 AudioConfig.json
drwxrwsr-x 3 pipedal_d pipedal_d 4096  9. Aug 15:13 audio_uploads
drwxrwsr-x 3 pipedal_d pipedal_d 4096  9. Aug 14:15 home
-rw-rw-r-- 1 pipedal_d pipedal_d  146  9. Aug 16:08 JackChannelSelection.json
drwxrwsr-x 2 pipedal_d pipedal_d 4096  9. Aug 16:01 plugin_presets
drwxrwsr-x 2 pipedal_d pipedal_d 4096 10. Aug 20:06 presets
```
journalctl -b 0 | grep pipedald:
```
Aug 11 12:09:17 raspberrypi systemd[1]: Starting pipedald.service - PiPedal Web Service...
Aug 11 12:09:18 raspberrypi pipedald[925]: Document root: /etc/pipedal/config Threads: 10
Aug 11 12:09:18 raspberrypi pipedald[925]: Scanning for LV2 Plugins
Aug 11 12:09:19 raspberrypi systemd[1]: Started pipedald.service - PiPedal Web Service.
Aug 11 12:09:19 raspberrypi pipedald[925]: Unable to read ALSA configuration for hw:V2. hw:V2 playback device not found. (No such device).
Aug 11 12:09:19 raspberrypi pipedald[925]: Audio device not configured.
Aug 11 12:09:19 raspberrypi pipedald[925]: Listening on raspberrypi.local:80
Aug 11 12:09:19 raspberrypi pipedald[925]: DNS/SD service announced.
Aug 11 12:11:25 raspberrypi pipedald[925]: Web Socket Address: 127.0.0.1:80
Aug 11 12:11:25 raspberrypi pipedald[925]: WebSocketSession opened. ([::ffff:127.0.0.1]:39914)
Aug 11 12:11:35 raspberrypi pipedald[925]: WebSocketSession closed. [::ffff:127.0.0.1]:39914
```
and after I reselected the usb adapter in the settings menu and it started working:
```
Aug 11 12:09:17 raspberrypi systemd[1]: Starting pipedald.service - PiPedal Web Service...
Aug 11 12:09:18 raspberrypi pipedald[925]: Document root: /etc/pipedal/config Threads: 10
Aug 11 12:09:18 raspberrypi pipedald[925]: Scanning for LV2 Plugins
Aug 11 12:09:19 raspberrypi systemd[1]: Started pipedald.service - PiPedal Web Service.
Aug 11 12:09:19 raspberrypi pipedald[925]: Unable to read ALSA configuration for hw:V2. hw:V2 playback device not found. (No such device).
Aug 11 12:09:19 raspberrypi pipedald[925]: Audio device not configured.
Aug 11 12:09:19 raspberrypi pipedald[925]: Listening on raspberrypi.local:80
Aug 11 12:09:19 raspberrypi pipedald[925]: DNS/SD service announced.
Aug 11 12:11:25 raspberrypi pipedald[925]: Web Socket Address: 127.0.0.1:80
Aug 11 12:11:25 raspberrypi pipedald[925]: WebSocketSession opened. ([::ffff:127.0.0.1]:39914)
Aug 11 12:11:35 raspberrypi pipedald[925]: WebSocketSession closed. [::ffff:127.0.0.1]:39914
Aug 11 12:14:23 raspberrypi pipedald[925]: WebSocketSession opened. ([::ffff:127.0.0.1]:56448)
Aug 11 12:14:29 raspberrypi pipedald[925]: Audio started. ALSA, hw:V2, 32-bit integer little-endian, 48000, 64x2, in: 1/2, out: 2/2
```

bbischler, 2024-08-11:
I have another question: Is it possible to display the full GUI of the LV2 plugins, including the background color and all the fancy UI elements? Right now, I can only see the parameter knobs.

rerdavies, 2024-08-11:
Regrettably, no. Doing so would require either a javascript implementation
of an X11 web server, or a javascript implementation of a VNC client in
order to display the remote UI of the plugin in a web browser. I did look
around to see whether such things exist. But there doesn't seem to be an
implementation of anything like that that hasn't been abandoned, or that
seems to have been remotely successful.

rerdavies, 2024-08-11:
OK. Problem identified. If the selected device isn't available at boot
time, PiPedal is supposed to delay for a while to see if the device is
there but has not yet come online. But it doesn't seem to be doing so.

Out of curiosity, what audio device are you using?

I'll get back to you soon.

bbischler, 2024-08-11:
This is the Audio interface I am using: Universal Audio Volt 2

rerdavies, 2024-08-11:
I have posted PiPedal 1.2.35 experimental which contains a fix that I think
will address your problem.

     https://github.com/rerdavies/pipedal/releases

Could you give it a try, please, and let me know what happens.

If it doesn't work, please run the following command after having rebooted,
and send me the output

   journalctl -b | grep pipedald

bbischler, 2024-08-12:
It works now! Thank you very much

rerdavies, 2024-08-12:
Non-experimentally fixed in v1.1.36 Beta 3.

closed 2024-08-12 by rerdavies