rerdavies/pipedal #155: 1.) Possibility to set colors of On/Off  symbols od plugins. 2.) Scale screen on sndroid app.

maarthome, 2024-08-10:
Hi Robin,

And... two requests...
1)
I would like to clearly see the status of a plugin on the screen. Now ON is highlighted abd OFF is grayed.
Ik would like to choose teh color that I find most clear. (I would choose light blueish for OFF an GreenGreen for ON)

2)
In the android app, that I have in landscape mode in front of me when playing, there is no possiblitity to scale the screen.... Also not with the 'normal' android finger rezize gestures. Would be greet to have the full width used for the pedal/preset chain overview.

[image]

Cheers,

Maarten

rerdavies, 2024-08-10:
Pinch zooming: Probably not, I'm afraid. Nested scrolling context are perilous in Android-style interfaces. The problem is  what do you grab onto in order to scroll the whole page, given that both the pedalboard view and the control view have drag-to-scroll behaviour already. 

Use of red/pink for on/off toggles is already on my radar. "Turns red when it's on?! Whutt?" The obvious solution: greengreen by default. There are complications with the web framework I'm using. It only allows "Primary" and "Secondary" colors, so if I change the "Primary" color to greengreen, I'm not sure what else comes along with it. But it will happen. ("OK" buttons that were red were also cognitively dissonant. Fixed in 1.2.33, I think). "off is gray" is a platform UI convention, I think.

maarthome, 2024-08-10:
has to laugh at cognitive dissonance... It's my life story
Green for ON and something without confusing connotation and yet visible for off.... blue?

Cheers

maarthome, 2024-08-10:
So default Green/Blue for ON/OFF

rerdavies, 2024-08-10:
So Green/Gray for ON/OFF, because Gray for off is an Android UI convention.

maarthome, 2024-08-10:
Jep that would work perfectly...
(Conventions...)

Cheers

rerdavies, 2024-08-18:
The cognitively-dissonant red OK buttons are now gone. As of 1.2.37, dialog buttons are white/gray.

I cannot for the life of me find a green color that isn't ugly. I tried a bunch of different colors without success.

Thinking about it,  a scale control that allows you to specify 75%/100%/125%/150% (similar to Window display scaling) might be doable.  (pinch zoom is definitely not doable).  

As things stand, the feature request is:

- Green toggle switches; not happening. 
- Change A/B and on off controls to use purple primary color.
- perversely, I like that the pink bypass toggle, because it calls dramatic attention to the control. Inclined to leave this one as is.
- research display scale 75%/100%/125%/150% feature.

rerdavies, 2024-08-20:
Plugin controls (toggles, and on/off buttons) now use the purple primary color instead pink/red.  In addition, toggle controls in the various settings dialogs have been changed purple instead of pink/red.

I have consciously chosen NOT to change the color of the bypass toggle. I want the bypass toggle to stand out dramatically so that it is clearly visible. Cognitive dissonance vs. high visibility... I have chose high visibility for that toggle only.

Fwiw, custom color selection was never really a candidate for a feature. But I hope the current fix addresses your quite reasonable objection to the original colors.

closed 2024-08-20 by rerdavies