rerdavies/pipedal #160: Saving presets is not OK in some situations...

maarthome, 2024-08-11:
**Describe the bug**
After renaming a preset and then pressing save brings back a previous previous old name..... (?)
Strange but true....

Cheers,
Maarten

**Screenshots**
[image]
[image]
[image]
[image]
[image]
[image]

** Environment (please complete the following information):**
PiPedal 1.2.34 beta2

Thanks,

Maarten

rerdavies, 2024-08-12:
Should be fixed in 1.2.36 Beta3 

    https://github.com/rerdavies/pipedal/releases/download/v1.2.36/pipedal_1.2.36_arm64.deb

Please reply once you've had a chance to verify the fix.

maarthome, 2024-08-12:
Hi Robin,
This is OK now 😊

Cheers,

Maarten

closed 2024-08-12 by rerdavies