rerdavies/pipedal #162: Filtering does not work in the pluging selection window.

maarthome, 2024-08-11:
**Describe the bug**
Filtering does not work in the pluging selection window.
For instance when I select the Reverb category on the right pulldown window, NOTHING is shown.... (and there are reverb plugins)
It used to work before (I think... but I also used Zynthian and MOD-ui... so I might mix up things...)

** Environment (please complete the following information):**
PiPedal 1.2.34 beta2
 - Browser [e.g. pipedal Android app, safari, Chrome]
Brave browser

Thanks and cheers,

Maarten

rerdavies, 2024-08-12:
Should be fixed in 1.2.36 Beta3 

    https://github.com/rerdavies/pipedal/releases/download/v1.2.36/pipedal_1.2.36_arm64.deb

Please reply when you have had a chance to test the fix.

maarthome, 2024-08-12:
Unfortunately.... the filtering does not work in 1.2.36....

Cheers,
Maarten

rerdavies, 2024-08-12:
Could you do me a favor, and try refreshing your browser. F5 to refresh.
It's possible that you have a running instance in your browser that has
stale plugin data.

maarthome, 2024-08-12:
Hi Robin,

I reloaded the page on Chrome and Brave... rebooted several times...
Filtering does not work... unfortunately...
😢

rerdavies, 2024-08-14:
I am not able to replicate this problem. Two questions.

1) In the filter dropdown in the Select Plugin dialog, what do you see? Do you see any filters?

2) Could you send me (attach it to your message) the contents of the following file:

     /etc/pipedal/config/plugin_classes.json

Awaiting your response.

maarthome, 2024-08-15:
Hi Robin,
I can select a sub category....
[image]
But no plugins are shown.

And the file you requested....
[plugin_classes.json](https://github.com/user-attachments/files/16624010/plugin_classes.json)

Cheers,
Maarten

rerdavies, 2024-08-18:
Aha! Could you please try clearing your browser cache, and see if you get
different results (files only, no need to clear cookies or other cached
data).

rerdavies, 2024-08-19:
@maarthome:

I have a theory that your problems are being caused by a particular plugin that you have installed.

I have added diagnostics to v1.2.39 to catch problems caused by plugins with incorrect plugin type declarations.

could you please install v1.2.39 experimental, and let me know what happens when you try to filter plugins.

maarthome, 2024-08-20:
Hi Robin,
I installed your experimental version but no difference. You did not specify what I should look for or whether I should send you some diagnostics file.....
When I install version 1.2.38 on a blank SD card, so a fresh install, the filtering works OK.
What is your theory? 

Cheers,
Maarten

rerdavies, 2024-08-20:
There was additional code around the filtering and sorting of plugins to catch unexpected exceptions. You would have seen a message box if there was a problem with the sorting and filtering code. I also had a theory that the locale you are using was causing problems when doing locale-specific sorting of the plugin entries, but the new instrumentation should have caught that as well. 

To be honest, I'm struggling to come up with a theory at this point.  Best theories are still:

- something to do with a plugin you have installed.
- something to do with the locale you are using.

To which I would add:

- a stale file left over from a previous install that's not getting upgraded properly, which could only be the `plugin_classes.json` file you sent me earlier, I think.

However, the instrumentation I added covered most of the code involved in filtering and sorting. It's not actually a lot of code, because displaying without filters executes almost all the same code.

Things for you to try, please:

1. Run the following command:

        sudo rm -rf /etc/pipedal

    Be careful to type that correctly! And then install pipedal again. All of your user data (presets and banks) are stored elsewhere, so this should be 
    harmless.  I can't think of why that should work, but if a clean install worked for you, this would be a good way to do a mostly 
   clean install on your non-working instance.

2. Filter for Chorus or Modulation effects. Does that work? (The amp filters are handled differently from other filters)

3.  Could you also let me know what locale you are using. Perhaps I can reproduce the issue here if I switch to the locale you are using. 

Your assistance and patience in hunting this down is much appreciated.  -R

In the meantime, I'm going to review the remaining code that I haven't covered so far to see if I can spot anything risky.

maarthome, 2024-08-20:
Hi Robin,

 sudo rm -rf /etc/pipedal -> still no filtering

during each install I always get the message:
N: Download is performed unsandboxed as root as file '/home/maart/downloads/pipedal_1.2.38_arm64.deb' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)

And in the journalctl messages I get every second:
Aug 20 17:37:00 pipedal dhcpcd[566]: wlan0: requesting DHCPv6 information
Aug 20 17:37:10 pipedal dhcpcd[566]: wlan0: requesting DHCPv6 information
Aug 20 17:37:20 pipedal dhcpcd[566]: wlan0: requesting DHCPv6 information
Aug 20 17:37:30 pipedal dhcpcd[566]: wlan0: requesting DHCPv6 information

Shoulds this be? And if so can the message be removed? It pollutes de logging.... 😢

Finally:
I re-installed and re-installed all the plugins ............. it works fine now... the filtering.
VERY happy!!! So much easier when creating an effect chain....
Keeps my choice stress to a minimum 🤣

Thanks Robin for all your effort.

maarthome, 2024-08-20:
Ow... have to laugh... my time goes fast....😂

maarthome, 2024-08-20:
eh.. my time goes fast in your time domain....

rerdavies, 2024-08-20:
> Finally: I re-installed and re-installed all the plugins ............. it works fine now... the filtering.
VERY happy!!! So much easier when creating an effect chain....
Keeps my choice stress to a minimum 🤣

Do you want to keep going on this issue, or would you like to close it? If so, please marke the bug as closed, and we'll let it go.

Please see log a new bug to cover the dhcp log messages.

> during each install I always get the message:
N: Download is performed unsandboxed as root as file '/home/maart/downloads/pipedal_1.2.38_arm64.deb' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)

Please confirm. Are you using this command to install PiPedal?

    sudo apt-get install pipedal_1.2.38_arm64.deb 

If not, that's my fault. The install instructions briefly said that you should use dpkg instead of apt-get. That turned out to be a incorrect.

> potential install problems caused by use of old dpkg installs....

Run the following command, and see if it corrects the issue.

    sudo apt install liblilv-dev

> And in the journalctl messages I get every second:
Aug 20 17:37:00 pipedal dhcpcd[566]: wlan0: requesting DHCPv6 information

It should not do that. Please log a new bug  for this.

maarthome, 2024-08-20:
Hi Robin,

'Do you want to keep going on this issue, or would you like to close it? If so, please marke the bug as closed, and we'll let it go.'
No... let's leave it. I caused it probably myself...

And yes I'll write a bug report for:
'And in the journalctl messages I get every second:
Aug 20 17:37:00 pipedal dhcpcd[566]: wlan0: requesting DHCPv6 information'

Have to laugh.... you made me use dpkg (because I got these messages at the end of the installation: 'Download is performed unsandboxed as root as file ...').... remember I wrote you I was using 'apt-get'? 😊
So yes: 'sudo apt-get install pipedal_1.2.38_arm64.deb '. (Conclusion does not matter which command is used...)

Cheers,
Maarten

rerdavies, 2024-08-20:
Number 1 theory: damage caused by using dpkg instead of apt-get.  :-(

Deeply sorry about that.

closed 2024-08-21 by rerdavies