rerdavies/pipedal #163: Split MIDI control

EE1977, 2024-08-12:
Hi Robin,

Thank you for all your amazing work over here. I noticed an issue with MIDI bindings related to the split function. When trying to assign a potentiometer to the "mix" option, instead of getting a full range adjustment, I only get a toggle function as soon as the pot passes through the 12 o'clock point. That would be great if the full adjustment range was available.

Thanks for your help with this!

rerdavies, 2024-08-12:
I think that's fixed in v1.2.34 beta 2.

Install it (if you haven't already), and let me know whether it fixes the
issue.

    https://github.com/rerdavies/pipedal/releases

rerdavies, 2024-08-12:
> This is the version I have already installed. I can also verify it from
> the "about" menu. I'm afraid that the issue persists.

I'm having trouble replicating the issue. Could you step me through what
you did, step by step, and expand on what you mean by "Toggle behaviour".
Is it possible you have the same midi control bound to bypass?

EE1977, 2024-08-12:
Hey there! All good now, things are clearing up. The problem popped up after I added multiple splits to an FX chain created by a previous version of pipedal. I just put together a new preset, and it looks like the issue is sorted out. Thanks a bunch for getting back to me so quickly!
By 'toggle,' I meant that when the pot value passed the midpoint, the value panned hard right, and vice versa. It was like an L- R switch, around the mid-rotation of the pot.

closed 2024-08-12 by rerdavies