rerdavies/pipedal #171: Install is interrupted (because prerequisite is missing: sudo apt-get install libzip4) and pipedald.service are not installed correctly; pipedal doe not start

maarthome, 2024-08-18:
**Describe the bug**
FIRST minor:

root@pipedal:/home/maart/downloads# sudo dpkg -i pipedal_1.2.37_arm64.deb
(Reading database ... 66976 files and directories currently installed.)
Preparing to unpack pipedal_1.2.37_arm64.deb ...
Removed "/etc/systemd/system/multi-user.target.wants/pipedald.service".
Removed "/etc/systemd/system/multi-user.target.wants/pipedaladmind.service".
Unpacking pipedal (1.2.37) over (1.2.36) ...
dpkg: dependency problems prevent configuration of pipedal:
 pipedal depends on libzip4 (>= 1.0); however:
  Package libzip4 is not installed.

dpkg: error processing package pipedal (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 pipedal

SECOND: Major

pipedald.service are not installed correctly; pipedal does not start.
I went back to version 1.2.36.

Here are the meassages generated during install:

[logging.txt](https://github.com/user-attachments/files/16652392/logging.txt)
[journalctl.txt](https://github.com/user-attachments/files/16652394/journalctl.txt)

I was looking forward for cognitive dissonant effects..... 🤣🤣🤣

I am glad that you are so commited again!! (Nice that you are back!! 👍🎶)

Cheers,
Maarten

rerdavies, 2024-08-18:
What locale is your os running in? (What language?)

rerdavies, 2024-08-18:
Ok. dpkg -i is not the right way to do things.  It should be 

    sudo apt-get install ./thepackage.

I will revise the documentation immediately.

In the meantime, I will await your response about which locale you are using.

rerdavies, 2024-08-18:
Never mind. I was able to replicate the problem by setting my locale to en-ca (Canadian). Fix coming shortly.

rerdavies, 2024-08-19:
Fixed in v1.2.38.

maarthome, 2024-08-19:
Hi Robin,

Thanks!!! Works great!!

.... except a few minor things....
1) I still cannot filter the plugins. I installed Firefox from start... so no coockies or what ever. Also in the Android app, it does not work.
2) In some plugins the selected IR or model the name is not shown in the selection window after the IR or model is selected and the selection is loaded. (I can hear that loading is OK)
Is that the plugin itself? Or something more general?
3) Did I misunderstand that the coloring of the on/off buttons is implemented? I do not see any colors... only the cognitive dissonant RED when the switch is on.... 😊
4) Perhaps a mechanism of creating symbolic links to directories with the models and IR's in order to reduce redundancy.... and improve clarity? I have to thing how... but perhaps in the file selection window offering the possibility to create links to directories... so the user can select the common IR and model directories. In that case no new directory is created but a link.

Cheers and thanks again for the fast response and time and energy!
Maarten

rerdavies, 2024-08-19:
1. Still not able to replicate that. Useful to know that it's a server-side problem. 
2. Could you please log a new bug to that effect, specifying which plugins.
3. Correct. That is not fixed in this release, but will be fixed in a forthcoming release. The fix: use the purple primary color instead of the pink secondary color. This release affects the color of dialog Cancel/OK buttons only.
4. The problem with that is I absolutely want to avoid allowing access through the web interface to any directory outside the uploads directory, which would constitute a significant security risk. So i'm not going to  provide a dialog that browses outside the /var/pipedal/audio_uploads directory. You can create the symbolic links yourself, if that's what you'd like to do. Just make sure the permissions in the target directory grant read access to the pipedal_d account, and read-write access if you want to manage samples in the shared directory through the web interface. Non-Toob plugins place their uploaded files in 
  
    /var/pipedal/audio_uploads/default

rerdavies, 2024-08-19:
A composite bug. #110 covers the color changes. There's an existing bug for not filtering. 4. is rejected. 2 needs a new bug report.

The bug in the title is fixes as of 1.2.38.

closed 2024-08-19 by rerdavies