rerdavies/pipedal #178: [BUG] Cab IRs not loaded using latest version.

38github, 2024-08-24:
I did notice something on my not primary SBC using this new release. When switching between presets I have made there are no IR's loaded in the Toob Cab IR. I will wait with the installation on my primary SBC that has to "just work".

This post first appeared in a discussion but I thought it would be better to put it here.

rerdavies, 2024-08-25:
I tried to replicate the issue, but wasn't able to.

Could you check whether audio has stopped. (visible in the status message at the bottom right if you haven't turned it off, or at the top of the settings page if you have. 

If audio isn't running, the IR filenames won't display. They are actually saved in the patch, but won't show up until you restart the audio thread.

If that's the case, it's on my radar as a low-ish priority bug (with high-ish effort required to fix it). 

If that's not the case, let's dig deeper.

38github, 2024-08-25:
About the interface not being connected is probably it. I guess we can close this and if it somehow wasn't that we can reopen it. 

Thank you!

rerdavies, 2024-08-26:
I'm going to leave it open for now. It should be fixed.  Thanks for bringing it up.

rerdavies, 2024-08-27:
Reproduced the issue. It's  related to an intermittent deadlock in the TooB convolution plugins particularly when using small buffer sizes (16x4, for example).

38github, 2024-08-28:
> Reproduced the issue.

That is good news!

rerdavies, 2024-08-29:
Should be fixed in v1.2.44, just released.

Please reopen the bug if it doesn't solve your problem.

closed 2024-08-29 by rerdavies