rerdavies/pipedal #185: [BUG] Corrupted bank json file when rpi shuts down unexpectedly

FoolHen, 2024-08-29:
**Describe the bug**
If at any point you save a preset and then hard shutdown the pi by removing the power (shouldn't be shut down like this, but can happen sometimes), the bank json file gets corrupted. My guess is that the file handler is not closed after saving.

** Environment:**
 - OS: bullseye, bookworm
 - pi4 4GB
 - v1.2.44 and previous ones

rerdavies, 2024-08-29:
Definitely not the case. All file IO in pipedal is done using c++
std::fstreams, so the file handles cannot remain open.

The bank file does sit in memory, so it's possible that it was corrupted by
a wild write. If you see it again, send me the file.

rerdavies, 2024-08-29:
For what it's worth, my pi gets shut down by cutting the power more often
than not. So this is not an uncommon occurrence for me.

FoolHen, 2024-08-29:
I can reproduce it every time I believe, will send you the file

rerdavies, 2024-08-29:
Just tried it myself, three times, a new preset created and saved each time. I was unable to reproduce the issue.

Are you using an external storage device? I am actually using an external USB SD drive. so that's not generically a problem with external storage. Using an external storage device, and changed PI OS's cached write policy, maybe? 

I am concerned by the report so keep me updated please.

FoolHen, 2024-08-29:
I'm using an SD card on the RPi SD slot. I just reproduced it and the bank file is empty after rebooting. The bank was Grupo.bank and had around 10 presets and now is 0 bytes. Before, in another instance of the bug, the file was not empty but had garbage data. I will try to reproduce it more often to see if it happens again, to send you the file.

rerdavies, 2024-08-29:
:-(

If you manage to capture the bank file before you do the hard reset,
knowing what was in it before the hard reset occurred would be useful as
well.

Plugin related, maybe? Plugins with file properties require special
handling by PiPedal. Actually, any plugin that uses the LV2 State feature
do.

Being able to shut down pipedal by pulling the power cord is absolutely a
must-have feature, as that's exactly what's going to happen when performing.

I did try it with TooB NAM, ToobML and toob Convolution reverb to see
whether it was related to File properties, just in case. Still did not
reproduce.

Sorry for the inconvience. But please know that anything that has the word
"corrupted" in close association to PiPedal has my most urgent attention.

FoolHen, 2024-08-29:
I've copied the file just before unplugging. 
[Before](https://github.com/user-attachments/files/16806353/Grupo.zip)

[After](https://github.com/user-attachments/files/16806358/Grupo-after.zip)

rerdavies, 2024-09-01:
I tried with your files, but still can't reproduce the issue. Both files look fine, with no complications of any kind. (Nice to see that the first plugin in most of  your presets is the TooB Input State plugin. I'm glad that somebody gets why it's useful and good there. 👍  

New theory: perhaps your SD-card filesystem has sustained minor damage. I think it might be worthwhile to run a consistency check.

A forum thread on the subject.

    https://forums.raspberrypi.com/viewtopic.php?f=63&t=179361

My understanding is that Raspberry Pi OS does basic consistency checks, and minor repairs automatically without user intervention on reboot; but that it does not do full consistency checks and repairs unless you follow this procedure. I have had personal experience where the default filesystem check and repair wasn't sufficient and manual intervention was required on previous versions of Raspberry PI OS. Messages about full repairs being needed used to show up on the boot screen, but bookworm doesn't seem to show the boot logs as the as the system comes up anymore. I assume warnings (if any0 would show up in systemd logs.

    journalctl -b0

The short and sweet procedure for doing a more thorough check of your boot drive:

At a command prompt:

     sudo touch /forcefsck
     sudo reboot

Wiat for the system to reboot. It will take longer than normal. After rebooting, check to see if the /forcesck file is still there. It should been deleted. If it has not, 

    rm /forcefsck

You can see results in the systemd log

    journalctl -b 0 | less 

To be honest, I'm not sure what you're looking for. Probably log messages that contain "fsck".

rerdavies, 2024-09-01:
ps: if you encounter cases where the Grupo.bank file has been damaged by a hard reboot, but is not empty, I'd be grateful if you could send me the contents of the corrupted file. Partial corruption sounds suspiciously like a cross-linked file (an issue that fsck would catch).  Seeing the corrupted content could provide a clue as to which process is writing bad data into the file.

rerdavies, 2024-09-01:
Ok. I think I have a fix for you!

I had naively assumed that Raspberry Pi OS wrote files to storage at the earliest possible opportunity (like Windows does). This turns out not to be true. ext4 filesystems can delay writes to disk for up to 5 minutes! I have added to code to make Linux write all pending file operations out to disk whenever PiPedal writes a file.

My best guess as to why I haven't been able to reproduce the issue is that Linux treats USB  drives and SD Card drives differently because it has to take into account the fact that a USB drive is a removable device.

Could you please install  PiPedal v1.2.46-Development, and let me now whether the problem goes away. 

You can either select Check for Updates... in the PiPedal settings dialog, and set the update stream to Development, or you can download and install it manuallly from 

&nbsp;&nbsp;&nbsp;&nbsp;[https://github.com/rerdavies/pipedal/releases/download/v1.2.46/pipedal_1.2.46_arm64.deb]( https://github.com/rerdavies/pipedal/releases/download/v1.2.46/pipedal_1.2.46_arm64.deb)

FoolHen, 2024-09-02:
I just tested it and seems to have fixed the issue indeed, thank you!

rerdavies, 2024-09-03:
Fixed in v1.2.47.

closed 2024-09-03 by FoolHen