rerdavies/pipedal #186: [BUG] cannot install v.1.2.44

bbischler, 2024-08-31:
On my pi 5 running piOS bookworm I cannot install v.1.2.44. The previous versions I was able to install via right click - > package installer. But it just finishes and pipedal did not update. I also tried sudo apt-get install, but it says package could not be found.

maarthome, 2024-08-31:
Hi,
Try sudo apt-get ./pipedal_1.2.44_arm64.deb (the./ makez tge difference...)

Cheers

maarthome, 2024-08-31:
forgot the install... i  between

bbischler, 2024-08-31:
Eh, I just noticed that the Double click to reset knobs is working, but the about pages says I am on v.1.2.42 Experimental.. Is the page maybe not updated?

rerdavies, 2024-08-31:
Unfortunately, you are correct. Version 1.2.44 says it's version "1.2.42 Experiemental" in the about box.  Fixed in the forthcomeing 1.2.45 release.

rerdavies, 2024-09-01:
Fixed in v1.2.45. The About page was displaying the wrong version.

closed 2024-09-01 by rerdavies