rerdavies/pipedal #191: Please add 32 and 64 buffer size to the selection pulldown menu

maarthome, 2024-09-03:
Hi Robin,

Please add the possibility to add a buffer size of 32 and 64 to the pulldown selection menu. See the snapshot below.
(Now it has to be defined manually in '/var/pipedal/AudioConfig.json')

[image]

Thanks and cheers,
Maarten

PS At the end of the install I still get:
'N: Download is performed unsandboxed as root as file '/home/maart/downloads/pipedal_1.2.47_arm64.deb' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)'

Is this important?

rerdavies, 2024-09-03:
Darn! :-/
[image: image.png]

I see the problem though. The minimum buffer size your device allows is 256
samples; mine has a minimum buffer size of 16 samples (but fails pretty
much immediately on anything less than 16x3). I should be allowing you to
use 64x4 (but not 64x3) on your device. Or maybe even 64x2 (stereo), It's
not clear from the documentation.

I'm going to push you a development build that allows you to select crazy
buffer sizes like 16x8. (which actually might be a very good configuration
for various reasons). Could you give it a try at let me know what buffer
sizes actually work on your device. More details to follow, once I push up
a build for you...

rerdavies, 2024-09-03:
p.s. Glad you asked about the "Download performed unsandboxed" message.. I
found the answer to that, but couldn't find an end of an active message
thread on which to provide an answer.

The message is a known issue in APT, and has been for ... close to a
decade. The message is completely harmless and can be ignored. Well.
99.99999% harmless. Nothing is certain when dealing with security issues.
But pretty darned harmless, and also completely expected.

When doing installs, APT attempts to downgrade the credentials of the
current process while fetching files from the internet, and then resumes
actual processing of the downloaded package using root credentials once the
download is complete. This is the "sandbox" the message refers to.The
theory being that doing the web download apis expose a lot of code to
potential abuse by hackers. If there's a bug, much less damage can be done
with a tightly constrained service account than could be done if the HTTPS
code were run with root credentials. Web APIs expose a lot of code to
potential abuse by hackers.

If apt is unable to transfer the file with the service account, it falls
back to using root credentials. If the package in question is a local file,
the service account will be unable to read the package because the file
lies outside of its sandbox. So it falls back to copying the file using
root credentials, and warns you that has done so. Because the transfer is
done using a minimal, and very stable set of APIs (open, read, write,
close), the security risk in this case is close to non-existent.

In case you were wondering, PiPedal also uses this approach when
downloading updates. The actual download of the update is performed on a
service account (pipedal_d), which won't process URLs that are not from
PiPedals github pages, or download files to anywhere other than PiPedal
temporary directories (not actually as tightly sandboxed as APT is, but
pretty close). And the cryptographic signature is checked before passing
the file to a process with root permissions via a very narrow API that
constrains incoming requests as tightly as it can. The admin install
process then verifies the signature again, to be completely certain, before
passing the update file to APT.

rerdavies, 2024-09-03:
PiPedal v1.2.48-Development will be posted in the next 10  minutes or so.  It allows you to select buffer configurations that probably won't work. It allows you to select more-or-less unconstrained buffer configurations. 

If you could perform the following experiments for me, I would be exceedingly grateful.

In a release implementation, I'd like to prevent selection of buffer configurations that clearly won't work. But it's difficult to tell from documentation what kind of behaviour to expect without having access to a device like yours. You can help with that.

It is absolutely crucial that you perform these tests with no load on the PiPedal server. Disconnect any VNC sessions you have, disconnect the HDMI cable (if you have one), reboot the server, and use a remote browser to access PiPedal. At this point, I am aware that the new Vulcan graphics in drivers in bookworm do not get along with real-time audio.

Create and select a preset (or use an existing one) that puts substantial load on the CPU (More than 40% less than 65%). and then save it. And then try the following buffer configurations.

16x4 , mono inputs and outputs - expected to fail
16x6 , mono inputs and outputs-  expected to fail
16x6 , stereo inputs and outputs-  Might work.
32x4, mono input and output - Will probably work, might be unusable due to underrun.
32x6, mono inputs and output - Should work, should be stable.
64x2 mono - Should work . Quite possible that it will underrun.
64x3 mono - Should be stable.

I think that's the minimal set of tests I need.  You may be able to spot other patterns on the fly if you are so inclined. Odd combinations like 16x8 or 16x12 may turn out to provide exceptionally stable behavior on Pi 4s or Pi 5s. But for my immediate needs, I'm more concerned with "will it run at all", and to allow users to experiment with various combinations of load and lead time and latency on their own.

Expected outcomes:
- outright failure on initial selection.
- Runs, but either underruns at a ferocious rate, or stops.
- Device XRuns for in the first couple of seconds, but then stabilizes, and doesn't underrun after that.
- Runs brilliantly. For at least a few minutes.

The tests are focussed on determining what your device's behavior is when BufferSize x  buffers >= 128, and whether mono or stereo configuration makes a difference, if you're looking for a pattern here. 

Also something to keep an eye on for me is whether your Pi is thermally throttling. Pi's automatically reduce their CPU speed if the CPU temperature exceeds 80 degrees C. A pi with heatsinks or fans shouldn't come anywhere close to that when running PiPedal. My Pi runs at between 45C and 60c depending on ambient temperature and load. But if you see CPU temperatures much over 65  degrees or so, let me know. We can have a separate discussion about that if heat throttling is potentially in play.

maarthome, 2024-09-03:
Hi Robin,

Thank you for tempting me to do this littele experiment.... while testing I found a wonderful chord progression om my guitar... 

However I did not fully follow your settings... I see now... (testing and musical improvisation. I am getting too oldd for multitasking 😊)

I have set my pipedal always to stereo in and stereo out.
Sample frequency 48000

**The 'fastest' setting at 65% load is 64 X 6**
**_Temp 65 to 70 degrees celcius._**
(64 X 4 gives once in a while some 'crackeling' in the audio)

No setting whatsoever reported xruns... in the screen....

With 32 buffersize pipedal crashes immediately and does not recover since the setting in the '/var/pipedal/AudioConfig.json' file is still at the 'crash' settings so it has to be adapted manually via the terminal.

Normally I have 30 to 40% load and with these values 64 X 4 is perfect... :-)

One observation/bug:
The dialog window for setting the buffersize always shows 128 when opened again. The  'udioConfig.json' shows the setting is still as I had set it; 64. (The application checks the buffersize of the audio interface again..?)

By the way, the Lewitt (I love this device for its mixer capabilities) is set at buffersize 64. (however I use a mobile usb3 input on the device for my pipedal. I use the Lewitt as my computer audio device (the 'normal' usb connection on the Lewitt)

Thank you for your quick and detailed response,
Cheers,

Maarten

rerdavies, 2024-09-03:
Ok. Not quite what I expected to happen; but I guess it gives me the
information I need to move forward. Thanks for time, and for your
forbearance.

 - R

rerdavies, 2024-09-04:
@mmarhome

> With 32 buffersize pipedal crashes immediately and does not recover since the setting in the '/var/pipedal/AudioConfig.json' file is still at the 'crash' settings so it has to be adapted manually via the terminal.

I'm just finishing up work on a fix for your problem. There's one detail that still needs to be addressed, and that's this one.

By "crashed", do you mean that the service terminates abnormally, or that the service runs, but audio is stopped?

I tried to replicate a crash on my device, by setting very editing a very small buffer size into the config file; but the service comes up fine (and Xruns a few times, and promptly stops, which is what the behaviour should be, I think).

And the code looks good. If device initialization fails, it looks like errors are handled properly.  We are doing something to the ALSA device which we should not do; and we're doing something that should be impossible with the new user interface for buffer selection. But the fact that the service crashes makes me worry that there's a lurking problem anyway.

So. 

If the service does indeed crash, I would really like to see systemd logs, if you wouldn't mind.  Do the deed -- edit the config file, and restart the pipedal service. No need to reboot. Just run this command:

    sudo systemctl restart pipedald

And then, if the service crashes, capture log files for me by running the following command. 

    journalctl -b 0 | grep pipedald 

And send me the results (or post them here). If the results are huge, the last couple of pages out output would be sufficient.

maarthome, 2024-09-04:
Hi Robin,

Here's the log file. I think it looks like the buffer is set to 48 in stead of 32.... (I set it to 32X4)....

PS:
 I use a RPI4 load at 68% and temp at 60 degrees celsius.

[Audio_buffer_error.txt](https://github.com/user-attachments/files/16878601/Audio_buffer_error.txt)

Let me know,
Cheers,
Maarten

rerdavies, 2024-09-05:
> Ow and I also get this message every 10 seconds (old issue if it is not
> intended)
>
> Sep 04 21:36:18 pipedal dhcpcd[567]: wlan0: requesting DHCPv6 information
> Sep 04 21:36:28 pipedal dhcpcd[567]: wlan0: requesting DHCPv6 information
> Sep 04 21:36:38 pipedal dhcpcd[567]: wlan0: requesting DHCPv6 information
> Sep 04 21:36:48 pipedal dhcpcd[567]: wlan0: requesting DHCPv6 information
> Sep 04 21:36:58 pipedal dhcpcd[567]: wlan0: requesting DHCPv6 information
> Sep 04 21:37:08 pipedal dhcpcd[567]: wlan0: requesting DHCPv6 information
> Sep 04 21:37:18 pipedal dhcpcd[567]: wlan0: requesting DHCPv6 information
> Sep 04 21:37:28 pipedal dhcpcd[567]: wlan0: requesting DHCPv6 information
> Sep 04 21:37:38 pipedal dhcpcd[567]: wlan0: requesting DHCPv6 information
> Sep 04 21:37:48 pipedal dhcpcd[567]: wlan0: requesting DHCPv6 information
>
> Cheers,
> Maarten

Please log a new bug to this effect.

rerdavies, 2024-09-22:
Fixed (and made generally available) in v1.2.51. 

None-power-of-two buffer sizes are theoretically permitted, but some plugins declare that they require buffers that are a power of 2, and would have to be excluded. Previous code required buffer sizes to be greater than the minimum buffer size that the device claims that it supports, which is incorrect. The fix allows buffer configurations in which buffersize x nBuffers is greater than the minimum buffer size the device supports.

closed 2024-09-22 by rerdavies