rerdavies/pipedal #193: [BUG] WiFi direct is not enabled in bookworm

FoolHen, 2024-09-03:
**Describe the bug**
Enabling WiFi direct through the interface does not work. The interface shows "Applying changes" and after some seconds it reconnects through WiFi.

**Logs**
The logs, right after enabling WiFi direct in the interface.
```
Sep 03 20:17:27 fulpi pipedald[1051]: WebSocketSession opened. ([::ffff:192.168.2.33]:54625)
Sep 03 20:17:28 fulpi pipedald[1051]: Closing audio session.
Sep 03 20:17:28 fulpi systemd[1]: Stopping pipedald.service - PiPedal Web Service...
Sep 03 20:17:43 fulpi systemd[1]: pipedald.service: State 'stop-sigterm' timed out. Killing.
Sep 03 20:17:43 fulpi systemd[1]: pipedald.service: Killing process 1051 (ppdl_main) with signal SIGKILL.
Sep 03 20:17:43 fulpi systemd[1]: pipedald.service: Killing process 1061 (ppdl_web_4) with signal SIGKILL.
Sep 03 20:17:43 fulpi systemd[1]: pipedald.service: Main process exited, code=killed, status=9/KILL
Sep 03 20:17:43 fulpi systemd[1]: pipedald.service: Failed with result 'timeout'.
Sep 03 20:17:43 fulpi systemd[1]: Stopped pipedald.service - PiPedal Web Service.
Sep 03 20:17:43 fulpi systemd[1]: pipedald.service: Consumed 15.582s CPU time.
Sep 03 20:17:43 fulpi systemd[1]: Starting pipedald.service - PiPedal Web Service...
Sep 03 20:17:43 fulpi pipedald[1270]: Document root: /etc/pipedal/config Threads: 5
Sep 03 20:17:43 fulpi pipedald[1270]: Locale: en_GB
Sep 03 20:17:43 fulpi pipedald[1270]: Scanning for LV2 Plugins
Sep 03 20:17:44 fulpi systemd[1]: Started pipedald.service - PiPedal Web Service.
Sep 03 20:17:44 fulpi pipedald[1270]: Found ALSA device hw:sndrpihifiberry.
Sep 03 20:17:44 fulpi systemd[1]: Stopping pipedald.service - PiPedal Web Service...
Sep 03 20:17:44 fulpi systemd[1]: pipedald.service: Deactivated successfully.
Sep 03 20:17:44 fulpi systemd[1]: Stopped pipedald.service - PiPedal Web Service.
Sep 03 20:17:44 fulpi systemd[1]: Starting pipedald.service - PiPedal Web Service...
Sep 03 20:17:44 fulpi pipedald[1275]: Document root: /etc/pipedal/config Threads: 5
Sep 03 20:17:44 fulpi pipedald[1275]: Locale: en_GB
Sep 03 20:17:44 fulpi pipedald[1275]: Scanning for LV2 Plugins
Sep 03 20:17:45 fulpi systemd[1]: Started pipedald.service - PiPedal Web Service.
Sep 03 20:17:45 fulpi pipedald[1275]: Found ALSA device hw:sndrpihifiberry.
Sep 03 20:17:45 fulpi pipedald[1275]: Audio started. ALSA, hw:sndrpihifiberry, 16-bit little-endian, 44100, 64x3, in: 1/2, out: 1/2
Sep 03 20:17:45 fulpi pipedald[1275]: Listening on fulpi.local:80
Sep 03 20:17:45 fulpi pipedald[1275]: Listening on 192.168.2.120:80
Sep 03 20:17:45 fulpi pipedald[1275]: DNS/SD service announced.
Sep 03 20:17:45 fulpi pipedald[1275]: WebSocketSession opened. ([::ffff:192.168.2.33]:54626)
Sep 03 20:17:49 fulpi pipedald[1275]: WebSocketSession closed. [::ffff:192.168.2.33]:54626
```

**Environment:**
 - OS: Bookworm, RPi 4 
 - Linux fulpi 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux
 - PiPedal Version 1.2.46

rerdavies, 2024-09-04:
Do you see any messages in your log files from the  `pipedal_nm_p2pd`
service after you attept to enable wi-fi direct?  That's the service that
manages the Wi-Fi Direct connection.

    journalctl -b 0 | grep pipedal_nm_p2pd

FoolHen, 2024-09-04:
systemd[1]: Starting pipedal_nm_p2pd.service - PiPedal P2P Session Manager...
systemd[1]: Started pipedal_nm_p2pd.service - PiPedal P2P Session Manager.
pipedal_nm_p2pd[1024]: NMP2PSettings::Load: Wifi Direct service is not enabled.
pipedal_nm_p2pd[1024]: P2PManager::Stop: Shutdown complete.
systemd[1]: pipedal_nm_p2pd.service: Deactivated successfully.

rerdavies, 2024-09-05:
Status update: two bugs, one easy, one deeply concerning. 

1) The pipedal_nm_p2p2 service was taking its configuration from the wrong location. This happened as part of an early fix that ensured that all user-selectable configuration survived an update. Easy fix.

2) An update to Raspberry Pi Os that was pushed to the apt update streams within the last week (sometime after 2024-08-04), seems to have broken Wi-Fi Direct support in Raspberry Pi OS. Not an easy fix at all.

I'm deep into issue #2.  But it's going to take a few days to fix. :-/

FoolHen, 2024-09-06:
Oh, that's unfortunate. Thanks for the update. No rush, for now I'm using the old version in bullseye for daily use.

rerdavies, 2024-09-22:
The entire Wi-Fi direct feature has been replaced with atomatic Wi-Fi hotspots in v1.2.51. Make sure you pick up the latest version of the Android PiPedal Remote app which should be published by google sometime this evening.

closed 2024-09-22 by rerdavies