rerdavies/pipedal #194: [BUG] Audio card is not in the list. Failed to start audio

BorisSutin, 2024-09-05:
After several reboots, the status shows the message "audio device not configured". On the settings and configuration page, the drop-down list among devices is missing. When checking through aplay -l, the list displays correctly. There are no other services that could use the audio device, only the pedalboard. Further reboots did not give any result. Updated to 1.2.48, the audio card works and is displayed in the list. After rebooting, the card disappears again in the pedalboard settings and a message appears that the card is not configured. Aplay -l and other services do not report errors. 
**** List of PLAYBACK Hardware Devices ****
card 0: UAC2Gadget [UAC2_Gadget], device 0: UAC2 PCM [UAC2 PCM]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 3: GtrBox [GtrBox], device 0: 1f000a4000.i2s-adau-hifi adau-hifi-0 [1f000a4000.i2s-adau-hifi adau-hifi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

In the status of the pipedald service, there is a message "Failed to start audio. hw:GtrBox playback device not found. (Device or resource busy)". Jack writes in the status that there are no problems.

● jack.service - JACK2 Audio Server
     Loaded: loaded (/lib/systemd/system/jack.service; disabled; preset: enabled)
     Active: active (running) since Thu 2024-09-05 12:59:54 MSK; 32s ago
    Process: 825 ExecStartPre=/bin/sleep 2 (code=exited, status=0/SUCCESS)
   Main PID: 972 (jackd)
      Tasks: 9 (limit: 4444)
        CPU: 2.178s
     CGroup: /system.slice/jack.service
             └─972 /usr/local/bin/jackd -t 2000 -R -P 95 -d alsa -d hw:GtrBox -r 48000 -p 64 -n 2 -X seq -s

Sep 05 12:59:56 pistomp jackdrc[972]: ALSA: final selected sample format for capture: 32bit integer little-endian
Sep 05 12:59:56 pistomp jackdrc[972]: ALSA: use 2 periods for capture
Sep 05 12:59:56 pistomp jackdrc[972]: ALSA: final selected sample format for playback: 32bit integer little-endian
Sep 05 12:59:56 pistomp jackdrc[972]: ALSA: use 2 periods for playback
Sep 05 12:59:56 pistomp jackdrc[972]: port created: Midi-Through:midi/playback_1
Sep 05 12:59:56 pistomp jackdrc[972]: port created: Midi-Through:midi/capture_1
Sep 05 12:59:56 pistomp jackdrc[972]: port created: touchosc:midi/playback_1
Sep 05 12:59:56 pistomp jackdrc[972]: port created: touchosc:midi/capture_1
Sep 05 12:59:57 pistomp jackdrc[972]: No jack_get_descriptor entry-point for mod-midi-merger
Sep 05 12:59:58 pistomp jackdrc[972]: No jack_get_descriptor entry-point for mod-midi-broadcaster

Raspberry Pi 5
Linux pistomp 6.6.23-v8-16k+ #1 SMP PREEMPT Thu Mar 28 14:31:44 MSK 2024 aarch64 GNU/Linux

BorisSutin, 2024-09-05:
The issue is closed, there were some old version jack  in the system. I cleaned it up and everything worked. Sorry for bothering you.

closed 2024-09-05 by BorisSutin

rerdavies, 2024-09-05:
:-) The probable issue: jack (and pipewire) are unable to share audio devices between desktop sessions and applications running as daemons (PiPedal).  Older version of jack allowed sharing of devices between sessions (jack, but not jack2?).  But it was a bit flaky; and jack-via-pipwewire makes no attempt at all to provide jack sessions to anything but desktop sessions. Which is why PiPedal switched to using ALSA instead of jack.  

And whoever gets the ALSA device first gets it. My guess: you had set up an old jack shared connection of some sort that was grabbing the ALSA device before PiPedal had a chance to get it. Or even see it. One of the quirks of ALSA is that you can't even see devices once somebody else has opened them.

But I think you probably had that figured out already.

BorisSutin, 2024-09-06:
I don't have pipewire, the version without desktop. Previously, there was a Modep on this build. But after installing PiPedal, I disabled all unnecessary services.