rerdavies/pipedal #195: Every 10 seconds this message appears (With changing time  and date of course 🤣): Sep 04 21:37:08 pipedal dhcpcd[567]: wlan0: requesting DHCPv6 information

maarthome, 2024-09-05:
**Describe the bug**
Every 10 seconds this message appears

Sep 04 21:36:18 pipedal dhcpcd[567]: wlan0: requesting DHCPv6 information
Sep 04 21:36:28 pipedal dhcpcd[567]: wlan0: requesting DHCPv6 information
Sep 04 21:36:38 pipedal dhcpcd[567]: wlan0: requesting DHCPv6 information
Sep 04 21:36:48 pipedal dhcpcd[567]: wlan0: requesting DHCPv6 information
Sep 04 21:36:58 pipedal dhcpcd[567]: wlan0: requesting DHCPv6 information
Sep 04 21:37:08 pipedal dhcpcd[567]: wlan0: requesting DHCPv6 information
Sep 04 21:37:18 pipedal dhcpcd[567]: wlan0: requesting DHCPv6 information
Sep 04 21:37:28 pipedal dhcpcd[567]: wlan0: requesting DHCPv6 information
Sep 04 21:37:38 pipedal dhcpcd[567]: wlan0: requesting DHCPv6 information
Sep 04 21:37:48 pipedal dhcpcd[567]: wlan0: requesting DHCPv6 information

RPI4
Bookworm
Linux pipedal 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux
Lewitt Connect 6 interface
PiPedal Version 1.2.48 development

Cheers,
Maarten

rerdavies, 2024-09-05:
Could you try the following for me.

- download this  attached file. [dhcpcd.conf.txt](https://github.com/user-attachments/files/16900919/dhcpcd.conf.txt)
- Copy it into place with these commands (pay attention to the different extensions):

    cd ~/Downloads
    sudo cp dhcpcd.conf.txt /etc/dhcpcd.conf

- Reboot and check to see whether the log message still appears.

maarthome, 2024-09-06:
Hi Robin,

Thanks!! 👍😊 The message disappeared...
So what was wrong? If anything...

Cheers,
Maarten

rerdavies, 2024-09-08:
The trigger: no IPv6 on your raspberry pi. The bug: IPv6 needs to be
disabled when pipedal enables DHCPCD for use with Wi-Fi Direct. And DHCPCD
needs to be more sensible about this.

Something like that.

closed 2024-09-22 by rerdavies