rerdavies/pipedal #204: [BUG] Auto Hotspot Not working

bbischler, 2024-10-03:
Hey,
I am running pipedal v1. 2.52 on a pi5 with bookworm (latest Version) and the auto WiFi Hotspot feature is not working properly.
Most of the time my Android Smartphone (Samsung s20) cannot find the WiFi. I tried multiple frequencies. Only sometimes it sees it, but only once I was able to connect. Then, opening the App (latest Version) it did find my pi, but got stuck loading the screen.
Is there something I can configure to have a more stable connection?
Its quiet a bummer, because without being able to connect via App, I dont want to take pipedal to Gigs.

rerdavies, 2024-10-03:
1. Which version of Android is your s20 using? Android 13?

2. Which Auto-connect mode are you using? Are you sure PiPedal has enabled
the hotspot on your Raspberry Pi?

3. Before doing anything else, reboot both your Raspberry Pi, and your
phone. If you use a VPN on your phone or  Raspberry Pi, make sure it is
disabled.

4. Can you connect using the Android Wi-Fi user-interface? Can you see the
hotspot? What happens if you try to connect to it using the Android Wi-Fi
UI?

5. Follow this procedure to get me log files that cover a period when
PiPedal should have made the hotspot available.

   - Reboot your Raspberry Pi so we get fresh log files.
   - Wait for about 20 seconds after the reboot to allow the PiPedal
   service to start up.
   - Do whatever you need to do to get PiPedal into a state where it will
   bring up the hotspot. Configuring the hotspot as  Setting the
hotspot to *Always
   on *is ideal. If not, unplug your ethernet cable, or power off your
   Wi-Fi router as appropriate for the auto-hotspot setting you are using. If
   the hotspot should be enabled at boot time, you don't need to do anything
   in this step.
   - Wait for another 30 seconds to allow wi-fi services to come up.
   - Try to connect to the hotspot from your phone using the Android's
   native Wi-Fi user interface (*not * the Pipedal client interface). Can
   you see the hotspot? Can you connect to it?
   - Do whatever you need to in order to gain access to your Raspberry Pi
   again without rebooting (plug in the ethernet cable, or turn on your Wi-Fi
   router again if necessary).
   - Without rebooting, from a bash prompt, issue the following command

                     journalctl -b 0 | grep pipedald

and send me the output. You can post to the current github conversation, or
if you have concerns about disclosing private information, you can email it
to me directly at *rerdavies at gmail.com <http://gmail.com>. *I would
recommend emailing it to me directly. Log files will contain internal IP
addresses, and names of computers on your personal network, but no
passwords, or other particularly personal information.

bbischler, 2024-10-07:
Hi, 
thanks for your help.
I wanted to follow yuor steps, but now pipedal crashes completly. here is the log:
```
Okt 07 21:06:24 raspberrypi pipedald[2159]: Scanning for LV2 Plugins
Okt 07 21:06:25 raspberrypi systemd[1]: Started pipedald.service - PiPedal Web Service.
Okt 07 21:06:25 raspberrypi pipedald[2159]: Found ALSA device hw:V2.
Okt 07 21:06:25 raspberrypi pipedald[2159]: Audio started. ALSA, hw:V2, 32-bit integer little-endian, 48000, 64x3, in: 2/2, out: 2/2
Okt 07 21:06:25 raspberrypi pipedald[2159]: Listening on mDns address raspberrypi:80
Okt 07 21:06:25 raspberrypi pipedald[2159]: Listening on Wi-Fi address 192.168.0.xxx:80
Okt 07 21:06:25 raspberrypi pipedald[2159]: DNS/SD service announced. (pipedal)
Okt 07 21:06:26 raspberrypi pipedald[2159]: DNS/SD group established.
Okt 07 21:06:26 raspberrypi pipedald[2159]: HotspotManager: Monitoring network status.
Okt 07 21:06:26 raspberrypi pipedald[2159]: HotspotManager: Enabling PiPedal hotspot.
Okt 07 21:06:26 raspberrypi pipedald[2159]: HotspotManager: Activation failed: [org.freedesktop.NetworkManager.Settings.Connection.InvalidProperty] 802-11-wireless.channel: »36« ist kein gültiger Kanal
Okt 07 21:06:26 raspberrypi pipedald[2159]: HotspotManager: PiPedal hotspot disabled.
Okt 07 21:06:26 raspberrypi systemd[1]: pipedald.service: Main process exited, code=killed, status=11/SEGV
Okt 07 21:06:26 raspberrypi systemd[1]: pipedald.service: Failed with result 'signal'.
```
This is with v1.3.53

rerdavies, 2024-10-09:
Fixed in v1.3.54.

closed 2024-10-09 by rerdavies