rerdavies/pipedal #206: Performance view

EE1977, 2024-10-06:
Hey Robin, awesome work with Pipedal as always! It's getting better and better!

I just gave the new release a try, and it looks really promising!

The only thing is, every time I try to change a snapshot, 
I see a "reconnecting" message on my Pi5 screen for a few seconds. 
Also, the remote Android app asks again to reconnect to a device. 
Am I doing something wrong?

P.S. And something more... I 'd love to see the ability to choose and add a few basic audio editing controls 
in the performance screen, so as to make basic adjustments on the fly using my pi5 touchscreen. 
Controls that I could select by myself (the same way I select my MIDI bindings) based on the lv2
effects my fx chain has loaded.

rerdavies, 2024-10-06:
That is, unfortunately, an indication that PiPedal has crashed (Something
severe, like a segment violation).

I've re-test with a basic set of plugins without problems, and it works
fine.

Is it possible that it's related to a particular plugin? WOuld you mind
trying with preset that contains only... (say..) TooB ML.

If that works, could you try stripping down the preset you're using that
has the problem. (Save As.... first to a new copy of the preset, and make
changes on the copy).)

rerdavies, 2024-10-06:
*> I 'd love to see the ability to choose and add a few basic audio editing
controlsin the performance screen, so as to make basic adjustments on the
fly using my pi5 touchscreen.Controls that I could select by myself (the
same way I select my MIDI bindings) based on the lv2effects my fx chain has
loaded.*
Yes. Me too. :-) I need to give some serious thought as to how to do that.

EE1977, 2024-10-08:
Hi Robin thanks for your prompt answer. I took the time to test the issue. It seems to be related to splits. Every time I tried to add a split to my signal chain I have the same issue. Reconnecting.

rerdavies, 2024-10-08:
Ok. I have a fix for that. Should be released later today.

rerdavies, 2024-10-09:
Fixed in v1.3.54.

closed 2024-10-09 by rerdavies