rerdavies/pipedal #213: Performance view

EE1977, 2024-10-10:
Hi Robin!! I just tried the 1.2.54 version but unfortunately, the issue with the splits persists for me... Creating or switching between snapshots, triggers a reconnecting message and pipedal freezes for some seconds. As soon as I remove the split everything works just fine.

rerdavies, 2024-10-10:
Did you edit the split type, or the split control values? (makes a big
difference as to which parts of my code I need to look at).

EE1977, 2024-10-10:
Just added a split, then clicked on a snapshot and ... then reconnect. my
fx strip has no issues.

rerdavies, 2024-10-10:
OK. Managed to reproduce the issue. Thanks.

EE1977, 2024-10-10:
I also encountered the same issue while using my USB MIDI. Whenever I attempt to use a split or adjust it, it just displays the reconnection message. Same thing when trying to adjust settings via MIDIe.g mute unmute from a gain plugin, adjust a pitchbend fx, etc...

rerdavies, 2024-10-11:
The MIDI controller issue was unrelated. It has been fixed in v1.3.56. If you could give it a try, and let me know whether all of the problems have been addressed, I'd be grateful.

EE1977, 2024-10-12:
Now, 1.3.56 the USB MIDI controller works just fine!! Thank you!! Problem solved!!

closed 2024-10-14 by rerdavies