rerdavies/pipedal #214: The problem with using PiPedal.

AlexLesPaul, 2024-10-10:
Hi. I have this problem after installation PiPedal.There is no sound. There is no way to select a model in plugins TooB Neural Amp Modeler, TooB ML Amplifier, TooB Cab IR, TooB Convolution Reverb. I select a model from the list, but nothing happens. The model is not selected. I tried installing version 1.2.47, 1.2.51, 1.2.52, 1.3.54. It's the same everywhere. It is impossible to select a model and there is no sound.That's how I did it.
                   System Raspberry Pi OS Lite (64-bit)
sudo apt update
sudo apt upgrade
sudo reboot
mkdir Downloads
cd Downloads
wget https://github.com/rerdavies/pipedal/releases/download/v1.3.54/pipedal_1.3.54_arm64.deb
sudo apt-get install ./pipedal_1.3.54_arm64.deb
sudo reboot
Configuring Audio 48000-64-2
What did I do wrong?

rerdavies, 2024-10-11:
Do the following for me please.

   - Reboot your Raspberry Pi.
   - Run the following command and send me the results.

        journalctl -b0 | grep pipedald

AlexLesPaul, 2024-10-11:
[image]

rerdavies, 2024-10-11:
Run the following command for me, and send me the results:

    amixer -c Audio contents

AlexLesPaul, 2024-10-11:
[image]

rerdavies, 2024-10-11:
Do the following for me, and let me know what the results are. 

From the main menu, select "Settings". In the Settings dialog, select "Input Channels". In the dialog, select:

  Mono (left channel only).

And see if that corrects the problem. If it fixes the problem, let me know.

If it doesn't work, reboot your Raspberry pi, and send me the output of the following command:

    journalctl -b0 | grep pipedald

AlexLesPaul, 2024-10-11:
[image]

AlexLesPaul, 2024-10-11:
I did a clean installation. All over again. I applied your recommendations. It seems to have worked.
[image]

rerdavies, 2024-10-11:
Great! The problem is that PIpedal wasn't dealing correctly with audio
devices that only have one input channel. There will be a fix for that in
the next release of PiPedal. But in the meantime, you have the workaround.
Thanks for your help in debugging the problem.

rerdavies, 2024-10-14:
No complete repro, but some issues definitely fixed.

closed 2024-10-14 by AlexLesPaul