rerdavies/pipedal #217: [BUG] Bank content changes after selecting a different preset in the preset creation window

maarthome, 2024-10-11:
**Describe the bug**
**Bank content changes after selecting a different preset in the preset creation window
I think you should at least press 'save' somewhere before changing the content of the bank.**

_(And.... what is a bank? The same as a preset but with the possibility of having differen sets of setings for ONE (1) preset?

I think its time to define what is what?

I would say: 
Bank is set of presets
Preset is a set of plugins and their settings. 

'Performance' (MIDI selectable) is a set of setings (MIDI assignable per preset) for ONE (1) preset  (selectable) from ONE Bank (selectable)
'Performance list' is the (one (1)) set of performances

Or? Am I stil missing something? :-)
)_

Cheers,

Maarten

**Screenshots**
If applicable, add screenshots to help explain your problem.

** Environment (please complete the following information):**
 - OS: [e.g. Raspbian, Ubuntu Gnome3 21.04, Windows 10]
- if Linux, run `uname -a` and include the results.
 - Device that the pipedal server is running on (pi 3, pi 4, &c, how much system memory)
 - PiPedal Version [See X/Settings/About,...]
If UI-related:
 - Browser [e.g. pipedal Android app, safari, Chrome]

maarthome, 2024-10-11:
AND...

When the settings of a plugin of a preset are changed in performance view, these settings are also save to the original preset.....
The original preset settings should not change when edited/saved in the performance view.

Cheersm,

Maarten

rerdavies, 2024-10-11:
> _Bank content changes after selecting a different preset in the preset creation window
I think you should at least press 'save' somewhere before changing the content of the bank._

Bank files are files that contain multiple presets. Similar idea to a midi bank. You can switch between banks (and manage banks) from the main menu. If you haven't discovered banks yet, I assume you're working in the "Default bank".  You might want to create a bank of presets that contains only the presets you use, with all the factory presets deleted. Or have a separate bank of presets that contain only the presets you use for a given set list. Lots of reasons why you might want to use different banks of presets for different things.

The bank file changes because it contains the preset itself, but also contains the most recently selected preset. When you reload a bank (or reload PiPedal), the first preset you see is the one you selected last. So when you select a new preset, the bank file changes.  

Is there a problem there? If you're tracking the bank file OUTSIDE of the pipedal UI, I think you'll find that only one property changes: the currently selected preset.

> When the settings of a plugin of a preset are changed in performance view, these settings are also save to the original preset..... The original preset settings should not change when edited/saved in the performance view.

Unfortunate, but I don't think I can get around it. When you save the preset, you save both the current control settings, and the snapshots. In this case, the **_current_** control settings ARE whatever the result of applying the most recent snapshot was (plus, potentially, changes to controls made by midi controllers).

And I actually do think it's the Right Thing To Do, although I'm having a hard time clearly articulating why that is the case. 

My best explanation: the Performance view is a view of the same thing the main page view shows. It's just arranged differently. Part of that thing is the concept of a currently loaded pedalboard, which matches what you hear on the realtime audio thread. When you save a preset, it should do the same thing, whether you do it from the Performance view or the main page view. And part of that is saving the state of the currently pedalboard. And the same for loading a preset: the results should be the same whether you do it from the main page or from the Performance view. The "current pedalboard" isn't in your face in the performance view, like it is in the main page view; but it is there nonetheless, because it's what you hear on the audio thread.

maarthome, 2024-10-11:
Thanks for your explanation.

I am still not sure what the use is of the performance screen;  different parameter settings of a same preset of plugins, not MIDI selectable.... (I always saved identical plugin configuration (preset) with different individual parameter settings with a different name... )

I am eager to see the use-case of the preformance view in the future.

Cheers and thanks again,

Maarten

rerdavies, 2024-10-11:
Snapshots ARE midi-selectable. In fact that's the primary use-case for the
entire Snapshot feature: selection of collections of controls using
footswitches on a MIDI pedalboard. The Performance View is sort of the
poor-man's version of Snapshots for those of us who don't have MIDI
pedalboards.

Midi bindings for snapshot selection can be found in the System MIDI
Bindings menu item in the settings dialog.

> I am eager to see the use-case of the preformance view in the future.

I'm with you on that one. :-/ I feel like I've delivered a high-cost
feature that's important to a very select few of PiPedal's users (those who
have midi footswitches). So it goes.

I'm still mulling over what next steps would be. Some way to get controls
in the Performance view that you and I would enjoy. Like preset selection
buttons, or maybe bindable customizable controls. Input and output VUs.

EE1977, 2024-10-11:
The discussion about controls over the Performance view is intriguing. I believe it would be beneficial to have bindable,  customizable controls such as sliders or pots, and it would be particularly useful to include a MUTE switch along with a volume control. That said, that would seriously resemble a compact touchscreen MIDI control, as said above.

rerdavies, 2024-10-11:
The thing I'd like to avoid is having to implement a fullblown layout
editor, which is almost a product in its own right (thinking of TouchOsc,
for example). Particularly challenging in PiPedal, which could be connected
to portrait or a landscape displays, desktop or tiniest phone displays,
maybe even both at the same time.

It's not so much what sort of controls that's the challenge (although I
have to admit I had not thought of sliders!). The tough question for me
right now, is how to edit the layout of the controls.

Something like the layout editor for widgets on an Android home screen
maybe, on a constrained 5x5 grid. Still a lot of work.

And a painful bit of UI ugliness when rotary dial controls are close to a
screen edge.

And how to deal with program/preset select buttons when the user loads a
new bank..

These are the sorts of things that keep programmers awake at night. And
cause programmer dreams (you may have to be a programmer to understand that
reference).

rerdavies, 2024-10-11:
@maarthome 

RE your original post, you almost have it, but not quite...

Banks contain one or more presets. Each preset contains up to six SNAPSHOTs. And the Performance View is the view that's optimized for use when you are performing, as opposed to when you are fiddling around creating and editing new presets (which you do in the main view). The idea being that, when you are performing, you want a display with as little clutter as possible. That's the concept anyway. Not sure I'm entirely delivering on that concept right now.. But it i think it's a good step in the right direction.

If that's confusing to you, then I see it as a problem with MY user interface. If you find it confusing, then  so will THOUSANDs of other users. I'm thinking of adding an old-school help button to the performance view to provide an explanation. Can you think of any tweaks to the UI that might have made things more obvious? 

I'm guessing that most of the confusion stemmed from the idea that the buttons were for selecting presents, not snapshots.

rerdavies, 2024-10-11:
I'm demoting this thread to a _Feature Request_ rather than a _BUG_ at this point. Just for my management purposes. I think that's where we currently stand.

maarthome, 2024-10-12:
Hi Robin,

Can the color of the active/selected SNAPSHOT be shown bright and the others dimmed (and the 2 levels 'bright' and 'dimmed' their intensity user defined )? Then on stage I cab see what is active. Otherwise the large colored fields do not have a use for me.

Cheers,

Maarten

maarthome, 2024-10-12:
.... and the text size adjustable so I can read it while standing.... (so... VERY large )

rerdavies, 2024-10-14:
Implemented in 1.3.57:  text size is larger (depending on available display size); there is also a now select shading (although it doesn't work that well with multi-colored buttons); and the select-border contrast has been increased.

closed 2024-10-19 by rerdavies