rerdavies/pipedal #219: [BUG] Description

FoolHen, 2024-10-12:
**Describe the bug**
Awesome with with the latest updates. There are some small regressions I've noticed, which were working fine some versions ago:

- After clicking "Update" on the updater dialog, the "updating.." overlay message isn't shown anymore
- GxTuner does not work anymore. The dial does not move
- The Android app does not pick up the system default theme setting. I have the color theme set to follow system but it does not turn to dark mode (which is my system theme)
- After connecting to the pi pedal app, going to another app (and so the websocket connection is stopped), and going back, the vu input and output meters do not update anymore. In the screenshot you can see how the plugin vu meter works just fine but not the input and output ones. Im not sure if it's a regression too.

Let me know if you prefer me to open different tickets for each issue.

**Screenshots**
[image]

rerdavies, 2024-10-12:
If you wouldn't mind opening separate tickets, I'd appreciate it. Makes it
easier not to lose bugs in the shuffle.

rerdavies, 2024-10-14:
Fixed in v1.3.57. The client now reconnects after being suspended, since Android appears to silently discard websocket messages when the app is suspended.

FoolHen, 2024-10-16:
Fixed, thanks

closed 2024-10-19 by rerdavies