rerdavies/pipedal #220: [BUG] NetworkManager.Settings.PermissionDenied

bbischler, 2024-10-12:
With v1.3.56 I get this error:
```
Okt 12 14:44:25 raspberrypi pipedald[2434]: HotspotManager: Monitoring network status.
Okt 12 14:44:25 raspberrypi pipedald[2434]: HotspotManager: Enabling PiPedal hotspot.
Okt 12 14:44:25 raspberrypi pipedald[2434]: HotspotManager: Activation failed: [org.freedesktop.NetworkManager.Settings.PermissionDenied] uid 999 has no permission to perform this operation
Okt 12 14:44:25 raspberrypi pipedald[2434]: HotspotManager: PiPedal hotspot disabled.
Okt 12 14:44:25 raspberrypi systemd[1]: pipedald.service: Main process exited, code=killed, status=11/SEGV
Okt 12 14:44:25 raspberrypi systemd[1]: pipedald.service: Failed with result 'signal'.
```

bbischler, 2024-10-12:
This is on a pi5 running bookworm

rerdavies, 2024-10-12:
Could you run the following command for me:

       sudo ls -l -h /etc/polkit-1/rules.d

There should be one file there, named 10-pipedal-networkmanager.rules with
a size of 276 bytes.

If there is more than one file, send me the results of the previous
command.

And if you have not previously done so, could you reboot to see if that
fixes the issue.

bbischler, 2024-10-12:
indeed, one file there:
```
basti@raspberrypi:~ $ sudo ls -l -h /etc/polkit-1/rules.d
insgesamt 4,0K
-rw-r--r-- 1 root root 276 27. Sep 16:57 10-pipedal-networkmanager.rules
```
I tried rebooting multiple times, even uninstalled pipedal und reinstalled with no luck

rerdavies, 2024-10-12:
1. Have you made any changes that would have a significant effect on system security? (It would have to be something major that significantly hardens system security.

2. Are you running full Raspberry Pi OS, or Raspberry Pi OS Lite?

3. Could you send me the output of the following command.

      sudo ls -l /usr/share/polkit-1/rules.d/

4.  Reboot pipedal, and send me the complete system log:
 
          journalct -b 0 > ~/systemlog.txt

     You can email me the file directly at rerdavies at gmail.com, instead of posting here if you prefer.

bbischler, 2024-10-13:
Hi,
It happened after I uploaded IRs to the toob reverb.
In the meantime I reinstalled bookworm completely as well as pipedal and it works now again. So lets just hope this doesnt Happen again. Thanks for your support!

closed 2024-10-13 by bbischler