rerdavies/pipedal #229: Tooltip in knobs title

bbischler, 2024-10-18:
Currently, if the title of a knob is too long, an ellipsis is shown, which is great. It would be cool to have tooltip on hover (or long press on mobile) to see the whole string.

bbischler, 2024-10-19:
Alternatively you could wrap the Text. But then the alignment of the knobs in a row should change accordingly I guess.. Or auto scale down the font size?! You'll be the judge ☺️

rerdavies, 2025-02-21:
Plugin controls now have long-hover/long-press tooltips. But non-plugin controls (bypass, info, *c) should have tooltips/long-press hints too /

rerdavies, 2025-03-13:
Actuallly implmented in desktop browsers, but non-function on mobile UIs. Should actually be implemented as long-press on mobile UIs. Deferred for re-work.