rerdavies/pipedal #232: [BUG] Import Preset and Bank

BorisSutin, 2024-10-21:
**Describe the bug**

Presets and banks are not imported. The upload procedure passes without errors , but the preset
does not appear in the list , and the same applies to banks.  No ability to exchange presets

vers 1.3.59

BorisSutin, 2024-10-21:
Thank you Robin, you are very quick to correct errors.  With 1.3.60 presets are imported correctly, but importing the bank again causes the error. I'll attach the video and the bank itself for the test. 

https://github.com/user-attachments/assets/d9f78bae-1230-4e36-9f2d-5ca70aefd3d2

[Bank Pistomp.piBank.zip](https://github.com/user-attachments/files/17465235/Bank.Pistomp.piBank.zip)

rerdavies, 2024-10-29:
Fixed in v1.3.61 or v1.3.62, I think. Sorry. I seem to have fumbled the release notes on this issue. It's definitely fixed in v1.3.62.

closed 2025-02-21 by rerdavies