rerdavies/pipedal #238: Hardware inputs outputs issue with a hifiberry soundhat on a pi5

EE1977, 2024-10-28:
Hi Robin!! Just tried to switch over my hifiberry hat on my pi5, just to find out that pipedal does not recognize its stereo nature. Using the same SD card on my pi4B+ I have no issues. Hardware channels are shown correct;y an the hat initialises perfectly, with just the L/R channels. On my pi5 though, instead of L/R channel or stereo options, I read 8 inputs and 8 outputs. Using either of them does not produce any sound at all. Any suggestions?

rerdavies, 2024-10-28:
1, Were you using Bullseye or Bookworm on your pi5?  I'm surprised that there would be a difference between pi4 and pi5 on the same OS level. 

2 Could you run the following command and let me know what you see:

```
   amixer contents --card CARDNAME
```
Where _CARDNAME_ is the name of the audio adapter as you see it in PiPedal.

rerdavies, 2024-10-28:
You might also want to ask the same question on Hifiberry support sites. If there's something I should be doing differently, i would be happy to accommodate. fwiw, I'm using the ALSA hw: device and using the number of channels that ALSA says the device has.So there shouldn't be any sort of problem with interventing ALSA adapters. Just what the ALSA says the device has.

EE1977, 2024-10-28:
The command points to an invalid card name when I use "snd_rpi_hifiberry_dacplusadcpro". This is what pipedal recognizes.
Instead, I took a couple of screenshots that depict the issue...
[image]
[image]
[image]

As for your OS question, it is Bookworm. Actually it is the same SD card. When I plug it in my Pi4, Pipedal runs smoothly, but when I plug the SD card in my pi5 I have the issue above.

EE1977, 2024-10-28:
Thank you for your prompt response! The SD card image was actually created on my Raspberry Pi 5. Despite my efforts, I haven’t been able to get it to work properly. However, when I tested the card in my Raspberry Pi 4, it worked perfectly. 

I’m also curious if there’s a way to check what ALSA detects and possibly force a left/right audio configuration to see if that helps. I recall reading a comment on a forum about someone who resolved a similar issue by manually adjusting the ALSA config file.

rerdavies, 2024-10-28:
Ok. That issue is taken care of then. Onto next steps... 

Let's take a look in system logs to see if anything odd is going on.  

1.    Select the configuration for your audio adapter as best you can, and reboot.
2.   After a clean reboot, issue the following command, and send me the result. Either post the results here, or mail the results to me directly at rerdavies at gmail dot com.  Log files may disclose details like ip addresses and network device names, but shouldn't otherwise disclose sensitive information. You may want to review the logs yourself before posting or emailing.
```
    journalctl -b 0 | grep "(pipedald)|(alsa)|(asound)"
```
   You might also want to scan the log files yourself to see if there is anything sensitive in the filtered logs (I don't think there will be).

4.    Run the following command, and post the results here. Substitute the name of the audio adapter as you see it in PiPeda for _CARDNAME_
```
 amixer contents --card CARDNAME
```

5. Does the audio adapter work with standard system utilities like `aplay` and `acapture`?  To test this, you will have to get PiPedal to let go of the audio device in question. The best way to do this is to temporarily disable pipedald (and re-enabled it when done).  The reboots are not optional. ALSA sometimes (often) gets itself into a state where it can't recover from previous errors.

    Use the following commands:

```
     sudo pipedalconfig --disable # disable the pipedal service, followed by a reboot
     sudo reboot

    sudo pipedalconfig -- enable # enable the pipedal service, and then reboot.
   sudo reboot
```

5. You asked   

    *I’m also curious if there’s a way to check what ALSA detects and possibly force a left/right audio configuration to see if that 
    helps. I recall reading a comment on a forum about someone who resolved a similar issue by manually adjusting the ALSA 
   config file).*

    There ARE ways to create ALSA devices that upmix or downmix channels, but PiPedal currently ignores them. Pipedal
    currently only allows selection  of "hw:" devices, in order to prevent users from accidentally selecting pseudo-devices that have 
    layers of audio processing and buffering between Pipedal and the actual hardware device.  This fine for casual or gaming audio; but far  from ideal for pro-audio low-latency audio. If you have a candidate solution, post it here, and I'll see whether PiPedal can do something sensible to accommodate the proposed solution. 

    It is possible that your device provides multiple capture and playback subdevices with different channel configuratons, and that Pipedal has chosen the wrong one. The output of the `amixer` command should  indicate whether this is the case. The fact that your device reports 8 channels is a bit disturbing. Unless it does actually have 8 channels.

fwiw, piPedal deals with N-channel devices (N > 2) as follows: it opens reads and writes ALL the channels, performing mixing between channels,  per pipedal's current channel selection. This is an untested configuration in piPedal, but it's a relatively safe path, not hugely different from the 2-channel submixing case. But logfiles should indicate whether there's a problem with that. 

The other possible point of friction between PiPedal and your device is the question of binding the input and output device together so that they share a common system clock. PiPedal does do this.  But there's some indication in internet chatter that there's something odd about the way Hifiberry devices do system-clock binding. (But WHY are there 8 channels! when you only seem to have 2?)

EE1977, 2024-10-28:
Running the  "amixer contents --card CARDNAME" command I have the results below:

numid=6,iface=MIXER,name='DSP Program'
  ; type=ENUMERATED,access=rw------,values=1,items=5
  ; Item #0 'FIR interpolation with de-emphasis'
  ; Item #1 'Low latency IIR with de-emphasis'
  ; Item #2 'High attenuation with de-emphasis'
  ; Item #3 'Fixed process flow'
  ; Item #4 'Ringing-less low latency FIR'
  : values=0
numid=27,iface=MIXER,name='ADC Left Capture Source'
  ; type=ENUMERATED,access=rw------,values=1,items=19
  ; Item #0 'No Select'
  ; Item #1 'VINL1[SE]'
  ; Item #2 'VINL2[SE]'
  ; Item #3 'VINL2[SE] + VINL1[SE]'
  ; Item #4 'VINL3[SE]'
  ; Item #5 'VINL3[SE] + VINL1[SE]'
  ; Item #6 'VINL3[SE] + VINL2[SE]'
  ; Item #7 'VINL3[SE] + VINL2[SE] + VINL1[SE]'
  ; Item #8 'VINL4[SE]'
  ; Item #9 'VINL4[SE] + VINL1[SE]'
  ; Item #10 'VINL4[SE] + VINL2[SE]'
  ; Item #11 'VINL4[SE] + VINL2[SE] + VINL1[SE]'
  ; Item #12 'VINL4[SE] + VINL3[SE]'
  ; Item #13 'VINL4[SE] + VINL3[SE] + VINL1[SE]'
  ; Item #14 'VINL4[SE] + VINL3[SE] + VINL2[SE]'
  ; Item #15 'VINL4[SE] + VINL3[SE] + VINL2[SE] + VINL1[SE]'
  ; Item #16 '{VIN1P, VIN1M}[DIFF]'
  ; Item #17 '{VIN4P, VIN4M}[DIFF]'
  ; Item #18 '{VIN1P, VIN1M}[DIFF] + {VIN4P, VIN4M}[DIFF]'
  : values=1
numid=22,iface=MIXER,name='ADC Left Input'
  ; type=ENUMERATED,access=rw------,values=1,items=5
  ; Item #0 'No Select'
  ; Item #1 'VINL1[SE]'
  ; Item #2 'VINL2[SE]'
  ; Item #3 'VINL2[SE] + VINL1[SE]'
  ; Item #4 '{VIN1P, VIN1M}[DIFF]'
  : values=1
numid=24,iface=MIXER,name='ADC Mic Bias'
  ; type=ENUMERATED,access=rw------,values=1,items=3
  ; Item #0 'Mic Bias off'
  ; Item #1 'Mic Bias on'
  ; Item #2 'Mic Bias with Bypass Resistor'
  : values=1
numid=28,iface=MIXER,name='ADC Right Capture Source'
  ; type=ENUMERATED,access=rw------,values=1,items=19
  ; Item #0 'No Select'
  ; Item #1 'VINR1[SE]'
  ; Item #2 'VINR2[SE]'
  ; Item #3 'VINR2[SE] + VINR1[SE]'
  ; Item #4 'VINR3[SE]'
  ; Item #5 'VINR3[SE] + VINR1[SE]'
  ; Item #6 'VINR3[SE] + VINR2[SE]'
  ; Item #7 'VINR3[SE] + VINR2[SE] + VINR1[SE]'
  ; Item #8 'VINR4[SE]'
  ; Item #9 'VINR4[SE] + VINR1[SE]'
  ; Item #10 'VINR4[SE] + VINR2[SE]'
  ; Item #11 'VINR4[SE] + VINR2[SE] + VINR1[SE]'
  ; Item #12 'VINR4[SE] + VINR3[SE]'
  ; Item #13 'VINR4[SE] + VINR3[SE] + VINR1[SE]'
  ; Item #14 'VINR4[SE] + VINR3[SE] + VINR2[SE]'
  ; Item #15 'VINR4[SE] + VINR3[SE] + VINR2[SE] + VINR1[SE]'
  ; Item #16 '{VIN2P, VIN2M}[DIFF]'
  ; Item #17 '{VIN3P, VIN3M}[DIFF]'
  ; Item #18 '{VIN2P, VIN2M}[DIFF] + {VIN3P, VIN3M}[DIFF]'
  : values=1
numid=23,iface=MIXER,name='ADC Right Input'
  ; type=ENUMERATED,access=rw------,values=1,items=5
[... 309 lines cut ...]
  ; Item #1 '2 samples/update'
  ; Item #2 '4 samples/update'
  ; Item #3 'Immediate'
  : values=0
numid=13,iface=MIXER,name='Volume Ramp Down Step'
  ; type=ENUMERATED,access=rw------,values=1,items=4
  ; Item #0 '4dB/step'
  ; Item #1 '2dB/step'
  ; Item #2 '1dB/step'
  ; Item #3 '0.5dB/step'
  : values=2
numid=14,iface=MIXER,name='Volume Ramp Up Rate'
  ; type=ENUMERATED,access=rw------,values=1,items=4
  ; Item #0 '1 sample/update'
  ; Item #1 '2 samples/update'
  ; Item #2 '4 samples/update'
  ; Item #3 'Immediate'
  : values=0
numid=15,iface=MIXER,name='Volume Ramp Up Step'
  ; type=ENUMERATED,access=rw------,values=1,items=4
  ; Item #0 '4dB/step'
  ; Item #1 '2dB/step'
  ; Item #2 '1dB/step'
  ; Item #3 '0.5dB/step'
  : values=2

EE1977, 2024-10-28:
And the rest of the commands: 
journalctl -b 0 | grep pipedald
Oct 28 17:29:44 pi systemd[1]: Starting pipedald.service - PiPedal Web Service...
Oct 28 17:29:45 pi pipedald[919]: Document root: /etc/pipedal/config Threads: 5
Oct 28 17:29:45 pi pipedald[919]: Locale: en_GB
Oct 28 17:29:45 pi pipedald[919]: Updater: Next update check at 2024-10-28 21:34:10
Oct 28 17:29:45 pi pipedald[919]: Scanning for LV2 Plugins
Oct 28 17:29:57 pi systemd[1]: Started pipedald.service - PiPedal Web Service.
Oct 28 17:29:57 pi pipedald[919]: Found ALSA device hw:sndrpihifiberry.
Oct 28 17:29:58 pi pipedald[919]: Audio started. ALSA, hw:sndrpihifiberry, 16-bit little-endian, 48000, 64x4, in: 1/8, out: 2/8
Oct 28 17:29:59 pi pipedald[919]: TooB Convolution Reverb: Loading /var/pipedal/audio_uploads/ReverbImpulseFiles/St. Margaret's Church.wav
Oct 28 17:29:59 pi pipedald[919]: Listening on mDns address pi:80
Oct 28 17:29:59 pi pipedald[919]: Listening on eth0 address 192.168.1.8:80
Oct 28 17:29:59 pi pipedald[919]: Listening on Wi-Fi address 192.168.1.10:80
Oct 28 17:29:59 pi pipedald[919]: DNS/SD service announced. (pipedal)
Oct 28 17:29:59 pi pipedald[919]: TooB Convolution Reverb: File loaded. Sample rate: 44100 Length: 2s.
Oct 28 17:29:59 pi pipedald[919]: TooB Convolution Reverb: Max T: 1.57s Feedback: 0.000466
Oct 28 17:29:59 pi pipedald[919]: TooB Convolution Reverb: Load complete.
Oct 28 17:30:00 pi pipedald[919]: DNS/SD group established.
Oct 28 17:30:00 pi pipedald[919]: HotspotManager: Monitoring network status.
Oct 28 22:45:23 pi pipedald[919]: Web Socket Address: 127.0.0.1:80
Oct 28 22:45:24 pi pipedald[919]: WebSocketSession opened. ([::ffff:127.0.0.1]:55236)

journalctl -b 0 | grep alsa
Oct 28 17:29:42 pi kernel: alsactl[642]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set
Oct 28 17:29:42 pi kernel: alsactl[412]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set
Oct 28 17:29:43 pi systemd[1]: alsa-state.service - Manage Sound Card State (restore and store) was skipped because of an unmet condition check (ConditionPathExists=/etc/alsa/state-daemon.conf).
Oct 28 17:29:43 pi systemd[1]: Starting alsa-restore.service - Save/Restore Sound Card State...
Oct 28 17:29:43 pi alsactl[719]: alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
Oct 28 17:29:43 pi alsactl[719]: alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:1 use case configuration -2
Oct 28 17:29:43 pi alsactl[719]: alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:3 use case configuration -2
Oct 28 17:29:43 pi systemd[1]: Finished alsa-restore.service - Save/Restore Sound Card State.

EE1977, 2024-10-28:
aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 3: sndrpihifiberry [snd_rpi_hifiberry_dacplusadcpro], device 0: HiFiBerry DAC+ADC Pro HiFi multicodec-0 [HiFiBerry DAC+ADC Pro HiFi multicodec-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

EE1977, 2024-10-28:
Also I disabled pipedal, rebooted and checked with Audacity. The hat seems properly installed and appears as a stereo device with 2 channels.

rerdavies, 2024-10-29:
Ok. I've identified the problem. Fix coming...

rerdavies, 2024-10-29:
Could you please try PiPedal v1.3.63-Experimental, which contains a fix that I think should address your problem.

I have a nagging feeling that PiPedal has accidentally picked an SPDIF output sub-device instead of the audio output device. If that's the case, I need to do some additional work to properly fix the problem.Whether the fix works or not, could you also run the following commands for me: 

    pipedalconfig --alsa-devices

Identify your device in the list, and then run:

    pipedalconfig --alsa-device YOUR_DEVICE_ID

And post the results here. The device ID should be of the form "hw:M2" (or whatever the ID of your device is).

EE1977, 2024-10-29:
pipedalconfig --alsa-devices
=== ALSA Hardware Devices ===

Sound Card hw:vc4hdmi0
  Name: vc4-hdmi-0
  Driver: vc4-hdmi

  Subdevice: hw:0,0 - Unknown
  ID: MAI PCM i2s-hifi-0
  Capabilities: PLAYBACK
  (device in use)

Sound Card hw:vc4hdmi1
  Name: vc4-hdmi-1
  Driver: vc4-hdmi

  Subdevice: hw:1,0 - Unknown
  ID: MAI PCM i2s-hifi-0
  Capabilities: PLAYBACK
  (device in use)

Sound Card hw:sndrpihifiberry
  Name: snd_rpi_hifiberry_dacplusadcpro
  Driver: HifiberryDacpAd

  Subdevice: hw:2,0 - Unknown
  ID: HiFiBerry DAC+ADC Pro HiFi multicodec-0
  Capabilities: PLAYBACK CAPTURE
  (device in use)

pipedalconfig --alsa-device hw:sndrpihifiberry
Device: hw:sndrpihifiberry
Channel range: 2 to 8

Supported channel configurations:
  2 channels (Stereo)
  4 channels (Quadraphonic)
  6 channels (5.1 Surround)
  8 channels (7.1 Surround)

EE1977, 2024-10-29:
Yes, now it works!!! Thanks for your immediate response!!!

rerdavies, 2024-10-29:
Excellent! I think I probably have a remaining bug with devices that have SPDIF outputs (which you do not have). But I'll have to wait until one turns up in the field in order to see what they look like. And very curious that the 7.1 channel configuration sends audio output to nowhere. Or maybe it's the 7.1 input that's the problem (audio input on center channel which would be the 3rd input channel?) . Anyway. Fixed for all practical purpose.

For now, stick with the Experimental release, but releases going forward will include the fix, so you can upgrade fearlessly.

And thanks for your patience and support working through the issue.

closed 2024-10-29 by rerdavies

EE1977, 2024-10-29:
Thank you, Robin! I wonder if the issue is related to the chipset's internal capability to handle up to a 7.1 configuration, even though the HiFiBerry implementation does not utilize it. Anyway, problem solved!! Thank you for all your help!!