rerdavies/pipedal #242: Instrument selector for TooB Tuner

rerdavies, 2024-10-29:
See #167 

TooB Tuner should sub-sample MORE for basses in order to gain extra low-frequency precision. TooB Tuner should subsample less for violins, in order to gain extra high-frequency precision. 

Add an Instrument control to TooB Tuner , allowing selection of Bass/Guitar/Violin, or consider running a pair of pitch detectors to increase range. (Does Violin need an 8096 sample pitch detector?)

- Ask @EE1977 to test the violin implementation when done.
- Check to see whether violins need Concert->instrument pitch adjustments.
- Consider viola/cello/standup bass pitch ranges.

EE1977, 2024-10-29:
Robin, thank you for your efforts in addressing the issues related to bowed instruments. Pipedal truly creates a powerful FX processor for violins, and I assume it works wonders for other instruments as well! I have one question regarding the tuners, specifically the Toob and Gx. What is the exact purpose of the sensitivity control? I find it a bit confusing, as none of the hardware tuners I've used so far have this setting. Personally, I use the lowest possible setting every time at -60dB.

rerdavies, 2024-10-29:
It sets the level above which the input signal has to be in order to turn
the tuner on. Probably a more important feature for guitars, which are
basically antennae connected via 20 feet of unshielded unbalanced cable to
a terrifying 1950's electrical device running at 600 volts that is--as
often as not -- improperly grounded.

Guitar input signals tend to have significant ground hum at 50 or 60 hz and
small signal to noise ratios, so its intended use is to prevent the tuner
from thrashing around or locking on to ground hum when you're not playing.

I imagine that violin electronics are more sophisticated. And I imagine
hardware tuners thrash around when guitarists use them in high-noise
environments, but they're less distracting than a big red needle on your
phone or tablet.

---------- Forwarded message ---------

#242)