rerdavies/pipedal #244: Audio started 16 bit in ASLA

BorisSutin, 2024-10-29:
Audio started. ALSA, hw:audioinjectoris, 16-bit little-endian, 48000, 32x4, in: 1/2, out: 2/2 

 expected 24 bit/audio

rerdavies, 2024-10-29:
Huh. Are you running 1.3.63 now? (I'm guessing you are, given that your I/O is now 1/2, 2/2 instead of 1/8, 2/8. 

Could you run the following commands for me and send me the results (both sets please):

    pipedalconfig --alsa-devices

Find your device in the list, and then run 

    pipedalconfig --alsa-device <DEVICE-ID-OF_YOUR DEVICE-FROM-STEP-1>

<note to self: make this a one-step process>

BorisSutin, 2024-10-29:
I use 1.3.62-Beta , you have already released 63, I don't see a new release

Сютин Борис
***@***.***

EE1977, 2024-10-29:
I also noticed 16bit instead of 24 on my hifiberry...

rerdavies, 2024-10-29:
Could I ask you both to try the PiPedal v1.3.64-Experimental release to see if it corrects the problem. It will, but I'd like to know that 24-bit formats work properly before I promote this to Beta status.  You may actually get 32-bit instead of 24-bit formats if your devices support it.  That's as-designed.

Available here:

    https://github.com/rerdavies/pipedal/releases/tag/v1.3.64

or through in-app update if you select **_Settings_** /  **_Check for Updates_** and choose the **_Development_** update channel.

BorisSutin, 2024-10-29:
Audio started. ALSA, hw:GtrBox, 32-bit integer little-endian, 48000, 32x4, in: 1/2, out: 2/2 

Thanks Robin. This solved the problem, according to the logs I see that it starts in 32 bit and this is good. And by the way, 8 channels are missing, now, as planned, 2 channels are displayed. I can't check the sound yet , it's already late, but I'll definitely play guitar tomorrow) Thank you again for your prompt response!

rerdavies, 2024-10-29:
@BorisSutin: Out of curiosity, what *IS* a GtrBox? I wasn't able to find any info the device online.

rerdavies, 2024-10-29:
Oops. Didn't mean to close the issue until the fix gets promoted to at least beta status.

rerdavies, 2024-10-29:
Fixed in v1.3.64 experimental (contains two fixes, both of which have been field tested and confirmed as working). Will be re-issued in a Beta release shortly.

BorisSutin, 2024-10-30:
> @BorisSutin: Out of curiosity, what _IS_ a GtrBox? I wasn't able to find any info the device online.

This is a codec based on adau1761, I rebuilt the kernel to include a new driver, and in the overlay I wrote the name GtrBox as in my project )

The board is designed for a guitar project, has a guitar HiZ input and  microphone input

BorisSutin, 2024-10-30:
Robin, two things are troubling me. I don't hear any difference in sound between the versions of(1.3.62 / 64), with 24 bit the sound should be more detailed.  And secondly, the adau1761 codec does not support 32 bit, the maximum depth is 24 bit, as it is written in the datasheet.  And the pipedal log says Audio started. ALSA, hw:GtrBox, 32-bit integer little-endian, i just wondered for a second if that was really the case

rerdavies, 2024-10-30:
Boris: That's interesting, because it may give you a unique perspective on what's going on. 

Here's what I know:

I know from EE1977's previous case that their device was reporting 8 channels on Raspberry Pi 5, and only 2 channels on Raspberry Pi 4 (same version of Raspberry Pi OS). The 8-channel configuration didn't work! The fix was to modify PiPedal to prefer a 2-channel configuration when the device reported that it supported 2 to 8 channels.  Your device was also reporting that it accepted 8 channels as well. PiPedal used to take the maximum number of channels offered because it would allow you to (for instance) route signals onto the 4th and 5th channels of a device with  8 physical inputs and outputs.

What Pipedal was told about your device: that it would NOT accept float32 data, but that it was perfectly happy to accept 32-bit data.  PiPedal didn't try 24-bit formats, because there it's possible (but unlikely) that there are commercially available ADCs out there that actually do have more than 24 bits of accuracy.

My expectation would be that the lower bits of that 32-bits of data would be ignored, since I don't think there's a DAC on the planet that's actually accurate to 32-bits of data. TBH I wouldn't expect your device to ACTUALLY have a 24-bit ADC/DAC. either It most likely has an 18- to 22-bit DAC, unless the actually chip you selected has freakishly good precision. It's also probable that your ADC has a bit or two less precision than the DAC does. fwiw datasheets that say "103db A-weighted SN/R" are actually 16-bit ADCs., whereas a device with actual 18-bit DACs would report "115db A-weighted SN/R". (No idea what mathematical trickery is involved in that oddity).

There is different between Raspberry Pi 4 and Raspberry Pi 5 hardware when it comes to I2C audio chips. The Raspberry Pi 5, apparently, has a newer, better, more capable I2C hardware. What that actually means, I don't know. I think it's faster, among other things, which would perhaps explain the discrepancy. Devices that could not support 8 channels on the rpi4 I2C stack because of bandwidth limitations, could PERHAPS now offer 7.1 audio formats that can be support on rpi 5. And MAY, because the rpi5 I2C hardware supports more bandwidth, be more inclined to offer 32-bit streams when rpi 4 devices might have offered only 24-bit streams. Also known: there was a ragged trail of bugs that appeared in the 10-2023 timeframe pertaining to I2C audio chips and audio formats that seemed to be related to something introduced in the 6.10 linux kernel.

So there are a few possibilities:

- Your device does actually support a 32-bit datastream, and uses only the most significant 24 (22?) bits.
- Your device supports both 32-bit and 24-bit datastreams, but in reality only has an ADC with 16-bit signal to noise ratio.
- Your device does not support a 32-bit data stream, but something in the Raspberry P  5's I2C audio driver stack has volunteered to do the conversion, and perhaps does it incorrectly.
- your device does actually support a 32-bit datastream but its driver has problems with capabilities being offered by Linux 6.10+ kernels on Raspberry 5 hardware (like for example, doing the 32 to 24 bit conversion wrong).

PiPedal currently prefers 32-bit formats to 24-bit formats because it's theoretically possible (but practically impossible) that one day there will be an actual ADC/DAC that supports more than 24 bits. I don't think there are actually any commercially available at a reasonable price ADC/DACs that support more than 24 bits of precision.  And -- as you probably know if you have designed this board yourself -- even if you do have an ADC that has more than 16-bits of precision, it is fiendishly difficult to implement a board layout that doesn't leak noise onto the actual physical traces of the board so that the analog signal delivered to the has a signal to noise ratio that is greater than 16 equivalent bits. (96db linear, but typically reported as 105db A-Weighted on data sheets).  Devices with ADCs that actually do have more than 16-bits of precision would specify that they have 113db A-weiighted SN/R, or better, I think). To do this right usually requires (1) extra ground planes in the board itself in order to protect the analog signal routes; (3) careful layout of audio traces -- much wider than normal, with minimum-bending-radius constrains, and (3) larger capacitors than are convenient to place on a Raspberry Pi HAT in order to remove noise from the power supply lines. (Based on my possibly out-of-date engineering experiences with audio devices).

To put this in perspective, I have a cheap M-AUDIO audio adapter that accepts 24-bit data streams, claims to have 105db A-Weight SN/R, which means that it is a 16-bit ADC, but in fact only delivers 12(!) bits of signal above its noise floor.  Most likely inadequate filtering of noisy power supply rails. The noise floor is most definitely noise from CPU and memory components causing power rails to swing.

So I have to ask: how CERTAIN are you that you're only getting 16-bit precision. And are you certain that you should be getting more? I agree that the difference is audible. Higher bit depths have a noticeable "transparency" or "clarity" than 16-bit signals do, but it is a subtle effect. And it relies on you having trimmed input and output signal levels very carefully.  

[... cut ...]  lot of the Gx gen-1 amp models that use matrix-based circuit simulation techniques show dramatic improvements with high-bit-depth inputs while NAM models (to my ears) do not. If you have amp models in your signal chain when doing your lsitening tests, all bets are off, I think.

Let me know what you think. If you're certain that the wrong thing is being done, I can offer you an experimental build that choose 24-bit samples ahead of 32-bit samples, and we can see whether that makes a difference.

rerdavies, 2024-10-30:
I pulled the datasheet on the adua1761. The datasheet says:

    24-bit stereo audio ADC and DAC: >98 dB SNR

Which would be (I think) 16-bits of ADC precision plus a little bit for dithering. A 24-bit audio stream format, with only 16.5 bits of valid data above the noise floor.

Let me know how you want to proceed.

rerdavies, 2024-10-30:
ARGH!!! One of the horrifying things about doing audio engineering work is that sometimes theory gives you insight into what you're actually hearing. And once you know what you're listening for, you can HEAR the difference. And once you can hear the difference, you can never unhear it. :-P

What I'm hearing now: I think one of the NAM models I've been using for testing seems to have been trained on data from cheap consumer ADCs that are delivering only 12 bits of digital precision. :-P

BorisSutin, 2024-10-30:
The codec and ADC are definitely 24-bit, I have been using Analog Device for several years now and know the entire ADAU line well. The codec acts as a Master, and the Raspberry is a Slave. In my SigmaStudio project, everything is strictly configured to use 24/48000, with other parameters the codec simply cannot start and work, the sound would immediately break. So it seems to me that all the transformations occur in ALSA itself. In Pi5, I made changes to the i2s bus, but it's a pity that this was poorly documented.
For the sake of interest, I will find my old Modep image today, launch it and compare the sound with PiPedal ) I think it will be the same in quality, although PIPedal significantly wins in speed!)  

I don't know how you specify device initialization in ALSA. You can simply add the option to choose between S16_LE S24_LE in the settings, then it will be easier to understand whether this affects the sound.

rerdavies, 2024-10-30:
I don't think you should be confusing stream configuration with actual ADC/DAC sizes.  Devices with 16-bit (or smaller) pretty much universally accept 24-bit stream formats even if their ADCs have less than 16 bits of precision. It's a shameless marketing trick that most consumers don't pick up on. The actual datasheet parameter you need to look for is the SN/R spec, which gives you a hint as what the precision of the underlying ADC is. SN/R measurements are almost always give in A-Weighted form (which is actually a useful measurement, but superficially makes it look like devices have more ADC bits than they do.  Or, if they are 16-bit devices don't bother reporting the SN/R at all. Also common, and a sure indicator that the device has a 16-bit ADC. Your datasheet falls into a bizarre intermediate category, where they do provide an SN/R ration, but don't say whether it is linear or A-Weighted. I suspect the latter. > 98db is an odd number: 16 and 2/3 bits of SN/R. Hard to say whether they have a 17th bit of signal that is not completely reliable, or whether they are representing dithering as a 17th bit (which should more properly yield a figure of <= 97.5db). Or whether they have given you an A-weight SN/R that actually reflects an ADC with only 14 bits of linear precision. It would not be unusual for a device of this class and purpose to have 14 bits of precision, which is ENTIRELY adequate for voice data on a phone.  I hate to be the bearer of bad news on this one, but I genuinely think you have a device with a 24-bit data stream, and a 16-bit ADC. With possibly a fraction of a 17th bit that isn't going to be truly useful (improperly representing averaged downsampling 16-bits as more than 16 bits, maybe? )

There are, in truth, only a small handful of appropriate ADC/DACs suitable for use in audio applications.  I have TWO usb adapters that use the TI Burr-Brown DAC (universally used on cheap consumer USB adapters. There is no other real choice).  The TI/Burr Brown datasheets are unequivocally plain that the ADCs are 16-bit. But spec sheets for both USB devices conceal this. One doesn't specify an SN/R at all; and the other quotes an a-Weighted SN/R ratio, which it doesn't remotely achieve because of inadequate power-rail filtering. Burr Brown reference implementations use four large 3/4" pot capacitors to filter power rails. You sometimes see them on DAC hats. I suspect they are not truly optional. My M2 device which is superb, uses two separate chips -- one for ADC, another for DAC. The ADC chip is seems to have been designed for use in laboratory test equipment, and is awkward to integrate if you are using it for pro-audio purposes. There doesn't seem to be an off-the-shelf single-chip ADC/DAC that provides more than 16-bits of ADC that I was able to find.  So I suspect that other pro-audio adapters use a similar two-chip approach. Research for a Pipedal hardware device incarnation, not currently active, which was abandoned PRIMARILY because the engineering and design challenges of doing a pro-audio adapter properly require deep (and rare) audio engineering design expertise, and because FCC and EU certifications cost for any such devices  make the whole project financially infeasible, in case you were wondering. I may revisit this with a potential GoFundMe campaign once PiPedal momentum picks up.  It's almost definitely infeasible using traditional funding channels.

A plausible way to test this is to use TooB Spectrometer (which I hope still works). That was how I discovered that my cheap USB adapters were unspeakable hunks of poo. You should be able to see the noise floor clearly on the spectrometer output as a shelf of noisy spikes some 80 or 90 or 100 db down from the maximum possible signal. You need to input a pure sine-wave signal at max trim to find the top of range of the signal, unfortunately, which is a bit awkward. But you do learn a lot in the process. I did, anyway.

On to your question...

Specifying the device configuration is a very bizarre process in ALSA. The basic premise is that the host starts with an empty list of conditions, and then adds additional conditions one by one. This is -- I think -- supposed to accommodate devices that will do (for instance) 24-bit formats in 2 channel configurations, but only 16-bit formats in 8-channel configuration, due to bandwidth constraints on an underlying USB or I2C or Firewire buses. ALSA (the underlying driver?) is free to reject constraints at any point along the configuration path.  Devices that rely on this are mostly historical at this point. Firewire 400 is probably extinct, and Firewire 800 doesn't have the problem. And USB 1.0 audio devices probably do exist, but neither of us would consider them fit for purpose. I rather suspect that I2C on rpi4 has bandwidth limitations, which is why it doesn't offer hardware-downmixed 7.1 formats as well that rPi 5 does not. (And I rather suspect that problems that EE had were because the order of channels on a 7.1 devices is actually Front-left, **_Front-Center_**, Front-Right .... so their input signal was on the 3rd channel instead of the second. If you happen to stumble upon code that indicates the order of 7.1 channels let me know.

PiPedal currently sets configuration parameters in the following order. At each step it adds the request to cumulative hw parameters, and either fails or accept device-supplied adjustments. The device is allowed to say "yes I accept your request for 4800 [... cut ...]
Channel configurations are actually queried in a separate step, so that the UI can be updated to reflect the actual number of device channels and frequencies and buffer sizes ahead of time. The query code also selects the best sample rate before the best channel size. PiPedal's code knows the channel configuration before it starts opening the device for actual use, but will (incorrectly) succeeds if the final channel configuration doesn't match what was expected. But any change to the number of channels would show up in the log message. The rest of the channel info (which channels are used) are mixing configuration that don't affect how the device is opened.

BorisSutin, 2024-10-30:
Robin, thank you for the detailed answer , I heard you) We will consider the topic closed. A little later, I would like to talk to you about the implementation of HOST USB Audio , I think you can easily add such a function. True, I tested UAC2Gadget on Raspberry, it works terribly, long latency and a lot of artifacts in the sound. They say that on Linux you need to use the Tiny USB library , it is more stable and devoid of disadvantages.

rerdavies, 2024-10-30:
Sure. I'd be interested in any info you have on the subject.

Things you should know:

   - There is no reason why you should NOT be getting exceptionally good
   latency from USB audio devices as long as you are using a reasonably modern
   (<8-ish year old?) device. I have three three test USB adaptors and one USB
   pedalbaord, all of which run stably at 32/4, and 16/4 (not extensively
   tested). Any USB Audio 1.1 device should work fine.
   - I think UAC2Gadget is an ancient USB Audio 1.0 device. It probably is
   not going to work well. And it probably does have horrible latency because
   it's a USB 1.0 device.
   - I *think* you will be able to see problems with USB devices that have
   bad buffer sizes in the UI. If you have a device that is offering you
   32xi8/32x10//32/12 or similar configurations, you're dealing with an old
   device that has an unreasonably large minimum buffer size. Reasonably
   modern devices (even cheap ones) should be offering you 32x2/32/3/32x4
   configurations.I have one field report of such a device (an old device)
   which had a minimum buffer size of 256, which was eventually coaxed in
   to running with OK/not great latency.
   - There were HUGE improvements made to USB audio in Linux 5.15. Prior to
   Linux 5.15, USB pro audio was a complete disaster. So TinyUsb might not be
   neccesary anymore.

BorisSutin, 2024-10-30:
I had in mind the organization of recording via Raspberry, so that it would act as an audio interface . So that you can capture audio from PiPedal and send it to the PC and return it back to PiPedal from the PC. Then you would get a real guitar processor with USB audio. There was an audio-bridge plugin on Modep https://github.com/falkTX/audio-bridge.

Unfortunately, it doesn't work in PiPedal because it doesn't have any incoming ports (this is also a topic that needs to be improved in PiPedal) There is for example a player for audio files , but it is not available so you do not have incoming ports, and it is very cool to play under backtracks, I hope you will implement this in the future.

closed 2025-02-21 by rerdavies