rerdavies/pipedal #250: [BUG+FIX] Compile error on Ubuntu 24.10 (x86_64)

TheMojoMan, 2024-11-08:
`themojoman@Acer:~/Develop/pipedal$ ./mk.sh 
[  1%] Building CXX object PiPedalCommon/src/CMakeFiles/PiPedalCommon.dir/NetworkManagerInterfaces.cpp.o
[  1%] Building CXX object PiPedalCommon/src/CMakeFiles/PiPedalCommon.dir/Lv2Log.cpp.o
[  2%] Building CXX object PiPedalCommon/src/CMakeFiles/PiPedalCommon.dir/DBusEvent.cpp.o
[  2%] Building CXX object PiPedalCommon/src/CMakeFiles/PiPedalCommon.dir/DBusVariantHelper.cpp.o
[  3%] Building CXX object PiPedalCommon/src/CMakeFiles/PiPedalCommon.dir/DBusLog.cpp.o
[  4%] Building CXX object PiPedalCommon/src/CMakeFiles/PiPedalCommon.dir/DBusDispatcher.cpp.o
[  4%] Building CXX object PiPedalCommon/src/CMakeFiles/PiPedalCommon.dir/ofstream_synced.cpp.o
[  5%] Building CXX object PiPedalCommon/src/CMakeFiles/PiPedalCommon.dir/ConfigSerializer.cpp.o
[  5%] Building CXX object PiPedalCommon/src/CMakeFiles/PiPedalCommon.dir/WifiRegs.cpp.o
[  7%] Building CXX object PiPedalCommon/src/CMakeFiles/PiPedalCommon.dir/WifiRegulations.cpp.o
[  8%] Building CXX object PiPedalCommon/src/CMakeFiles/PiPedalCommon.dir/WifiDirectConfigSettings.cpp.o
[  8%] Building CXX object PiPedalCommon/src/CMakeFiles/PiPedalCommon.dir/WifiConfigSettings.cpp.o
[  9%] Building CXX object PiPedalCommon/src/CMakeFiles/PiPedalCommon.dir/ServiceConfiguration.cpp.o
[  9%] Building CXX object PiPedalCommon/src/CMakeFiles/PiPedalCommon.dir/WriteTemplateFile.cpp.o
[ 10%] Building CXX object PiPedalCommon/src/CMakeFiles/PiPedalCommon.dir/ChannelInfo.cpp.o
[ 11%] Building CXX object PiPedalCommon/src/CMakeFiles/PiPedalCommon.dir/HtmlHelper.cpp.o
In file included from /home/themojoman/Develop/pipedal/PiPedalCommon/src/HtmlHelper.cpp:20:
/home/themojoman/Develop/pipedal/PiPedalCommon/src/include/HtmlHelper.hpp:46:55: error: ‘uint32_t’ has not been declared
   46 |     static void utf32_to_utf8_stream(std::ostream &s, uint32_t uc);
`

FIX:
Add `#include <cstdint>` to pipedal/PiPedalCommon/src/HtmlHelper.cpp and pipedal/PiPedalCommon/src/include/HtmlHelper.hpp.

TheMojoMan, 2024-11-08:
Thanks for your reply!
I will do the testing tomorrow (it is time to go to bed here where I live).
I did the compile on amd64 but I can also do the compilation on arm64 (on my Xiaomi Pad 5 which runs Ubuntu 24.10 on a 6.1.10 Kernel).

TheMojoMan, 2024-11-09:
I have started testing PiPedal on my Acer TravelMate P215-54 (Intel Core i5-1235U) using Ubuntu 24.10 this morning  and ran into the following problems:

- My build of PiPedal only installs CabSim.ttl into /usr/lib/lv2/ToobAmp.lv2/. All other Toob plugins are missing. So I downloaded the amd64 .deb package of the plugins from your github page but cannot install it because Ubuntu 24.10 comes with libflac++10 and libboost-iostreams1.83.0:
```shell
themojoman@Acer:~/Downloads$ LANG=en sudo dpkg -i ./toobamp_1.0.17_amd64.deb
(Reading database ... 252571 files and directories currently installed.)
Preparing to unpack ./toobamp_1.0.17_amd64.deb ...
Unpacking toobamp (1.0.17) over (1.0.17) ...
dpkg: dependency problems prevent configuration of toobamp:
 toobamp depends on libboost-iostreams1.74.0 (>= 1.74.0); however:
  Package libboost-iostreams1.74.0 is not installed.
 toobamp depends on libflac++6v5 (>= 1.3.1); however:
  Package libflac++6v5 is not installed.

dpkg: error processing package toobamp (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 toobamp

themojoman@Acer:~/Downloads$ LANG=en sudo apt search libflac++
flac/oracular 1.4.3+ds-2.1ubuntu2 amd64
  Free Lossless Audio Codec - command line tools

libflac++-dev/oracular,now 1.4.3+ds-2.1ubuntu2 amd64 [installed]
  Free Lossless Audio Codec - C++ development library

libflac++10/oracular,now 1.4.3+ds-2.1ubuntu2 amd64 [installed,automatic]
  Free Lossless Audio Codec - C++ runtime library

libflac-dev/oracular,now 1.4.3+ds-2.1ubuntu2 amd64 [installed,automatic]
  Free Lossless Audio Codec - C development library

libflac-doc/oracular,oracular 1.4.3+ds-2.1ubuntu2 all
  Free Lossless Audio Codec - library documentation

libflac-ocaml/oracular 0.5.0-1build3 amd64
  OCaml interface to the flac library -- runtime files

libflac-ocaml-dev/oracular 0.5.0-1build3 amd64
  OCaml interface to the flac library -- development files

libflac12t64/oracular,now 1.4.3+ds-2.1ubuntu2 amd64 [installed,automatic]
  Free Lossless Audio Codec - runtime C library

themojoman@Acer:~/Downloads$ LANG=en sudo apt search libboost-iostreams
libboost-iostreams-dev/oracular,now 1.83.0.2ubuntu1 amd64 [installed]
  Boost.Iostreams Library development files (default version)

libboost-iostreams1.83-dev/oracular,now 1.83.0-3.2ubuntu2 amd64 [installed,automatic]
  Boost.Iostreams Library development files

libboost-iostreams1.83.0/oracular,now 1.83.0-3.2ubuntu2 amd64 [installed,automatic]
  Boost.Iostreams Library
``` 
- Compiling ToobAmp plugins failed:
```shell
themojoman@Acer:~/Develop/ToobAmp$ LANG=en ./build.sh 
/usr/bin/cmake -S/home/themojoman/Develop/ToobAmp -B/home/themojoman/Develop/ToobAmp/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/themojoman/Develop/ToobAmp/build/CMakeFiles /home/themojoman/Develop/ToobAmp/build//CMakeFiles/progress.marks
/usr/bin/gmake  -f CMakeFiles/Makefile2 all
gmake[1]: Entering directory '/home/themojoman/Develop/ToobAmp/build'
/usr/bin/gmake  -f modules/RTNeural/RTNeural/CMakeFiles/RTNeural.dir/build.make modules/RTNeural/RTNeural/CMakeFiles/RTNeural.dir/depend
gmake[2]: Entering directory '/home/themojoman/Develop/ToobAmp/build'
cd /home/themojoman/Develop/ToobAmp/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/themojoman/Develop/ToobAmp /home/themojoman/Develop/ToobAmp/modules/RTNeural/RTNeural /home/themojoman/Develop/ToobAmp/build /home/themojoman/Develop/ToobAmp/build/modules/RTNeural/RTNeural /home/themojoman/Develop/ToobAmp/build/modules/RTNeural/RTNeural/CMakeFiles/RTNeural.dir/DependInfo.cmake "--color="
gmake[2]: Leaving directory '/home/themojoman/Develop/ToobAmp/build'
/usr/bin/gmake  -f modules/RTNeural/RTNeural/CMakeFiles/RTNeural.dir/build.make modules/RTNeural/RTNeural/CMakeFiles/RTNeural.dir/build
gmake[2]: Entering directory '/home/themojoman/Develop/ToobAmp/build'
gmake[2]: Nothing to be done for 'modules/RTNeural/RTNeural/CMakeFiles/RTNeural.dir/build'.
gmake[2]: Leaving directory '/home/themojoman/Develop/ToobAmp/build'
[  1%] Built target RTNeural
[... 13 lines cut ...]
/usr/bin/gmake  -f modules/lv2cairo/src/lv2c/CMakeFiles/lv2c.dir/build.make modules/lv2cairo/src/lv2c/CMakeFiles/lv2c.dir/build
gmake[2]: Entering directory '/home/themojoman/Develop/ToobAmp/build'
[  4%] Building CXX object modules/lv2cairo/src/lv2c/CMakeFiles/lv2c.dir/Lv2cTableElement.cpp.o
cd /home/themojoman/Develop/ToobAmp/build/modules/lv2cairo/src/lv2c && /usr/bin/c++  -I/home/themojoman/Develop/ToobAmp/modules/lv2cairo/src/lv2c/include -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/sysprof-6 -I/usr/include/harfbuzz -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/fribidi -I/usr/include/pixman-1 -I/usr/include/librsvg-2.0 -I/usr/include/gdk-pixbuf-2.0 -fPIC -O3 -DNDEBUG  -Wno-psabi -Werror -Wall -pedantic -Wno-reorder -Wno-restrict -std=gnu++20 -MD -MT modules/lv2cairo/src/lv2c/CMakeFiles/lv2c.dir/Lv2cTableElement.cpp.o -MF CMakeFiles/lv2c.dir/Lv2cTableElement.cpp.o.d -o CMakeFiles/lv2c.dir/Lv2cTableElement.cpp.o -c /home/themojoman/Develop/ToobAmp/modules/lv2cairo/src/lv2c/Lv2cTableElement.cpp
In file included from /home/themojoman/Develop/ToobAmp/modules/lv2cairo/src/lv2c/include/lv2c/Lv2cTableElement.hpp:22,
[... cut ...]
      |                          ^~~~~
cc1plus: all warnings being treated as errors
gmake[2]: *** [modules/lv2cairo/src/lv2c/CMakeFiles/lv2c.dir/build.make:93: modules/lv2cairo/src/lv2c/CMakeFiles/lv2c.dir/Lv2cTableElement.cpp.o] Error 1
gmake[2]: Leaving directory '/home/themojoman/Develop/ToobAmp/build'
gmake[1]: *** [CMakeFiles/Makefile2:1159: modules/lv2cairo/src/lv2c/CMakeFiles/lv2c.dir/all] Error 2
gmake[1]: Leaving directory '/home/themojoman/Develop/ToobAmp/build'
gmake: *** [Makefile:169: all] Error 2
```

TheMojoMan, 2024-11-09:
> Ubuntu apt install dependencies should be pretty darned close.

Installing the dependencies on Ubuntu 24.10 following your build guide worked perfectly. The only thing missing was `npm`.

rerdavies, 2024-11-10:
I'm going to have to do this step by step. First step: fix the ToobAmp build for Ubuntu.  Supporting x64 in the pipedal build is downright messy; so let's figure out whether this is going to work at all before proceeding. 

So, here's what I've done with the ToobAmp build. I'd be grateful if you could test and confirm that the fixes work. 

1.  Problematic dependencies have been removed. The flac libraries are now statically linked (have been for some time, but that change hasn't been pushed into the ubuntu .deb package.  The boost-dependent code has been removed entirely. 
2.  The build has been updated to ignore the warning you're getting. I don't get the warning on arm64 builds, or Ubuntu 22.04 builds. I'm relatively indifferent as to whether the fix sticks for you on an Ubuntu 22.10 build, so you can test it or not, as you see fit. Changes have been committed to the main branch.
3.  Whenever changes are posted to github, there is a github action that does a test build of ToobAmp using Ubuntu 22.04 LTS amd64 already. I have modified the build to publish a .deb package for Ubuntu 22.04 amd64 as an artifact. Github doesn't provide an Ubuntu 22.10 builder, but the 22.04 LTS build should work fine on 22.10.
4.  If this all works, I will start publishing the amd64 artifact whenever I publish an arm64 build. 

The same is true of the arm64 build, which should also install fine on Ubuntu 22.04 LTS arm64 without problems, since Ubuntu 22.04 and Raspberry Pi OS both derive from Debian 12.

Debian packages have been posted here:

-    https://github.com/rerdavies/ToobAmp/releases/tag/v1.1.53

I will do appropriate testing on Ubuntu 22.04 x64/amd64, but I would be grateful if you could confirm that the .deb packages will work for you on Ubuntu 22.10 (either arm64 or amd64, or both if that's convenient). 

The PiPedal build has not yet been updated to support Ubuntu. If you have an otherwise working build of Pipedal, you can test PiPedal by 

1.  Building and installing Pipedal with the fixes you have used so far.
2.  Install using ./install.sh - This will install arm64 binaries for Pipedal WITH the problematic dependencies so...
3. ... fix that by installing the appropriate ToobAmp .deb package listed above on top of the PiPedal install.

As an INTERIM measure only, after installing ToobAmp, you should run 

     cd src/pipedal
     ./provisionLv2

which will copy the installed binaries for ToobAmp from /usr/lib/liv2 into pipedal source directories, overwriting the existing arm64 binaries, potentially with am64 binaries.  (ToobAmp can't be built as a submodule, because the build runs out of memory on my 8GB raspberry pi, so that's the procedure that's used for the current aarm64 build).

If all of this works, I will deal with the (rather unpleasant) build steps required to deal with aarch64/amd64 Pipedal builds.

Let me know how it goes.

TheMojoMan, 2024-11-11:
Thanks for your reply and great work!

-  toobamp_1.1.53_amd64.deb now installs without problem on my machine which runs Ubuntu 24.10 (not 22.10)
- Plugins load and work as expected when using Reaper, especially 'Toob Neural Amp Modeler'!
- My build of PiPedal works and loads the plugins as expected. All of the shipped presets function as expected.

Remaining problems:
- Cannot load .nam files when using PiPedal & Toob NAM (but works well with Reaper). Error: "Upload failed. Mailformed request". Tested with Firefox.
- Tuner does not work too well. I need to dial back input level on my interface for not getting buzzes when using the presets. But for tuner to function better I need to increase input up to a level where it clearly overdrives.
- Compiling toobamp plugins myself still fails ('all warnings being treated as errors' seems to be too restricting for successful compilation especially because the error occurs in a submodule):
```shell
themojoman@Acer:~/Develop/ToobAmp$ LANG=en ./build.sh 
[...]
[ 16%] Building CXX object modules/lv2cairo/src/lv2c/CMakeFiles/lv2c.dir/Lv2cSvg.cpp.o
In file included from /home/themojoman/Develop/ToobAmp/modules/lv2cairo/src/lv2c/Lv2cSvg.cpp:23:
/usr/include/librsvg-2.0/librsvg/rsvg.h:1331:1: error: this use of "defined" may not be portable [-Werror=expansion-to-defined]
 1331 | #if LIBRSVG_CHECK_FEATURE(PIXBUF)
      | ^~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
gmake[2]: *** [modules/lv2cairo/src/lv2c/CMakeFiles/lv2c.dir/build.make:636: modules/lv2cairo/src/lv2c/CMakeFiles/lv2c.dir/Lv2cSvg.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1156: modules/lv2cairo/src/lv2c/CMakeFiles/lv2c.dir/all] Error 2
gmake: *** [Makefile:166: all] Error 2
``` 

TODO:
- I did not test Wifi hotspot, yet.
- I did not test Midi, yet.

TheMojoMan, 2024-11-11:
Some more testing:
- MIDI is working. Great!
- JAZZ preset indicates that 'Toob Power Stage' is missing. It is also not shown in the list that shows up after pressing LOAD button. However, there is a file `PowerStage2l.ttl` in folder `/usr/lib/lv2/ToobAmp.lv2/`.

TheMojoMan, 2024-11-11:
When connected to the same Wifi network:
- PiPedal Android app works well.

Using the hotspot functionality:
- After enabling Wifi hotspot in PiPedal Ubuntu shows the hotspot icon in the top bar (on Gnome).
- pipedal shows up in the list of WLAN devices when using my phone or tablet.
- When I tap on pipedal it displays the dialog to enter the password.
- After entering the password pipedal is added as a saved network but there is no connection.
- Taping on pipedal in the list of saved networks does not connect to it.

rerdavies, 2024-11-23:
Just to keep you updated.... There seem to be enough differences that I need to do actual testing on actual Ubuntu installs.

The following fixes have been applied and published on main branches of PiPedal and ToobAmp.  

- _Cannot load .nam files when using PiPedal & Toob NAM (but works well with Reaper). Error: "Upload failed. Mailformed request". Tested with Firefox._

  Fixed.

- _Compiling toobamp plugins myself still fails ('all warnings being treated as errors' seems to be too restricting for successful compilation especially because the error occurs in a submodule):_

    -WError has been removed from the build for the ToobAmp submodule. (The warning is still given).

    PiPedal and ToobAmp both compile cleanly (with warnings in the submodule) on Ubuntut 24.04/GCC 13. 

-  Pipedal installation of  x64 toob amp plugins

     Added a build and install procedure to PiPedal for x64 ToobAmp plugins;  but I think I have not updated the TooBAmp amd64 .deb package in the PiPedal source tree with the latest ToobAmp amd64 .deb package.

-  x64 Plugins failing to load. 

    Interestingly, a buglet in GCC code generation relating to handling of  +/- inf, on x64. Fixed in ToobAmp source. The fix doesn't seem to have bee propagated to the PiPedal x64 source tree. Probably because the x64 .deb package hasn't been updated. 

======

Not yet fixed:

-   PAM adjustments (rt priority for pipedal_d, NetworkManager permissions for pipedal_d) seem to be applied correctly, but don't take effect until the system is rebooted. PAM does get poked to reload it's configuration in the .deb install, which works fine on Raspberry Pi OS, but doesn't seem to work on Ubuntu. Need to investigate. Somewhat concerning: default installs of Ubuntu on a VM, and default installs on a live machine behave differently.

-   Configuration for realtime audio on Ubuntu.

    Not terrible, but not as good as it should be. Probably needs Threaded IRQ kernel parameters for Ubuntu. Maybe also needs  preempt=full kernel argument, and one other (can't remember what it is). Need to investigate differences between governor settings on VMs and live Ubuntu instances. I think the governror is disabled when running in a VM.

   If you have any checkists for what needs to be fixed on a default Ubuntu install, I'd be interested.  I think I have most of the good stuff. 

    - Forcing Governor settings.
    - Select PREEMPT=full kernel, with threaded IRQs, and one other kernel argument relating to low latency (can't remember what it is)
    - Realtime scheduling works! (bit behavior seems to differ between VMs and live instances)
    

- ...hotspots...

    Not fixed. Possibly because network names ("eth0", "wlan0") don't match Ubuntu conventions.  There is a risk at
    this point that the hotspot feature may be removed in Ubuntu builds, although I will look for trivial fixes.

I currently have a live  Ubuntu x64 system (VMs weren't good enough). I'm currently debugging through the issues. 

- CPU temperature

    Seems to be different on Ubuntu. I'm not sure what temperature I'm picking up; but it's not the right temperature.

I will keep you updated on progress.

In summary:

- Builds should work. .deb packages won't. 
- Performance issues need work. Latency is ok. You can probably decrease latency by applying well-known configuration adjustments manually. (kernel parameters, principally). If you are using 24.x Ubuntu Studio, these fixes will have been applied already.
- monitoring needs some tweaking.

rerdavies, 2024-11-23:
Oh. If you're going to try building sources, you need to do the following:

In the toobamp build, you will need to update the submodules. Pull the latest ToobAmp sources, and then, from the toobamp project directory:

     git submodule update --init --recursive

&lt;seething intense hatred for the whole git submodule feature /&gt;

rerdavies, 2025-02-20:
Fixed in 1.4.70. Pipedal now builds and runs and installs  on Ubuntu 24.04 LTS, and 24.10 on both Arm64 and amd64/x64 platforms.

closed 2025-02-20 by rerdavies