rerdavies/pipedal #251: IR files loading

EE1977, 2024-11-08:
Hi Robin!! 

I had a hard time trying to upload IR (.wav) files from my USB flash drive attached to my pi.
I cannot see a way to directly browse and load the files from the Android remote app.
In the end, I copied the files directly in the SD card, and had to delete the audio plugins folder in var/pipedal 
(as suggested from panhardleft) for the pipedal to recognize them.

Did I miss something here?

rerdavies, 2024-11-08:
That is, regrettably, behaviour by design. For security reasons, I cannot allow the web interface to browse (or upload files) outside of the audio_uploads directory.

You can copy files into place on your Raspberry Pi; but if you copy using a method that also copies permissions from the source directories, there will be problems.

The permissioning of the audio_uploads directory and subdirectories thereof is fairly complicated. Fortunately, if there are permissioning problems, you can fix them by running the following command:

     pipedalconfig --install

(which , among other things, sets appropriate permissions on all existing folders and files in the audio_uploads directory).

If that doesn't work, run the following command, and post the results:

     sudo ls -lR /var/pipedal/audio_uploads

For your information only, the remainder of this post describes what permissions should be.  Offered only because it may give you insight into how permissions were broken. I would recommend using the procedure given above to fix permissions, as doing it through commandline utilities is perilous.

Permissions on /var/pipedal/audio_uploads should be 

     drwxrwsr-x 4 pipedal_d pipedal_d

or 

     drwxrwsr-x 4 root pipedal_d

(Note the 's' permission, particularly, which gives the pipedal_d  service account write access  to directories).

Permissions on files should be:

      -rw-rw-r--  <root or pipedal_d> pipedal_d

If you have collections of files elsewhere, you can symbolically link them into pipedal upload directories. If you do so, you should ensure the the pipedal_d service account has read access to the target directories. I would recommend against granting write access to files and folders in symbolically linked directories.

rerdavies, 2024-11-20:
I have replicated the bug. It happens on the first install, but goes away after subsequent installs. Deleting the audio_uploads directory will correct the issue; but a better solution is to run the following command:

    pipedalconfig --install

which will non-destructively reconfigure pipedal services, and also apply the correct permissions to the audio_uploads directory, without deleting any files that may be there already .  Any setup and configuration you have performed so far should be preserved.

Expect a fix for this in an imminent release. In the meantime, the best work around is to run the `pipedalconfig` command given above.

guysoft, 2024-12-03:
[image]
It looks like it will upload wav files but not WAV files.

closed 2025-02-21 by rerdavies