rerdavies/pipedal #258: Important additions / improvements

BorisSutin, 2024-11-27:
Hi Robin .  I'll write a list of fixes/improvements in one post

1. I have long asked to add the function of selecting the profile/impulse file when changing the line in list, so that the window does not close.

2. Plugin Aida-X problems with changing, loading, and displaying the selected profile. Check the entire plugin.

 3.  When loading a (preset / pedalboard), the default snapshot should be activated, the one that was active when the preset was last saved.  The preset is currently loading , but you have to select the snapshot manually. 

 4. You promised to add a file download function . If you remember, I built a plugin for recording.  
 https://github.com/BorisSutin/skhrecord.lv2 

 Your recommendation, files are saved in /var/pipedal/audio_uploads/shared/audio/Tracks, files saved in disk but are not displayed in the browser.   

5. Support for plugins that do not have "audio_input_port"
     2024-11-27 19:29:44.760 DBG: Plugin Audio File (http://kxstudio.sf.net/carla/plugins/audiofile) skipped. No audio inputs.

6. Add group or function displays only  favorite plugins.  

Thank you in advance !)

rerdavies, 2025-02-20:
A list of responses, in one post.... 

> I have long asked to add the function of selecting the profile/impulse file when changing the line in list, so that the window does not close.

Sorry. This one got lost in the shuffle. Good feature. please open a separate feature request so I can accept and track it.

> Plugin Aida-X problems with changing, loading, and displaying the selected profile. Check the entire plugin.

Regrettably, unfixable.  Aida-X is missing functionality required to handle LV2 path properties properly without using their custom UI. If you REALLY need to use Aida-X models, try Ratatouille.

> When loading a (preset / pedalboard), the default snapshot should be activated, the one that was active when the preset was last saved. The preset is currently loading , but you have to select the snapshot manually.

Rejected. 

There are 6+1 sets of patch properties in a preset: one for each of the snapshots, plus one more for the current control settings.  The current snapshot selectiion is saved, but ONLY if the preset settings haven't changed since a a snapshot was loaded or saved.   If the preset settings have changed, they will be different from the last-saved snapshot, so it would be an error to display the snapshot as selected in that case. Any control change will stop the active snapshot from being selected when you reload the preset. Bizarrely hateful feature. That's the way it needs to be implemented in order for the feature to make any sense at all.

> You promised to add a file download function . If you remember, I built a plugin for recording.
https://github.com/BorisSutin/skhrecord.lv2

Implemented. I'll deal with all of this in a separate message, as there's quite a bit to discuss.

> Support for plugins that do not have "audio_input_port"
2024-11-27 19:29:44.760 DBG: Plugin Audio File (http://kxstudio.sf.net/carla/plugins/audiofile) skipped. No audio inputs.

I took a significant kick at implementing this, but was ultimately unhappy with the resulting UI. The challenge is coming up with either UI to handle mixing inline out an audio route, or providing a general N channel mixer UI -- which is way out of scope for what Pipedal is. I tried impleenting the first approach, and disliked the implementation enough that I have disabled it for now.

The big problem: the most significant category of plugins without audio inputs are midi instrument plugins. Pipedal is nowhere near being able to support MIDI plugins at present (requires a big chunk of work),  but I don't want to cut off or impede future implementations of MIDI plugins in PiPedal with a half-thought-out feature that was added specifically and only to support zero-input audio playback plugins.

So my current feeling on this, after having experimented with hacking PiPedal's UI to support zero-input audio plugins:  There is exactly one and only one sort of plugin that currently needs this functionality -- and that is an audio file player once you take MIDI instruments out of the picture.  

So the apropriate interim solution -- I think -- is to provide a custom record/playback plugin specifically designed with the oddities of PiPedal's remote interface, and LV2's inadequate support of remote file properties in mind. 

I think we're close on that. You seem to have the functionality; and I have a sample project with a prototype UI that provides a relatively streamlined UI for both recording AND playback, but none of the actual file load/save functionality. And I'm pretty sure you'll find that fixes to better support LV2 toggle controls will provide you an easy path toward writing a looper plugin as well.

The third solution, and the one I actually prefer: to provide seperate input and output stream for external apps that want to record or play backing tracks. I actually have an Android implementation of a backing track player, which I was planning to hook up to a playback stream on the PiPedal server. And that may be the better solution in the long term: expose a connection point that apps can use to playback and/or record audio remotely .

rerdavies, 2025-02-21:
@BorisSutin 

So. I need a thread on which to discuss the recorder changes, so I am coopting this one. If you want to raise points on ANY other issue than the record/playback plugin, please start a new bug or feature request. 

One and only one issue per thread please.  Raising multiple issue in a single issue does not do me any favors. It makes it hard to track the status of where those issue stand.

From this point on, this thread deals only with 

> You promised to add a file download function . If you remember, I built a plugin for recording.

In this release (1.4.70) I have added a bunch of features specifically to support recording, playback and looper plugins. 

Most significant of these changes is correct support for LV2 trigger controls (and MIDI bindings thereof).  You may find that off-the-shelf loopers work much better. I did test the LV2 trigger controls against herman's looper plugin, which now works but probably isn't an actually usuable looper becuase the UI is a bit off base. 

With respect to recording, specifically, I coded up a test plugin so that I could test the functionality you were promised. The test plugin implements a plausibly decent user interface, but implements none of the background off-thread loading and saving of audio files.  You should probably take a look at it. Yes, PiPedal UI was tweaked to make the UI look better than was previously possible. yes, you should carefully consider basing your UI off mine, I think. 

The UI I designed for you: 

[image]

The LED to the right of record button blinks red while recording, and the  LED to the right of the play button blinks green when playing back.  (Note that to get the LEDs to blink, you have to change the value of the associated LV2 output control on and off in the realtime thread. Blinking doesn't happen by itself). 

You get the buttons in the screenshot if the title of an LV2 trigger control is a single character -- Unicode characters in this case for the stop, record and play Unicode characters..

Sources for the prototype are available here:

     https://github.com/rerdavies/lv2cairo/tree/main/src/download_test_plugin

and are currently built and distributed as part of the [lv2cairo](https://github.com/rerdavies/lv2cairo) project. There's a fair bit of hackery going on to get the UI to display they way it does. You should definitely take a look at the ttl used by the prototype; and may want to look at the .cpp implementation of the plugin to see how blinking lights are handled. If you need help creating a from-scratch project, let me know. 

As currently implement, you can download from the Recordings directory, but not upload; and you can upload to all other directories, but not download. This behavior is by design. You can, however, move recordings into other audio directories if you want.

SkyhawkBS, 2025-02-21:
Thank you for responding.  What about the recorded file download function? Will there be a download button next to upload

rerdavies, 2025-02-21:
*1. Downloading recordings.*

Yes. Follow the prototype's declaration of the path property to get the
desired result.  The prototype plugin declares that it wants files in the
Recordings folder, and the Audio folders. Users see both folders in the
File dialog.

As currnetly implemented in PiPedal , you can download from the
recordings directory only. I think we've discussed the rationale for that
before. Users can, however, move files in the recordings directory to any
of the audio directories.

The proposed implementation is this: that you create recorded files in the
Recordings directory with a name that includes the current date and time.
You then signal to the host that you have created a new file by sending a
Property Set message. That should cause the host to display the name of the
new file. Users can use the File Dialog to rename and/or move and/or
download recorded files.

I think you should also provide a play button so that users can preview
their recordings before downloading; and as a side-effect also allows the
same plugin to be used for loading files other than recorded files. The
prototype says that it is interested in fields from either the Recordings
director or the general-purpose Audio directory, which allows users to play
uploaded files as well.

I am toying with alternatives. Possibly a setting that allows or disallows
downloading of files, which allows users to opt in to potential security
issues when exposing directories from which external programs can both
upload and download files. Also a possibility -- limiting the number of
network hops that TCP/IP packets from the  web server can take. Limiting
web server traffic to two network hops would prevent the web server from
being accessed from outside the local network. Or MAYBE allow downloads
only to pre-configured IP addresses (does not work well with IPv6
connections, unfortunately). But for the meantime, the restriction is: can
upload to any directory except the Recordings directory; can download only
from the Recordings directory. :-/

*2. Supporting compressed recording files.*

Another point worth bringing up while we're discussing your project... MP3
patents expired last month!! That means that you could use ffmpeg libraries
to compress files using MP3 as an option. It seems like a useful feature to
allow users to record to WAV, FLAC or MP3.

ffmpeg is licensed under LGPL, so it's also compatible with the PiPedal
license (if you ever decide that you would like to contribute your plugin
to the ToobAmp/PiPedal projects. It may actually be a safer approach, as
long as you only allow FLAC and MP3 encodings. M4A patents are still live,
so you should probably avoid M4A support.  The easiest way to implement
compression would be to make sysexec  the FFMPEG commandline program to
perform the compression which -- I think -- would avoid the potential for
painful linking problems if you want to distribute one binary that runs on
Raspberry Pi OS AND other debian distros like Ubuntu. This may not be true
for FFMPEG, but it is most definitely true of the FLAC libraries, which
have odd problems with cross-platform support due to library versions and
availability of static libraries on Ubuntu and Rasbian.

*3. Native UI for your plugin.*

Also worth mentioning at this point. If you want a native UI for your
plugin, I have not yet added the UI hacks for trigger controls to the
Lv2Cairo package (which generates native UIs from Lv2 .TTL files). If you
decide that you want to use Lv2cairo to provide a native UI as well as a
Pipedal UI, let me know when you have your plugin up and running, and I
will make the required changes to Lv2Cairo  that you would need for your
project.  (Lv2Cairo can provide a native UI even if your core plugin isn't
based on the Lv2Cairo plugin library).

Regards,

R.

SkyhawkBS, 2025-02-22:
Robin, the plugin is interesting, I'll be waiting for the release.   Add only the threshold parameter to start recording. It seems to me that plug-ins such as tuner, looper, and record player can always be kept in the signal chain outside of the preset so that they are always at hand.

rerdavies, 2025-02-22:
Cart before the horse, I'm afraid. This beta has two major features (plus a
handful of bug fixes):

- Support for Ubunu (arm64 and x64)

- Imrpovement to UI controls required to support record/playback/looping.

To be promoted to a full release, I need two things:

1. A couple of weeks to let the fairly substantial changes required to
support Ubunutu get some field testing.

2. A working record/playback/looper plugins that demonstrate that the UI
changes are fit for purpose.

So it won't go full release until item 2 has been completed.

The UI changes have been tested using the Lv2Cairo Proof-of-concept plugin.
The next step would be to fill in the actual audio backend for that, (and
build a looper using that audio backend).

I was -- to be perfectly honest -- kind of offering you  a chance to put
any work that you have done so far on the recording plugin to good use. But
if you're not comfortable with that, I'm perfectly happy to implement the
backend myself. it seemed unkind to me to stomp all over any work you have
put in so far.

Let me know which path you would prefer. (tbh, for me, it's probably less
work for me personally to do it myself).  And more efficient implementomg
them as TooB Amp plugins since they would share substantial code with the
TooB Amp project.

Regards,

R.

SkyhawkBS, 2025-02-22:
I will be waiting for your implementation.

rerdavies, 2025-03-13:
TooBb One-Button Looper, and TooB 4Looper plugins added. Will be available in the next release.

closed 2025-03-21 by rerdavies