rerdavies/pipedal #261: looper issue

EE1977, 2024-12-31:
Hi Robin!!
I am still exploring the possibilities of Pipedal and have encountered a strange issue with the SooperLooper plugin. When I use my touchscreen to control the REC, PLAY, and RESET buttons, everything works fine. However, when I try to use my footswitch MIDI pedal for the same functions, the RESET feature causes problems, primarily resulting in reconnections with no audio. Do you have any suggestions for resolving this issue?

rerdavies, 2025-01-02:
I have some revisions to how trigger controls pending, that should be published very soon (I've been working on UI controls that Boris will want for his various plugins).  The changes may or may not help.  If they don't I'll take a look.

I'll ping you when I publish them.

EE1977, 2025-01-02:
Thank you all!! Boris I'll definitely give it a try!! Wishing you all a happy and resourceful New Year!

rerdavies, 2025-02-20:
Took me a bit longer to get this out than I expected. 

The new 1.4.70 release contains fixes to the behaviour of LV2 Trigger controls, and to MIDI bindings to LV2 trigger controls. If you could give it a try, I'd be grateful. And if that doesn't fix your problem, I should be able to turn around a fix a little more promptly.

rerdavies, 2025-03-13:
Added support for `http://moddevices.com/ns/mod#preferMomentaryOffByDefault` and `http://moddevices.com/ns/mod#preferMomentaryOnByDefault` port properties which provide Buttons and midi bindings with long-press (and double-tap) support.

rerdavies, 2025-03-21:
Probably fixed in v1.4.74. (Plus two new TooB Looper plugins that definitely work).

closed 2025-04-02 by rerdavies