rerdavies/pipedal #263: MIDI snapshot saving

rerdavies, 2025-01-02:
### Discussed in https://github.com/rerdavies/pipedal/discussions/260

<div type='discussions-op-text'>

<sup>Originally posted by **EE1977** December 19, 2024</sup>
Hey Robin!

I hope you're doing well! I have a fun idea I wanted to share with you. Wouldn't it be great to have an option to save snapshots directly from MIDI? For instance, I could use my MIDI foot controller to pick the snapshot I want, make a few tweaks right from the controller, and then save my updated snapshot just by long-pressing the snapshot button switch. What do you think? Can't wait to hear your thoughts!</div>

rerdavies, 2025-01-02:
Out of curiousity, what have you bound your footswitches to, in order to select presets? MIDI Program changes? Or are you using the Next/Previous program system bindings?

EE1977, 2025-01-02:
Currently, I am using the next/previous system bindings.