rerdavies/pipedal #265: Add alsamixer section in documentation

GentlemanRider, 2025-01-18:
**The issue I had:**
I recently purchased a short scale bass and a cheap guitarlink USB card. The setup went all together really easy but I had to keep the volume on the bass very low and everything was always on the edge of digital saturation. I was thinking that I cheaped out too much, or most of the effects and plugins were made for guitars and didn't sound right. 

**What was happening**
The input gain was not set properly at a lower level than the Pipedal gui allows to tweak. The guitarlink has a peak LED on the input and I could not manage to light it even with a volume way above what I needed to use to avoid distortion.

**How I managed to fix it**
- ssh to the host
- run _alsamixer_
- press F6 to select the sound card  (USB Audio Codec in my case)
- press TAB to scroll across pages / channels until I reached the CAPTURE slider
- use up / down arrow to adjust the gain, keeping Pipedal gui visible, so that I can use the volume knob on the instrument at full range
- press esc to close alsamixer
- run _sudo alsactl store_ to save the configuration

rerdavies, 2025-01-18:
That is.... unfortunate. Do you have to reset the levels every time you reboot, or are the settings persistent?

I may add some UI to support this scenario.  I think at least one of my devices has level settings that accessible via ALSA. Doubly unfortunate because many (most?) audio adapters have knobs on the device to set input and output trim, so having PiPedal set the levels at boot time would be the Wrong Thing To Do. :-(

Let me know. Thanks.

GentlemanRider, 2025-01-18:
The settings are immediately effective when adjusting with pipedal on the browser and alsamixer on the shell. The last command makes them persistent and they survive reboots. It is a one time task and it can be considered part of the initial setup procedure.

If you ask me I think that mentioning this aspect in the documentation totally solves the issue, but I am more a computer guy than a musician. It's just a couple of commands by the way: I think that also less technology oriented people can perform the task from the shell. I can prepare some screenshots and instructions if helps.

rerdavies, 2025-03-13:
Added the documentation. Should show up in the next release.  

Thanks for your detailed description of the configuration procedure.

rerdavies, 2025-03-21:
Fixed in v1.4.74. Thanks.

closed 2025-03-21 by rerdavies

GentlemanRider, 2025-03-23:
Glad to support you project in a way that helps running low budget devices