rerdavies/pipedal #270: USB UAC2 function

SkyhawkBS, 2025-02-22:
Robin, can you tell me if it is possible to add such a feature?  It would be very cool to implement a bridge between the current audio device and the USB UAC2 gadget. I tried to do this using Linux, but I didn't get the right result. I have launched UAC2, if you select it as the interface in PiPedal, then everything works. It would be great to make Pipedal a full-fledged guitar processor with USB recording capability on a PC. Maybe you can try adding a route programmatically between UAC2 and the current audio interface.  Send an audio stream from the end of the chain to UAC2 and return the playback  from the computer to the current audio port. I hope I explained the task correctly and you understood me.

rerdavies, 2025-02-22:
No. I'm not getting what it is you want to do. You want to have TWO audio devices, and copy from the primary audio device to the secondary device? (To put things in perspective, from my perspective, the NORMAL and most common use-case of PiPedal is that the UAC-2 would be the primary audio device, rather your PiPedal audio hat. With current Linux kernels, there's not much reason to prefer an internal audio hat to a USB hat, tbh, and maybe some reason to prefer external USB audio devices (probably lower SN/R being the most important among them). The UAC-2 (unlike most cheap USB adapters) actually has a reputation for providing a remarkably clean audio signal, fwiw.

One of the significant challenges with two active ALSA devices is the issue of clock-rate matching. The physical master sample clock of two audio devices varies depending on the components, and will invariably differ by a few herz.  So whenever two USB devices are involved, one of them has to use a "slave" sample clock. Slaved clocks on USB devices are perilous. Slaving is performed by estimating the clock rate of the other USB device based on the rate at which samples arrive over the USB bus.   This is a messy and dangerous and inexact procedure, which often involves adding an extra buffer in the signal chain, and runs the risk of generating odd "phasing" audio artifacts. And I'm not honestly sure that all USB devices can do it, or whether I2C audio devices like yours can do it at all. The problem of clock-rate matching is the reason why PiPedal does not allow you to select input from one device, and output audio to a different device, but instead requires both channels to be on the same device.  Simply put, I don't trust ALSA, and audio devices involved to perform sample-clock slaving without introducing odd audio artifacts or additionally latency.  AND, it's a nightmare feature to test properly. 

A simpler solution would be for PiPedal to implement a custom ALSA virtual audio device. This is doable, I think, and is currently somewhere on the current development map. I intend to do it eventually. And probably also provide a secondary audio stream for playback and recording via HTML at the same time so that external software could do it using either PiPedal's custom ALSA device OR stream audio remotely via HTTP. Not currently top of my development backlog, but I have to admit I would REALLY like a solution for remotely playing backing tracks for my personal use.

Or in the shorter term, one could do it via a record/playback plugin.  ;-)  Which has the advantage that any such work could be quickly leveraged to provide a sensible functional looper plugin. I've actually tried a few LV2 looper plugins, but so far, the ones I've found seem to be universally awful and not truly usable. I have a good friend who regularly performs live using looping, so I have a reasonable idea of what a pro-level looper should look like.  The big advantage on PiPedal is that the UI controls are always going to be kind of sloppy and probably unusable no matter what the host is, but MIDI-bound controls would be perfectly good and precise for pro-level looping. The core issue: MIDI signals are accurate to within a millisecond or so; UI controls and mouse/touch interactions are often (always?) going to have 100ms+ jitter.

If you're having trouble implementing the audio backend for your plugin, let me know and I will implement it myself. It's actually a non-trivial piece of code to implement well, and I was hoping you could save me some time. (And it seemed unkind just to do it myself, invalidating any work that you have done yourself). If you would rather not do it, it would probably take me a bout a week or so to write it myself. And a record/playback/looper plugin is probably the top of my development backlog at present (significantly ahead of a secondary record/playback channel).

SkyhawkBS, 2025-02-22:
OK, I got it. Since the UAC2 option is difficult to implement, I will do my own via dsp. I'm currently using the adau1701 as an audio interface. I plan to add an STM32 with USB Audio 2.0 to it and create an internal route using dsp.

closed 2025-02-23 by rerdavies