rerdavies/pipedal #278: [BUG] pipedal_1.4.72_arm64.deb NOK on my PI4

maarthome, 2025-02-27:
**Describe the bug**
Some pugins do not work anymore. More annoying is a hight pitched peep.....

Luckily version pipedal_1.3.64_arm64.deb does work, so I am using this one...

** Environment (please complete the following information):**
 - OS: [Raspbian]
Linux pipedal 6.6.74+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux

Cheers,
Maarten

SkyhawkBS, 2025-03-01:
What kind of hight pitched peep in the sound are we talking about. Do you have any artifacts in the sound? Or a problem with power interference ? What kind of equipment do you have?

maarthome, 2025-03-01:
Hi @SkyhawkBS,

Thanks for your message.

With version 1.3.69 everything works fine.

With 1.4.72 there is something NOK with the right channel. Almost all effect sound is coming from the left channel.
With some pedals there is a high pitched sound in the right channel. 

Some plugins do not work like ShiroVerb....

I am using a Focusrite Scarlett2i2. I cannot imagine that this is causing the problem.
Also the power supply. It is the same I always use.

I can live with using 1.3.69 only and wait for a future release. So do not spend to much time on this.
But I am surprised that nobody else encounters these issues.... 

Cheers,
Maarten

SkyhawkBS, 2025-03-01:
I think that Robin will soon introduce an updated version, in which we will see a lot of new things and fix all the bugs.

maarthome, 2025-03-01:
I can't wait! :-)

rerdavies, 2025-03-03:
@maarthome: 

The high-pitched whine on the right channel is because you haven't configured your audio adapter properly in PiPedal. Your audio device transmits audio from the left input jack on the left channel, and audio from the right input jack on the right channel. My guess is that you have a device plugged into the left plug, and nothing plugged into the right plug. The best solution is to configure PiPedal to only process the right channel. In the Settings page,  right below the device configuration, you can select the channels: Left, Right, or Stereo.  Select "Left", and the problem should go away. You should probably also turn the volume trim for the right channel down to zero, just to minimize the amount of noise the rest of the adapter is being exposed to. 

As for broken plugins.... I have no particular reason to think that the current version breaks plugins. Or that the version after will fix problems with plugins. So if you could provide details on the plugins that are not working for you I would be grateful.

rerdavies, 2025-03-03:
ps. if it's Shiroverb Mk II, that's a plugin with stereo inputs, so every reason to think that it's going to be unhappy if it's getting nothing but noise on the right channel. Let me know which plugin(s) were broken, and how they were broken.

maarthome, 2025-03-03:
Hi @rerdavies,

Thanks for your explanation.

But I do not understand that when I use a preset with version 1.3.69 it sounds fine and when I use the same preset with 1.4.72 it is NOK.
I alternated  the installation of 1.3.69 and 1.4.72 several times and each time the same issue.
I experimented with switching to only the left channel (guitar) and indeed the pitched sound was gone but effects where missing on the right audio channel and shiroverb was not working wit 1.4.72...

Nobody else has this issue?

Cheers and thanks,
Maarten

rerdavies, 2025-03-03:
The reason why it differs when you change the input configuration is that
PiPedal is now feeding the left Jack input on both left and right channel
to the first effect on your preset chain instead of a left channel
containing the signal from the left Jack and a right channel containing
noise from your right input jack, which is noise because the input signal
is floating,  and because you have turned up the volume on the right input
on your sapphire. The preset files contain plugin configuration, but not
the configuration of the audio device, which is a system setting, not a
preset setting.

I'm not sure why you're not getting effects on the right output. Could it
possibly be that you have configured the audio output to Left channel only
on the old version, but to stereo on the new version?

Other than configuration of the audio device, the effects should be running
identically on both versions. There are changes in how files are handled,
but I don't think your reverb unit is a convolution reverb, is it?

If you think you're reverb isn't loading a convolution reverb file, let me
know which plugin you are using, and where you got it from and I will give
it a try myself. And maybe post the preset here as well so that I can see
what's going on.

maarthome, 2025-03-04:
Hi Robin,

Thanks for offering to look at my preset.

Here is my preset:

[MDin_song1_part1_Git_Voc_v2.piPreset.docx](https://github.com/user-attachments/files/19068435/MDin_song1_part1_Git_Voc_v2.piPreset.docx)

Please remove the .docx at the end.
(It is not possible to upload .z or .zip or .rar files)

Cheers,

Maarten

rerdavies, 2025-03-12:
OK. So the prooblem is the LR split at the start of the chain, and the fact that you most probably have not configured your audio input device the same on both installs. 

IF your device provides one channel, the L/R split will send the same signal to both chains; if your device has two audio inputs and you the audio device is configured for stereo input, then the input from the left jack of your Focusright Scarlet will be sent through the top chain, and the input from the right jack of your Focusright Scarlet will be sent through the bottom chain.  If your audio device has only one audio channel, then the single channel will be converted to stereo by sending the signle channel to both left and right inputs of the L/R split, so you get the same signal being sent on both the upper and lower routes. So the patch will behave very differently if Input Channels is set to Stereo. Setting the split type to Mix might have been a better choice.  A Mix splitter sends the same input to both upper and lower routes (stereo, if it is stereo), but also provides a control that allows you control the mix of the two effect chains when they are combined at the end of the split.

The solution is straightforward. Go to PiPedal Settings, and tap the Input Channels setting. Select either Left Only, or Right Only depending whether you have plugged your instrument into the left or right jack. You should -- as a matter of good practice -- also turn down the volume knob on your unused jack to zero just to make sure you aren't leaking noise into the Focusright Scarlet. The unconnected jack will act like a radio antenna, and will pick up some amount of RF emission from nearby computer devices, which will be quite audible if you have the volume input knob maxed out.   

All of this assumes that you aren't deliberately trying to process two independent input channels.

rerdavies, 2025-03-13:
OK! Got it. Will be corrected in the next release.

rerdavies, 2025-03-13:
@maarthome: Ridiculously great reverb, by the way!

maarthome, 2025-03-14:
The shiroverb is think you mean? or the parralel combination of dragofly and shiroverb?

maarthome, 2025-03-19:
Thanks Robin,

1.4.73 beta solved the problem :-)

Cheers,
Maarten

closed 2025-03-19 by maarthome