rerdavies/pipedal #279: Works perfect for my needs - some thoughts

shrjne, 2025-03-01:
Thank you for the current state of pipedal. As a guitarist this solved a problem I created of wanting to run a NAM setup on my pedalboard. This is a non issue for performing, but for QOL I would love the option to change the name of the blocks, or give them custom labels in addition to. 

Thought it was worth mentioning as I am running multiple instances of NAM to get access to some distortion/OD pedals. The folder feature is what made this my definitive choice to run it, but the labels would be the cherry on top

rerdavies, 2025-03-03:
Requested definitely noted. :-) Thanks.

rerdavies, 2025-03-13:
Regrettably, there isn't enough tap-target in the main UI when being used from a phone or tablet -- the label is too ; and not really enough space to add a button for this functionality. :-(

However, I think I can splice it into the Info dialog.  Not a piece of I that is particularly pretty; but I think it's a good piece of functionality.

Deferred to Next Release+1.

rerdavies, 2025-06-25:
Top on the name/author of the plugin, launches a dialog which allows users to override the default name of a plugin.

rerdavies, 2025-06-25:
Available in the next published build.

rerdavies, 2025-07-01:
Implemented  in version v1.4.77 (just published).

Tap on the plugin name to open a dialog which allows you to override the default name. Good feature request. Thanks.