rerdavies/pipedal #281: [BUG] Android app seems to have communication problems

dcoredump, 2025-03-12:
**Describe the bug**
I use pipedal_1.4.72_arm64 on a Raspi5(8GB). The whole thing seems to work fine when I use the web interface. When using the Android app, the app (or rather the app interface) seems to hang up at some point and you can't make any changes

**Screenshots**
-

** Environment (please complete the following information):**
 - OS:  Raspbian (Debian GNU/Linux 12 (bookworm))
 - Linux pipedal 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
 - Pi5 (8GB)
 - PiPedal Version 1.4.72
If UI-related:
 - pipedal Android app (make problems), Chromium on PC works

rerdavies, 2025-03-12:
How much RAM does your phone have?  If it's < 2GB, let me know, and I'll review to see what I can do to reduce memory usage in the app.

rerdavies, 2025-03-12:
I've tried it on a variety of phones (the smallest phone/table I have has 2gb), and I haven't been able to replicate the problem.  Perhaps we should look at system logs to see if something odd is going on. 

1. Reboot your raspberry pi.
2. Reproduce the issue using the steps you gave above.  
3. At a system prompt on the PiPedal device, run the following command:

```
journalctl -b0 | grep pipedal > capture.txt
````

And post the output here.

dcoredump, 2025-03-13:
Thanks for your fast response. In fact I used my old tablet (which says it has 2GB RAM), but using my much newer phone works like a charm! So no further debugging is necessary!

Thanks for this nice software. Hope I can try it soon on the PA in the rehearsal room!

closed 2025-07-19 by rerdavies