rerdavies/pipedal #290: [Feature Request] Output gain control on TooB Tone Stack

DuraMorte, 2025-03-22:
Being able to add some gain at the output of the Tone Stack would be very helpful.
Hopefully this won't be terribly tough to accomplish, and would make life much easier.
Thanks for all your hard work; I'm loving PiPedal, and I can't wait to show my new "guitar rig" off to my friends!

rerdavies, 2025-03-23:
I need to think about that one a bit. UI space in PiPedal is kind of precious.  In theory, NAM models are supposed to set a trim level internally so that output is normalized to a proper volume. Newer NAM models always do this automatically as part of the training process; older models sometimes do not. 

Are you using a model that gives drastically wrong results? If so let me know which model it is, and I'll take a look. 

As a workaround, you can adjust the final volume by tapping the final output pin on the pedalboard.  Since most effects that go after an amp don't have non-linear response, this would be relatively harmless. A reverb effect, for example, doesn't really care whether it's input level is over- or under-driven, since all calculations are done in floating point arithmetic. Not to trivialize your problem. But maybe it will get you through for now.

If, on the other hand, you're trying to match bypassed and un-bypassed volume levels on something like an overdrive pedal.. Well... ok. I've thought about it. That probably would make a good feature request. I will implement that, probably in the next release. 

But let me know if it's a problem only with specific models anyway. There are some oddities in the original NAM Library implementation relating to output normalization, and some recent changes that the ToobAmp plugin may not have merged. I think the model trim level is working properly. It has been tested. But it's possible that the plug-in is out of sync with the upstream library.

@

rerdavies, 2025-03-29:
Huh.  Whether or not to provide a makeup gain control to the Tone Stack.... I'll have to sit n that for a bit. Maybe.  I'm leaning toward adding a utility Level plugin that does nothing but adjust gain and/or pan the input signal.  I haven't actually a utility plugin like that in any of the various LV2 libraries, despite the fact that it seems like a very sensible plugin. And I've been meaning to add one to the TooB plugin collection for a while. 

In the meantime.... I have a ridiculous workaround for you. Not sure if you're aware, but the EQ in the NAM plugin is actually pre-model. And the EQ units in the NAM plugin are using exactly the same code as the Tone Stack plugin. So.... If you *really* need to gain match the EQ (because you  want it to be bypass-able), you COULD  insert another instance of the NAM plugin. Do not load a model; and set the Noise Gate to off (minimum position).  I just took a peak at the code. The EQ unit will still be active even though there isn't a loaded model; and so will the input gain. if you are not trying to gain-match a bypassable EQ plugin, then you might as well just use the input gain in the NAM plugin.

rerdavies, 2025-05-14:
So. The most recent release includes a TooB Volume plugin. A plugin that does nothing but adjust volume. I've been meaning to add it for a while,  but it was partly motivated by your feedback.  

On re-reading your post, though, I'm not really sure it addresses your issue properly. On reflection, I think I'm going to add the makeup gain control as well in a forthcoming release.

rerdavies, 2025-07-01:
Implemented  in version v1.4.77 (just published).

Sorry for the vacillation on this issue. There is tension between useful functionality and unnecessary UI complexity. In the end, your use case was compelling enough.

closed 2025-07-19 by rerdavies