rerdavies/pipedal #291: [Feature Request] Toob Tone Stack adjustment arrows should use .1 increments

DuraMorte, 2025-03-22:
The Toob Tone Stack is extremely useful in it's current iteration, but having finer control of the knobs via the adjustment arrows would be incredibly helpful.
Either .1 or .05 increments would be great. The total travel is 0.00 to 1.00, so having increments of 1 isn't the best.

rerdavies, 2025-03-22:
How to do find control: on a desktop, the and/or CTL  keys (can't remember
which off the top of my head) allow fine and ultra-fine adjustments. On a
touch display, having two fingers down gives fine control, and three
fingers down gives ultra fine control. I find the easiest way to do the
gesture is to touch with your middle finger first, and then your index
finger and then drag  from there.

DuraMorte, 2025-03-23:
> How to do fine control: on a desktop, the ALT and/or CTL  keys (can't remember
> which off the top of my head) allow fine and ultra-fine adjustments. On a
> touch display, having two fingers down gives fine control, and three
> fingers down gives ultra fine control. I find the most comfortable way to do the two-finger
> gesture is to touch with your middle finger first, and then your index
> finger and then drag  from there.
> […](#)

Very cool! Thanks for the rapid reply! Your software is awesome!